Xilinx/mlir-aie #7: Invalid compilation target `aie`

hanchenye, 2021-09-08:
During the regression tests, quite some fails are caused by this command:
```
llc acdc_project/core_7_3.stripped.ll -O2 --march=aie --filetype=obj -o acdc_project/core_7_3.o
```
with an error:
```
llc: error: error: invalid target 'aie'.
```
I dug a little into this and found the path of `llc` is the common `llvm-project` build, which doesn't support `aie` as target. Any idea on this?

Thanks,
Hanchen

stephenneuendorffer, 2021-09-08:
Yes...I'm aware of this.  Basically we need to have a cmake configuration option that sets the default options in aiecc.py to be -xchesscc -xbridge.  Currently the default is trying to run LLVM directly.  In the meantime you can locally modify this:
```
xbridge_found = False // Change to True
xchesscc_found = False // Change to True
```
in tools/aeicc/aiecc/cl_arguments.py

hanchenye, 2021-09-08:
Thanks, Steve! `xchesscc_wrapper` works now, but I encountered another issue:
```
Command Output (stdout):
--
Error encountered while running: llvm-link acdc_project/core_1_3.chesshack.ll acdc_project/chess_intrinsic_wrapper.ll -S -o acdc_project/core_1_3.chesslinked.ll

--
Command Output (stderr):
--
llvm-link: acdc_project/chess_intrinsic_wrapper.ll:13:83: error: expected metadata or 'align'
  store volatile i8* inttoptr (i32 1 to i8*), i8** @_ZL23__chess_separator_dummy, noalias_sidechannel i8** undef, align 4, !tbaa !2
                                                                                  ^
llvm-link: error:  loading file 'acdc_project/chess_intrinsic_wrapper.ll'

--
```
It seems the `chess_intrinsic_wrapper.ll` generated by `xchesscc_wrapper` cannot be parsed by `llvm-link`. This error is emitted by the LLParser. I'm not sure whether it's the `xchesscc_wrapper` or the LLVM side that is causing this issue. Has this been encountered this before?

stephenneuendorffer, 2021-09-08:
Can you make sure that the llvm-link in your path is the the one that you've compiled (close to LLVM head) and not the one that you're compiling with?

hanchenye, 2021-09-08:
Right, I tried this using both the CIRCT LLVM and the LLVM suggested by the README. They are generating the same error.

hanchenye, 2021-09-08:
I'm trying this offline by:
```shell
~/workspace/llvm-project/build/bin/llvm-link \
    ~/mlir-aie/build/test/unit_tests/00_itsalive/acdc_project/core_1_2.chesshack.ll \
    ~/mlir-aie/build/test/unit_tests/00_itsalive/acdc_project/chess_intrinsic_wrapper.ll -S
```
and the error generated is:
```shell
/home/hanchenye/workspace/circt/llvm/build/bin/llvm-link: ./test/unit_tests/00_itsalive/acdc_project/chess_intrinsic_wrapper.ll:13:83: error: expected metadata or 'align'
  store volatile i8* inttoptr (i32 1 to i8*), i8** @_ZL23__chess_separator_dummy, noalias_sidechannel i8** undef, align 4, !tbaa !2
                                                                                  ^
/home/hanchenye/workspace/circt/llvm/build/bin/llvm-link: error:  loading file './test/unit_tests/00_itsalive/acdc_project/chess_intrinsic_wrapper.ll'
```

hanchenye, 2021-09-08:
Attached the wrapper `ll` file generated on my side for your reference: [chess_intrinsic_wrapper.txt](https://github.com/Xilinx/mlir-aie/files/7132143/chess_intrinsic_wrapper.txt)

stephenneuendorffer, 2021-09-08:
Thanks, I see the problem.  We're playing a little fast and loose with the llvm IR we're feeding to the Chess compiler.

closed 2021-09-09 by stephenneuendorffer

hanchenye, 2021-09-09:
Nice! Thanks, Steve :D

PR #10 (2021-09-09): Improve integration with Chess compiler
fixes #7
commit 7da58141 (2021-09-09): More adaptation between modern LLVM and Chess

Chess uses (approximately) llvm-8.
This patch removes the noalias_sidechannel attribute (not used in
LLVM HEAD) and formal argument names (not used in llvm-8).

fixes #7