Xilinx/mlir-aie #8: Could NOT find LibXAIE (missing: XILINX_XAIE_INCLUDE_DIR XILINX_XAIE_LIBS)

hanchenye, 2021-09-08:
I built up the vck190 bare platform following the [board setup instructions](https://xilinx.github.io/mlir-aie/Platform.html), but the generated rootfs at `mlir-aie/platforms/vck190_bare/petalinux/sysroot/sysroots/aarch64-xilinx-linux` doesn't contain `/opt/xaiengine`, making cmake fail to find the `LibXAIE` package. I must have missed something here, any ideas?

Thanks,
Hanchen

jackl-xilinx, 2021-09-09:
The reference for `LibXAIE` package in the `test/CMakelists.txt` is hardcoded but was changed in the latest 2020.1 bare platform + sysroot. I've fixed it so the reference is correct for this platform but going forward, if we change platforms and need to point to a different location for the `LibXAIE` package, we can add the cmake argument `-DLibXAIE_ROOT=<sysroot pointer to directory containing libxaie/lib and libxaie/include`. For example, to compile for the older platform, we would add `-DLibXAIE_ROOT=<sysroot base>/opt/xaiengine`

jackl-xilinx, 2021-09-09:
There's another location where sysroot type information is hardcoded (`aiecc.py`) so that needs to be adjusted as well. I will create a new issue to track it until it is fixed but you'll encounter this when moving onto the `ninja check-aie` step.

stephenneuendorffer, 2022-04-19:
@jackl-xilinx I think this is fixed now, but we don't document the LibXAIE build process.  Can you add some notes in GettingStarted.md?

jackl-xilinx, 2022-11-17:
Build process is being updated so this will be incorporated in the new build instructions.

closed 2022-11-17 by jackl-xilinx