Xilinx/mlir-aie #9: Error during test compilation (ninja check-aie, when running aiecc.py) (Error generated: cassert non found, bits/c++config.h not found)

jackl-xilinx, 2021-09-09:
The `aiecc.py` has some hard coded references to xaiengine headers/libs in the sysroot that doesn't match the current platform. This reference should be corrected and made more generic and flexible. 

Manually having the clang call in aiecc.py start with the below change works but is itself too inflexible.
`clang --target=aarch64-xilinx-linux -std=c++11 --sysroot=<sysroot base> -I<sysroot base>/usr/include/c++/9.2.0 -I<sysroot base>/usr/include/c++/9.2.0/aarch64-xilinx-linux -I<sysroot base>/usr/include/c++/9.2.0/backward` ....

hanchenye, 2021-09-12:
Thanks for tracking this! By manually referencing the `xaiengine` headers and libs, I've successfully generated some executables and mapped them to a VCK190 board :partying_face:

stephenneuendorffer, 2022-12-05:
Using `clang --target=aarch64-xilinx-linux -std=c++11 --sysroot=<sysroot base> --gcc-toolchain=<sysroot>/usr` should be sufficient.

closed 2022-12-05 by stephenneuendorffer