Xilinx/mlir-aie #22: Build direct-to-binary backend target

stephenneuendorffer, 2021-09-22:
Current backend targets use the LibXAIE API to configure the AIE array through memory-mapped accesses.  This has a number of downsides:
1) LibXAIE has significant memory usage and we want to run this configuration on an embedded microblaze.
2) LibXAIEv1 lacks a way to aggregate configuration.  (LibXAIE does have a mechanism that bundles reconfiguration)
3) LibXAIE assumes that a processor is doing configuration, while we may need to push configuration through AXI streams

LibXAIEv2 may have some pieces that can be leveraged to achieve this, or at least verify that the result is correct.

jgmelber, 2023-03-14:
#243 is a start towards this issue, my current plan is to refactor this to target a binary format familiar to Vitis

jgmelber, 2023-11-30:
#668 addresses this for AIE1 on VCK5000