Xilinx/mlir-aie #39: -aie-standard-lowering doesn't handle AIE.wire

hanchenye, 2021-10-17:
Hi,

I'm playing with a test case like this:
```
  %0 = AIE.tile(1, 1)
  ... ...
  %266 = AIE.core(%0)  {
    ... ...
  }
  ... ...
  %339 = AIE.switchbox(%0)  {
    AIE.connect<DMA : 0, East : 0>
  }
  ... ...
  AIE.wire(%0 : Core, %339 : Core)
  AIE.wire(%0 : DMA, %339 : DMA)
```

When `aiecc` calls `aie-opt --aie-standard-lowering=tilecol=1 tilerow=1 acdc_project/input_with_addresses.mlir -o acdc_project/core_1_1.mlir`, the `AIE.tile` is erased while `AIE.wire`s still using it. Error massage:
```
acdc_project/input_with_addresses.mlir:8:8: error: failed to legalize operation 'AIE.tile' marked as erased
  %0 = AIE.tile(1, 1)
       ^
acdc_project/input_with_addresses.mlir:8:8: note: see current operation: %0 = "AIE.tile"() {col = 1 : i32, row = 1 : i32} : () -> index
acdc_project/input_with_addresses.mlir:2409:3: note: found live user of result #0: AIE.wire(%0 : DMA, %300 : DMA)
  AIE.wire(%0 : DMA, %339 : DMA)
```

Did I miss anything, such as `AIE.wire` should be handled in an earlier stage of the pipeline?

Thanks,
Hanchen

stephenneuendorffer, 2021-10-17:
I doubt you're missing anything.  I've gone back and forth as to whether AIE.wire ops are useful or not.  Currently they exist primarily for the --aie-find-flows pass, but we've discussed removing them entirely.

AndraBisca, 2025-04-02:
To follow-up on this, the final decision is to remove AIE.wires entirely and as a result the AIEFindFlows pass and the visualization tool will need to be reworked.

closed 2026-03-17 by hunhoffe