Xilinx/mlir-aie #74: Does deleting the "#include" code affect the result?

Gakkkkkki, 2022-01-12:
When I built AIE tools on x86, I met a fatal error " 'aie/Dialect/ADF/ADFDialect.cpp.inc' file not found."  as the following  picture.
[image]
Then I checked my build folder, and I do miss the file. So I delete the code in the cpp file and built successfully. Does deleting the "#include" code affect the following test results?

stephenneuendorffer, 2022-01-12:
Hi..  the .inc files are automatically generated by the build.  This is usually an indicator of an incorrect build setup that is missing some dependencies, or perhaps a bug in an older version of cmake.  you should be able to work around it by running 'make -k 0'.  Generally, we use ninja which is more predictable about dependencies on generated files.  What version of cmake are you using?

Gakkkkkki, 2022-01-12:
Thanks for your reply.
According to your /docs/Building.md, I installed cmake 3.17.5, ninja 1.8.2, Xilinx Vitis 2020.2(differ from 2020.1 in your doc) and jobilb.
[image]
[image]

First of all, I git clone the llvm-project and then git checkout commithash(provided in /utils/clone-llvm.sh) like this.
[image]

Next, I cmake using the following instruction(provided in /utils/build-llvm.sh) in the created build folder but I met an error like this:
[image]
[image]
So I delete the LLD button like this:
[image].
After I cmake sucessfully, I use [image] and compile LLVM successfully.

To build AIE tools, 
[image] were both git. Then I cmake using the following instruction(provided in /.github/workflows/buildAndTest.yml) in the create build folder in mlir-aie project:
[image]
[image]
Then I met the fatal error:
[image]

I had post what I did after meeting the error in this issue.  This does confuse me for a whole while.
Look forward to receiving your reply ASAP!  :-)

stephenneuendorffer, 2022-01-17:
It looks like you're still using make to compile.  To use ninja, run "cmake -GNinja ...." and use ninja to build.

closed 2022-07-07 by stephenneuendorffer