Xilinx/mlir-aie #87: xchesscc compilation issue with floating-point vectorized C++

hanchenye, 2022-02-03:
Hi @pssrawat, I generated a vectorized GEMM kernel using the aievec-to-cpp exporter. The int32 version can be successfully compiled, but the float version cannot. The error message is:
```
segmentation violation
This error occurred while running: mist -B -I/tools/Xilinx/Vitis/2020.1/cardano/data/cervino/lib -I/tools/Xilinx/Vitis/2020.1/cardano/data/cervino/lib/isg -r +H/tools/Xilinx/Vitis/2020.1/cardano/data/cervino/lib/elongation +Omodo -k64 +Omsbr=100 +Opnll +A +pnopALU +pnopLDA +pnopLDB +pnopSTS +pnopVEC +Ofexm +Onzmem +Onombt +Ochex +Omsmfi +Omslactc=lckLdaRsrc_E1,lckLdbRsrc_E1:2 gemm.kernel-extern_kernel_ me
xchesscc Failed 
The Program xchesscc has encountered an unexpected error exiting ... xchesscc Failed 
/tools/Xilinx/Vitis/2020.1/cardano/bin/xchesscc: line 58: kill: (-717567) - No such process
```
I attached the MLIR and C++ file for your reference. My compilation command is:
```
    ${VITIS_DIR}/cardano/bin/xchesscc -g -p me \
      -P ${VITIS_DIR}/cardano/data/cervino/lib \
      -o kernel.o -c gemm.kernel.cc
```
[gemm.kernel.cc.txt](https://github.com/Xilinx/mlir-aie/files/7997687/gemm.kernel.cc.txt)
[gemm.polyaie.mlir.txt](https://github.com/Xilinx/mlir-aie/files/7997688/gemm.polyaie.mlir.txt)

Any ideas?

Thanks,
Hanchen

david-vc, 2022-02-04:
Hi Hanchen:

I was able to reproduce the error that you are seeing.
This seems to be a bug in the chess tool chain.
I also ran your test case with xchess from Vitis 2021.2 and this completes
successfully.
If you have access to Vitis 2021.2, you could give this a try.
Thanks,

David

pssrawat, 2022-02-04:
@hanchenye,
Your testcase passes with Vitis 2021.2. I am attaching my log below.
I am no longer affiliated to Xilinx, so I am not in a position to reproduce your issue with 2020.1. However, could you try adding '+Wnoodle,+Olbb=200' to your xchesscc command? 

[2021.2-log.txt](https://github.com/Xilinx/mlir-aie/files/8005353/2021.2-log.txt)

closed 2022-05-10 by stephenneuendorffer