Xilinx/mlir-aie #92: opt: unrecognized architecture 'aie' provided.

ywshao, 2022-04-11:
I was trying to build the toolchain. My goal was to generate some executables, but some test cases didn't pass check-aie. The following are some parts of the outputs when executing 'ninja check-aie'.

The summary from 'ninja check-aie' using aiengine:
```
  Unsupported:  14
  Passed     : 118
  Failed     :  25
```
The summary from 'ninja check-aie' using aienginev2:
```
  Unsupported:  14
  Passed     : 127
  Failed     :  25
```
Both aiengine and aienginev2 had the same issue:
```
Command Output (stdout):
--
Error encountered while running: opt -O2 -strip -S acdc_project/core_1_3.ll -o acdc_project/core_1_3.stripped.ll

--
Command Output (stderr):
--
opt: unrecognized architecture 'aie' provided.

--
```
The issue might be caused by a line in acdc_project/core_1_3.ll:
```llvm
target triple = "aie"
```
The result I got was similar to [this](https://github.com/Xilinx/mlir-aie/issues/77#issuecomment-1017596380).

According to the link above, I think the issue may come from building LibXAIE. The building method I tried might be wrong since I built both aiengine and aienginev2 by running Makefile.Linux and Makefile. If this is true, I would like to have documentation about building these things. It may also help others since somebody also faced the building issue in the link above.

Since the file passed to opt has specified aie target, but The LLVM with specified commithash doesn't have aie target, I think the issue may also come from somewhere that isn't in the documentation. Maybe it modifies LLVM to support aie target, or some conversions were done before opt, or something I don't know.

stephenneuendorffer, 2022-04-11:
Thanks... I need to fix this....

stephenneuendorffer, 2022-04-11:
I don't have an easy way to check this at the moment, but I think we need something like:  It would be wonderful if you could check to see if this fixes the problem.

```
diff --git a/tools/aiecc/aiecc/main.py b/tools/aiecc/aiecc/main.py
index 832e4d5..7203cf5 100644
--- a/tools/aiecc/aiecc/main.py
+++ b/tools/aiecc/aiecc/main.py
@@ -90,6 +90,11 @@ def run_flow(opts, tmpdirname):
         do_call(['aie-translate', file_with_addresses, '--aie-generate-ldscript', '--tilecol=%d' % corecol, '--tilerow=%d' % corerow, '-o', file_core_ldscript])
         file_core_llvmir = tmpcorefile(core, "ll")
         do_call(['aie-translate', '--mlir-to-llvmir', file_opt_core, '-o', file_core_llvmir])
+
+        # If we're compiling with chess, use the chess target triple.
+        if(opts.xchesscc):
+          do_call(['sed', '-i', 's/target triple = \".*\"/target triple = \"pdarch-unknown-unknown-elf\"/', file_core_llvmir])
+
         file_core_llvmir_stripped = tmpcorefile(core, "stripped.ll")
         do_call(['opt', '--passes=default<O2>,strip', '-S', file_core_llvmir, '-o', file_core_llvmir_stripped])
         file_core_elf = elf_file if elf_file else corefile(".", core, "elf")
```

ywshao, 2022-04-11:
Thanks for your help.
But unfortunately, it doesn't recognize pdarch either.
```
opt: unrecognized architecture 'pdarch' provided.
```

stephenneuendorffer, 2022-04-11:
What about this?  Sorry for the back and forth.
```
diff --git a/tools/aiecc/aiecc/main.py b/tools/aiecc/aiecc/main.py
index 832e4d5..9f89eac 100644
--- a/tools/aiecc/aiecc/main.py
+++ b/tools/aiecc/aiecc/main.py
@@ -90,13 +90,11 @@ def run_flow(opts, tmpdirname):
         do_call(['aie-translate', file_with_addresses, '--aie-generate-ldscript', '--tilecol=%d' % corecol, '--tilerow=%d' % corerow, '-o', file_core_ldscript
])
         file_core_llvmir = tmpcorefile(core, "ll")
         do_call(['aie-translate', '--mlir-to-llvmir', file_opt_core, '-o', file_core_llvmir])
-        file_core_llvmir_stripped = tmpcorefile(core, "stripped.ll")
-        do_call(['opt', '--passes=default<O2>,strip', '-S', file_core_llvmir, '-o', file_core_llvmir_stripped])
         file_core_elf = elf_file if elf_file else corefile(".", core, "elf")
         file_core_obj = tmpcorefile(core, "o")
         if(opts.xchesscc):
           file_core_llvmir_chesshack = tmpcorefile(core, "chesshack.ll")
-          do_call(['cp', file_core_llvmir_stripped, file_core_llvmir_chesshack])
+          do_call(['cp', file_core_llvmir, file_core_llvmir_chesshack])
           do_call(['sed', '-i', 's/noundef//', file_core_llvmir_chesshack])
           do_call(['sed', '-i', 's/noalias_sidechannel[^,],//', file_core_llvmir_chesshack])
           file_core_llvmir_chesslinked = tmpcorefile(core, "chesslinked.ll")
@@ -114,6 +112,8 @@ def run_flow(opts, tmpdirname):
             do_call(['clang', '-O2', '--target=aie', file_core_obj, me_basic_o, libm,
             '-Wl,-T,'+file_core_ldscript, '-o', file_core_elf])
         else:
+          file_core_llvmir_stripped = tmpcorefile(core, "stripped.ll")
+          do_call(['opt', '--passes=default<O2>,strip', '-S', file_core_llvmir, '-o', file_core_llvmir_stripped])
           do_call(['llc', file_core_llvmir_stripped, '-O2', '--march=aie', '--filetype=obj', '-o', file_core_obj])
           if(opts.xbridge):
             link_with_obj = extract_input_files(file_core_bcf)
```

ywshao, 2022-04-11:
Sorry for the late replies.
My computer is very slow, but it's still my pleasure to participate in this.
Here's the result of one of them:
```
Command Output (stdout):
--
WARNING: XILINX_VITIS environment variable unset, python unavailable
Error encountered while running: xchesscc_wrapper -d -f +P 4 acdc_project/core_1_3.chesslinked.ll kernel.o +l acdc_project/core_1_3.bcf -o ./core_1_3.elf

--
Command Output (stderr):
--
chess-clang: warning: argument unused during compilation: '-I /tools/Xilinx/Vitis/2020.1/cardano/data/cervino/lib' [-Wunused-command-line-argument]
chess-clang: warning: argument unused during compilation: '-I /tools/Xilinx/Vitis/2020.1/cardano/data/cervino/lib/runtime_cxx/libcxx-lite/include' [-Wunused-command-line-argument]
chess-clang: warning: argument unused during compilation: '-I /tools/Xilinx/Vitis/2020.1/cardano/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include-lite' [-Wunused-command-line-argument]
chess-clang: warning: argument unused during compilation: '-I /tools/Xilinx/Vitis/2020.1/cardano/data/cervino/lib/runtime/include' [-Wunused-command-line-argument]
chess-clang: warning: argument unused during compilation: '-include me_llvm.h' [-Wunused-command-line-argument]
warning: overriding the module target triple with pdarch-unknown-unknown-elf [-Woverride-module]
Stack dump:
0.	Program arguments: /tools/Xilinx/Vitis/2020.1/cardano/tps/lnx64/target/bin/LNa64bin/chess-clang -cc1 -triple pdarch-unknown-unknown-elf -S -disable-free -disable-llvm-verifier -discard-value-names -main-file-name core_1_3.chesslinked.ll -mrelocation-model static -mthread-model single -mdisable-fp-elim -fmath-errno -no-integrated-as -mconstructor-aliases --chess-proc-name=me --chess-proc-dir=/tools/Xilinx/Vitis/2020.1/cardano/data/cervino/lib --chess-proc-binary -mllvm -chess-config=/tools/Xilinx/Vitis/2020.1/cardano/data/cervino/lib/isg/me_chess_llvm.cfg -fno-threadsafe-statics -dwarf-column-info -debugger-tuning=gdb -coverage-notes-file /home/ywshao/Desktop/mlir-aie/build/test/unit_tests/chess_compiler_tests/01_precompiled_core_function/chesswork13762/core_1_3.chesslinked.gcno -nostdsysteminc -resource-dir /tools/Xilinx/Vitis/2020.1/cardano/tps/lnx64/target/bin/LNa64bin/../../chessdir/clangdir/9.0.0 -O2 -Werror=return-type -std=c++2a -fno-dwarf-directory-asm -fdebug-compilation-dir /home/ywshao/Desktop/mlir-aie/build/test/unit_tests/chess_compiler_tests/01_precompiled_core_function -ferror-limit 19 -fmessage-length 0 -fno-builtin-memcpy -fno-implicit-modules -fobjc-runtime=gcc -fdiagnostics-show-option -vectorize-loops -vectorize-slp -mllvm -disable-lsr -mllvm -replexitval=never -mllvm -enable-load-pre=false -mllvm -chess-disable-add-to-or -mllvm -chess-combine-gep-indices=none -mllvm -chess-disable-fold-phi-of-loads -mllvm -chess-aainfo2chains-algo=4 -mllvm -chess-aggressive-aainfo=false -mllvm -chess-enable-indvarsimplify=0 -fchess-register-c++-static-destructors-as-global-dtors -o chesswork13762/core_1_3.chesslinked.sfg -x ir acdc_project/core_1_3.chesslinked.ll 
1.	Code generation
2.	Running pass 'Function Pass Manager' on module 'acdc_project/core_1_3.chesslinked.ll'.
3.	Running pass 'PD Emit Ipd Functions - SFG emission' on function '@core13'
/tools/Xilinx/Vitis/2020.1/cardano/tps/lnx64/target/bin/LNa64bin/chess-clang(_ZN4llvm3sys15PrintStackTraceERNS_11raw_ostreamE+0x1a)[0x158b42a]
/tools/Xilinx/Vitis/2020.1/cardano/tps/lnx64/target/bin/LNa64bin/chess-clang(_ZN4llvm3sys17RunSignalHandlersEv+0x35)[0x1589305]
/tools/Xilinx/Vitis/2020.1/cardano/tps/lnx64/target/bin/LNa64bin/chess-clang[0x158941c]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x12890)[0x7f4a69f06890]
/tools/Xilinx/Vitis/2020.1/cardano/tps/lnx64/target/bin/LNa64bin/chess-clang[0xa13d40]
/tools/Xilinx/Vitis/2020.1/cardano/tps/lnx64/target/bin/LNa64bin/chess-clang[0xa37e56]
/tools/Xilinx/Vitis/2020.1/cardano/tps/lnx64/target/bin/LNa64bin/chess-clang[0xa3ca77]
/tools/Xilinx/Vitis/2020.1/cardano/tps/lnx64/target/bin/LNa64bin/chess-clang[0xcfc7c7]
/tools/Xilinx/Vitis/2020.1/cardano/tps/lnx64/target/bin/LNa64bin/chess-clang(_ZN4llvm13FPPassManager13runOnFunctionERNS_8FunctionE+0x344)[0x107b194]
/tools/Xilinx/Vitis/2020.1/cardano/tps/lnx64/target/bin/LNa64bin/chess-clang(_ZN4llvm13FPPassManager11runOnModuleERNS_6ModuleE+0x6b)[0x107b3ab]
/tools/Xilinx/Vitis/2020.1/cardano/tps/lnx64/target/bin/LNa64bin/chess-clang(_ZN4llvm6legacy15PassManagerImpl3runERNS_6ModuleE+0x33b)[0x107bc0b]
/tools/Xilinx/Vitis/2020.1/cardano/tps/lnx64/target/bin/LNa64bin/chess-clang[0x1777757]
/tools/Xilinx/Vitis/2020.1/cardano/tps/lnx64/target/bin/LNa64bin/chess-clang(_ZN5clang17EmitBackendOutputERNS_17DiagnosticsEngineERKNS_19HeaderSearchOptionsERKNS_14CodeGenOptionsERKNS_13TargetOptionsERKNS_11LangOptionsERKN4llvm10DataLayoutEPNSE_6ModuleENS_13BackendActionESt10unique_ptrINSE_17raw_pwrite_streamESt14default_deleteISM_EE+0x459)[0x1778f79]
/tools/Xilinx/Vitis/2020.1/cardano/tps/lnx64/target/bin/LNa64bin/chess-clang(_ZN5clang13CodeGenAction13ExecuteActionEv+0x5db)[0x2096d5b]
/tools/Xilinx/Vitis/2020.1/cardano/tps/lnx64/target/bin/LNa64bin/chess-clang(_ZN5clang14FrontendAction7ExecuteEv+0x191)[0x1bf87a1]
/tools/Xilinx/Vitis/2020.1/cardano/tps/lnx64/target/bin/LNa64bin/chess-clang(_ZN5clang16CompilerInstance13ExecuteActionERNS_14FrontendActionE+0x390)[0x1bbcfe0]
/tools/Xilinx/Vitis/2020.1/cardano/tps/lnx64/target/bin/LNa64bin/chess-clang(_ZN5clang25ExecuteCompilerInvocationEPNS_16CompilerInstanceE+0xa4c)[0x1c98c6c]
[... cut ...]
InstalledDir: /tools/Xilinx/Vitis/2020.1/cardano/tps/lnx64/target/bin/LNa64bin
chess-clang: note: diagnostic msg: PLEASE submit a bug report to mailto:asipsupport@synopsys.com and include the crash backtrace, preprocessed source,  associated run script and used processor configuration (isg/*_chess_llvm.cfg) file.
chess-clang: note: diagnostic msg: Error generating preprocessed source(s) - no preprocessable inputs.
xchesscc Failed 
The Program xchesscc has encountered an unexpected error exiting ... xchesscc Failed 

--
```

gyreflyr, 2022-04-13:
> What about this? Sorry for the back and forth.
>
> ```
> diff --git a/tools/aiecc/aiecc/main.py b/tools/aiecc/aiecc/main.py
> index 832e4d5..9f89eac 100644
> --- a/tools/aiecc/aiecc/main.py
> +++ [...]

ywshao, 2022-04-15:
Thank [stephenneuendorffer](https://github.com/stephenneuendorffer) and [gyreflyr](https://github.com/gyreflyr)!
After switching to Vitis 2021.1 and messing around with compiler and LibXAIE, I finally have the same result as [gyreflyr](https://github.com/Xilinx/mlir-aie/issues/92#issuecomment-1098234707)'s.

While fixing some errors from check-aie, I found I need to compile LibXAIE with aarch64 compiler and change the filename.
Maybe it's worth mentioning the building workflow of LibXAIE in the documentation to help people after me.

commit d79c5e99 (2022-04-19): Update to newer version of LLVM

This requires some changes to the way we call opt and more downgrading
of llvmir in order to target chess.  In addition, fix a bug where
previously we were calling opt even in cases where we don't have peano.

Fixes #92
closed 2022-04-19 by stephenneuendorffer