Xilinx/mlir-aie #100: Enable mlir-aie to aiesimulator flow

ywshao, 2022-05-03:
There is an aiecompiler to aiesimulator flow in Vitis. I would like to know if there is a way to replace aiecompiler with mlir-aie, i.e., having a mlir-aie to aiesimulator flow.

stephenneuendorffer, 2022-05-10:
There's not a direct way to do this today.  Conceptually, the ADF dialect would be the best place to integrate with the aiecompiler tools.  However, this level of abstraction doesn't yet map into the AIE dialect.  Conversely, aiesimulator doesn't readily exist as a separate component that the AIE dialect could interface to.

jackl-xilinx, 2023-01-19:
We are looking into enabling some level of integration from mlir-aie to aiesimulator and hope to have an update in the next few weeks.

jackl-xilinx, 2023-02-13:
This should now be enabled. Please see tutorial-2b for more details

closed 2023-02-13 by jackl-xilinx