Xilinx/mlir-aie #106: build problems?

HadXu, 2022-05-27:
```
-- Found v++:/data/soft/Xilinx/Vitis/2021.2/bin/v++
-- Found xchessmk:/data/soft/Xilinx/Vitis/2021.2/aietools/bin/xchessmk
-- Found libme.a:/data/soft/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/Release/libme.a
-- Found AIE libc.a:/data/soft/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime/lib/Release/libc.a
-- Found AIE libm.a:/data/soft/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime/lib/Release/libm.a
-- Unable to find Vitis include folder
-- Could NOT find Vitis (missing: VITIS_INCLUDE_DIR) 
-- Building with -fPIC
-- Using MLIRConfig.cmake in: /home/lay/work/xilinx/llvm-project/build/lib/cmake/mlir
-- Using LLVMConfig.cmake in: /home/lay/work/xilinx/llvm-project/build/lib/cmake/llvm
-- Found v++:/data/soft/Xilinx/Vitis/2021.2/bin/v++
-- Found xchessmk:/data/soft/Xilinx/Vitis/2021.2/aietools/bin/xchessmk
-- Found libme.a:/data/soft/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/Release/libme.a
-- Found AIE libc.a:/data/soft/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime/lib/Release/libc.a
-- Found AIE libm.a:/data/soft/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime/lib/Release/libm.a
-- Unable to find Vitis include folder
-- Could NOT find Vitis (missing: VITIS_INCLUDE_DIR) 
-- Could NOT find LibXAIE (missing: XILINX_XAIE_INCLUDE_DIR XILINX_XAIE_LIBS) 
-- Building with -fPIC
Translations: MLIRTargetCpp;MLIRSPIRVDeserialization;MLIRSPIRVSerialization;MLIRSPIRVBinaryUtils;MLIRSPIRVTranslateRegistration;MLIRArmNeonToLLVMIRTranslation;MLIRArmSVEToLLVMIRTranslation;MLIRAMXToLLVMIRTranslation;MLIRLLVMToLLVMIRTranslation;MLIRNVVMToLLVMIRTranslation;MLIROpenACCToLLVMIRTranslation;MLIROpenMPToLLVMIRTranslation;MLIRROCDLToLLVMIRTranslation;MLIRX86VectorToLLVMIRTranslation;MLIRTargetLLVMIRExport;MLIRToLLVMIRTranslationRegistration;MLIRTargetLLVMIRImport
-- Configuring done
-- Generating done
```

how to set the Vitis and LibXAIE config, and I had build LibXAIE with embeddedsw repo?

HadXu, 2022-05-30:
I follow your advice and found

cmd 
```
    -DLLVM_DIR=/home/lay/work/xilinx/llvm-project/build/lib/cmake/llvm \
    -DMLIR_DIR=/home/lay/work/xilinx/llvm-project/build/lib/cmake/mlir \
    -DCMAKE_MODULE_PATH=/home/lay/work/xilinx/cmakeModules \
    -DLIBXAIE_DIR=/home/lay/work/xilinx_nqdtan/embeddedsw/XilinxProcessorIPLib/drivers/aienginev2 \
    -DCMAKE_BUILD_TYPE=Debug \
```

and info below
```
-- Unable to find Vitis include folder
-- Could NOT find Vitis (missing: VITIS_INCLUDE_DIR) 
-- Could NOT find LibXAIE (missing: XILINX_XAIE_INCLUDE_DIR XILINX_XAIE_LIBS) 
-- Building with -fPIC
Translations: MLIRTargetCpp;MLIRSPIRVDeserialization;MLIRSPIRVSerialization;MLIRSPIRVBinaryUtils;MLIRSPIRVTranslateRegistration;MLIRArmNeonToLLVMIRTranslation;MLIRArmSVEToLLVMIRTranslation;MLIRAMXToLLVMIRTranslation;MLIRLLVMToLLVMIRTranslation;MLIRNVVMToLLVMIRTranslation;MLIROpenACCToLLVMIRTranslation;MLIROpenMPToLLVMIRTranslation;MLIRROCDLToLLVMIRTranslation;MLIRX86VectorToLLVMIRTranslation;MLIRTargetLLVMIRExport;MLIRToLLVMIRTranslationRegistration;MLIRTargetLLVMIRImport
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    LIBXAIE_DIR
```

according to this info, ```LIBXAIE_DIR``` do not used.

HadXu, 2022-05-30:
I had build it successfully. And I pass test. But I do not know how to run on vck5000?  And I do not find a simple running on vck5000.

HadXu, 2022-05-30:
Lincense need.
```
AIETOOLS=/data/soft/Xilinx/Vitis/2021.2/aietools python aiecc.py  ~/work/xilinx/mlir-aie/test/unit_tests/00_itsalive/aie.mlir
AIEBuild license not found !!No such file or directory
The Program xchesscc has encountered an unexpected error exiting ... AIEBuild license not found !!No such file or directory
Error encountered while running: xchesscc_wrapper -c -d -f +f +P 4 /home/lay/work/xilinx/mlir-aie/build/bin/aiecc/../../runtime_lib/chess_intrinsic_wrapper.cpp -o acdc_project/chess_intrinsic_wrapper.ll
```

gyreflyr, 2022-05-31:
You will need to get 2 licenses from
https://www.xilinx.com/support/licensing_solution_center.html
The AIE compiler license (AIEbuild) and the VCK5000 synthesis license
(XCVC1902.)

HadXu, 2022-06-02:
```
// RUN: source %S/../../../settings.sh
// RUN: aiecc.py --sysroot=%VITIS_SYSROOT% --aie-generate-xaiev2 %s -I%aie_runtime_lib% %aie_runtime_lib%/test_library.cpp %S/test.cpp -o test.elf
// RUN: %run_on_board ./test.elf
```

what does ***sysroot*** mean, and how to set it(on VCK5000)? Thanks.

gyreflyr, 2022-06-02:
For VCK5000 add this to cmake command line:
-DVitisSysroot=/

Adding the -DVitisSysroot=/ to the cmake invocation will have cmake use the X86 host environment for include files instead of the embedded aarch64 environment.  

You will also need to go into tools/aiecc/aiecc/main.py line 144 and change this:
    cmd = ['clang','--target=aarch64-linux-gnu', '-std=c++11']
to this:  
    cmd = ['clang', '-std=c++11'] 
That will build x86_64 executables instead of aarch64.

If your ninja check-aie fails due to missing files, you can patch the include path to match your local environment in the tools/aiecc/aiecc/main.py lines 149-154.  You will notice that line 153 points to /opt/xaiengine/include and 154 includes the /opt/xaiengine/src/libxaiengine.so library.  Either create a symbolic link called /opt/xaiengine that points to your embeddedsw/XilinxProcessorIPLib/drivers/aienginev2 or change main.py to point to it.

HadXu, 2022-06-06:
I followed your advice, cmd
```
AIETOOLS=/data/soft/Xilinx/Vitis/2021.2/aietools python aiecc.py  --sysroot=/ ~/work/xilinx/mlir-aie/test/unit_tests/01_memory_read_write/aie.mlir -I/home/lay/work/xilinx/mlir-aie/runtime_lib /home/lay/work/xilinx/mlir-aie/runtime_lib/test_library.cpp  ~/work/xilinx/mlir-aie/test/unit_tests/01_memory_read_write/test.cpp -o test.elf
```

but found
```
/home/lay/work/xilinx/mlir-aie/runtime_lib/test_library.h:97:3: error: unknown type name 'XAieGbl_Config'; did you mean 'XAie_Config'?
  XAieGbl_Config *AieConfigPtr;
  ^~~~~~~~~~~~~~
  XAie_Config
/opt/xaiengine/include/xaiengine/xaiegbl.h:185:3: note: 'XAie_Config' declared here
} XAie_Config;
  ^
In file included from /home/lay/work/xilinx/mlir-aie/runtime_lib/test_library.cpp:13:
/home/lay/work/xilinx/mlir-aie/runtime_lib/test_library.h:98:3: error: unknown type name 'XAieGbl'
  XAieGbl AieInst;
```

and I found those ***XAieGbl_Config*** type  do  not exist in ***aienginev2***. Thanks.

HadXu, 2022-06-06:
run with 
```
(py38) xxx@ubuntu-MU72-SU0-00:~/work/xilinx/mlir-aie/build/bin$ AIETOOLS=/data/soft/Xilinx/Vitis/2021.2/aietools python aiecc.py  --sysroot=/ --aie-generate-xaiev2 ~/work/xilinx/mlir-aie/test/unit_tests/01_memory_read_write/aie.mlir -I/home/lay/work/xilinx/mlir-aie/runtime_lib /home/lay/work/xilinx/mlir-aie/runtime_lib/test_library.cpp  ~/work/xilinx/mlir-aie/test/unit_tests/01_memory_read_write/test.cpp -o test.elf
Found Vitis at /data/soft/Xilinx/Vitis/2021.2
chess-clang: warning: argument unused during compilation: '-I /data/soft/Xilinx/Vitis/2021.2/aietools/data/cervino/lib' [-Wunused-command-line-argument]
chess-clang: warning: argument unused during compilation: '-I /data/soft/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libcxx-lite/include' [-Wunused-command-line-argument]
chess-clang: warning: argument unused during compilation: '-I /data/soft/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include-lite' [-Wunused-command-line-argument]
chess-clang: warning: argument unused during compilation: '-I /data/soft/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime/include' [-Wunused-command-line-argument]
chess-clang: warning: argument unused during compilation: '-include aie_core.h' [-Wunused-command-line-argument]
warning: overriding the module target triple with pdarch-unknown-unknown-elf [-Woverride-module]
1 warning generated.
```

and generate test.elf, but look test error.
```
....
MW: 0x20000872000, 0x2, 0x0
MW: 0x20000872008, 0x8080, 0x8000
MW: 0x20000872000, 0x1, 0x1
MW: 0x200008b2000, 0x2, 0x0
MW: 0x200008b2008, 0x8080, 0x8000
MW: 0x200008b2000, 0x1, 0x1
R: 0x200008c100c, 0x0
R: 0x200008c100c, 0x0
ERROR After memory writes. Check [3]=14: mlir_aie_read_buffer_a(_xaie, 3) expected 14, but was 0!
R: 0x200008c1014, 0x0
R: 0x200008c1014, 0x0
ERROR After memory writes. Check [3]=14: mlir_aie_read_buffer_a(_xaie, 5) expected 8, but was 0!
R: 0x200008c1024, 0x0
R: 0x200008c1024, 0x0
ERROR After memory writes. Check [3]=14: mlir_aie_read_buffer_a(_xaie, 9) expected 14, but was 0!
Fail!
test done.
```

cabreraam, 2022-06-06:
> _**Update**_ I find run with _**--aie-generate-xaiev2**_

> /home/lay/work/xilinx/mlir-aie/runtime_lib/test_library.h:97:3: error: unknown type name 'XAieGbl_Config'; did you mean 'XAie_Config'?
>   XAieGbl_Config *AieConfigPtr;
>   ^~~~~~~~~~~~~~
>   XAie_Config [...]

If you pass `--aie-generate-xaiev2` to `aiecc.py`, it should trigger [this condition](https://github.com/Xilinx/mlir-aie/blob/ecbaa1d4db4d1a3ba44e0edbb0c55fce44d75ad5/tools/aiecc/aiecc/cl_arguments.py#L69) in the `parse_args` function in `cl_arguments.py`. When that is triggered, `xaie` gets set to `2`. In `aiecc.py`, `xaie == 2` should trigger [this condition](https://github.com/Xilinx/mlir-aie/blob/ecbaa1d4db4d1a3ba44e0edbb0c55fce44d75ad5/tools/aiecc/aiecc/main.py#L148), which will define `-DLIBXAIENGINEV2` during compilation. In `test_library.h`, `-DLIBXAIENGINEV2` being defined results in [this code block](https://github.com/Xilinx/mlir-aie/blob/ecbaa1d4db4d1a3ba44e0edbb0c55fce44d75ad5/runtime_lib/test_library.h#L35) being active (lines 35-80). That code block **does not use `XAieGbl_Config`.** So, if you're passing `--aie-generate-xaiev2` correctly, that error message should not come up.

I hope this helps!

HadXu, 2022-06-06:
Yes, you are right. Thanks.  But according this [link](https://github.com/Xilinx/mlir-aie/issues/106#issuecomment-1146982467), unit test do not pass. And I do not how to debug.

stephenneuendorffer, 2022-06-06:
@jackl-xilinx Can you look into debugging, please?

HadXu, 2022-06-08:
@jackl-xilinx please?

HadXu, 2022-07-01:
from this [link](https://github.com/Xilinx/mlir-aie/blob/main/test/unit_tests/01_memory_read_write/aie.mlir#L11)
where can I get ```valid_xchess_license```?

HadXu, 2022-07-13:
> For VCK5000 add this to cmake command line: -DVitisSysroot=/
>
> Adding the -DVitisSysroot=/ to the cmake invocation will have cmake use the X86 host environment for include files instead of the [...]

I do not how to solve this problems according this link [link](https://github.com/Xilinx/mlir-aie/issues/124#issuecomment-1181227514), Thank you!

gyreflyr, 2022-07-13:
I am not able to speak to how the mlir-aie team is approaching pcie host
support for the vck5000.  I was just attempting to affirm that after a few
modifications, it was possible to run the test suite on the vck5000
hardware.

Sharing library paths using the LD_LIBRARY_PATH environment variables might
be helpful.

export LD_LIBRARY_PATH=/opt/xaiengine/src:/opt/xilinx/xrt/lib

HadXu, 2022-07-14:
> I am not able to speak to how the mlir-aie team is approaching pcie host support for the vck5000. I was just attempting to affirm that after a few modifications, it was possible to run the test [...]

You are right. Thanks.

closed 2022-07-14 by HadXu