Xilinx/mlir-aie #108: Using --aie-generate-xaiev2 to run unit tests with xaiev2

cabreraam, 2022-06-02:
Hi all, 

I am still very new to this project and still get very lost, but reading through some issues (#106 #77) and @nqdtan's [fork](https://github.com/Xilinx/mlir-aie/commit/14c00ed3234ed7698c6720cb457cc5491d2d9999) of this project, I have updated the RUN: scripts in the unit tests to pass the `--aie-generate-xaiev2--` option to `aiecc.py`. To do this, I navigate to the [this directory](https://github.com/Xilinx/mlir-aie/tree/main/test/unit_tests) and issue the commands:

```
find ./ -name aie.mlir | xargs sed -i 's/--sysroot=%VITIS_SYSROOT%/--sysroot=%VITIS_SYSROOT% --aie-generate-xaiev2/g'
find ./ -name aie_row.mlir | xargs sed -i 's/--sysroot=%VITIS_SYSROOT%/--sysroot=%VITIS_SYSROOT% --aie-generate-xaiev2/g'
```
I probably could have just cherry-picked from @nqdtan's commits, but this seemed easier since I didn't need the hard-coded include and library paths.

With this, when I issue `ninja check-aie`, I go from this:

```
  Unsupported:  15                                                                                                                                                                                                  
  Passed     : 130                                                                                                                                                                                                  
  Unresolved :   2                                                                                                                                                                                                  
  Failed     :  23   
```
to this:
```
  Unsupported:  15
  Passed     : 153
  Unresolved :   2
```

I don't understand the AIE versions or all of the available options you can pass to `aiecc.py`, but I just wanted to share this in case it was helpful to anyone else trying to follow the instructions from [here](https://xilinx.github.io/mlir-aie/Building.html).

stephenneuendorffer, 2022-07-07:
@jackl-xilinx Is there still something here that needs to be done?

stephenneuendorffer, 2022-11-03:
I think this obsolete

closed 2022-11-03 by stephenneuendorffer