Xilinx/mlir-aie #135: On VCK5000, when I run unit_tests/01_memory_read_write unit test, find these errors

HadXu, 2022-07-14:
When I run unit_tests/01_memory_read_write unit test, find these errors
```bash
>>> python aiecc.py --sysroot=/ --aie-generate-xaiev2 ~/work/new-aie/mlir-aie/test/unit_tests/01_memory_read_write/aie.mlir -I/home/lay/work/new-aie/mlir-aie/runtime_lib /home/lay/work/new-aie/mlir-aie/runtime_lib/test_library.cpp ~/work/new-aie/mlir-aie/test/unit_tests/01_memory_read_write/test.cpp -o test.elf
>>> ./test.elf
....
MW: 0x20000072000, 0x2, 0x0
MW: 0x20000072008, 0x8080, 0x8000
MW: 0x20000072000, 0x1, 0x1
MW: 0x200000b2000, 0x2, 0x0
MW: 0x200000b2008, 0x8080, 0x8000
MW: 0x200000b2000, 0x1, 0x1
MW: 0x200000f2000, 0x2, 0x0
MW: 0x200000f2008, 0x8080, 0x8000
MW: 0x200000f2000, 0x1, 0x1
MW: 0x20000872000, 0x2, 0x0
MW: 0x20000872008, 0x8080, 0x8000
MW: 0x20000872000, 0x1, 0x1
MW: 0x200008b2000, 0x2, 0x0
MW: 0x200008b2008, 0x8080, 0x8000
MW: 0x200008b2000, 0x1, 0x1
R: 0x200008c100c, 0x0
R: 0x200008c100c, 0x0
ERROR After memory writes. Check [3]=14: mlir_aie_read_buffer_a(_xaie, 3) expected 14, but was 0!
R: 0x200008c1014, 0x0
R: 0x200008c1014, 0x0
ERROR After memory writes. Check [3]=14: mlir_aie_read_buffer_a(_xaie, 5) expected 8, but was 0!
R: 0x200008c1024, 0x0
R: 0x200008c1024, 0x0
ERROR After memory writes. Check [3]=14: mlir_aie_read_buffer_a(_xaie, 9) expected 14, but was 0!
Fail!
test done.
```
How to debug this problem?

HadXu, 2022-07-18:
Can mlir-aie run on VCK5000 through x86 host?

HadXu, 2022-07-19:
Unlike the ARM-based platform (e.g., VCK190), the AIE address space is not visible to the x86 host.

jgmelber, 2022-07-19:
What platform design are you using for VCK5000 and MLIR-AIE?

closed 2025-04-29 by jgmelber