Xilinx/mlir-aie #140: Linking external object file compiled

cabreraam, 2022-07-26:
Hi all,

I am trying to compile an example I was given where a kernel function is defined. It looks something like this:

`kernel_code.cc`:
```c
void func(int32_t *a, int32_t *b)
{
...
}
```

The `mlir` code references the function like this:
`aie.mlir`
```llvm
call @func(%buf13_0, %buf13_1) : (memref<256xi32>, memref<256xi32>) -> ()
```

I am able to use `xchesscc`  to compile `kernel_code.cc`, but when I use the `aiecc.py` driver, I get the following error message:

```bash
aie.mlir:25:3: error: custom op 'func' is unknown (tried 'builtin.func' as well)
  func private @func(%A: memref<256xi32>, %B: memref<256xi32>) -> ()
  ^
Error encountered while running: aie-opt --lower-affine --aie-assign-buffer-addresses -convert-scf-to-cf aie.mlir -o acdc_project/input_with_addresses.mlir
```

__This example works for me in a containerized environment where everything was pre-built for me__. However, with upstream `mlir-aie`, I get the above error message.

I took a look at the driver in the container, and I saw this piece of `python`:
```python
do_call(['aie-opt', '--lower-affine', '--aie-assign-buffer-addresses', '-convert-scf-to-std', opts.filename, '-o', file_with_addresses])
```

In the upstream `mlir-aie` repo, I see:
```python
do_call(['aie-opt', '--lower-affine', '--aie-assign-buffer-addresses', '-convert-scf-to-cf', opts.filename, '-o', file_with_addresses])
```

The difference is the between the flags `-convert-scf-to-std` and `-convert-scf-to-cf`. I'm not sure if this is the issue, but this is at least one difference that I have observed.

For extra context, I am able to generate a `.o` file with `xchesscc`. 

Is anyone else experiencing similar issues? #87 is kind of similar, but it's not a type issue; it just looks like `aie-opt` doesn't know that `func` exists, even though the object file it's defined in does exist.

Any help is appreciated!

Thanks,
Anthony

cabreraam, 2022-07-27:
@jgmelber let me know that the example that I was referring to is actually already in the test suite and is located [here](https://github.com/Xilinx/mlir-aie/tree/main/test/unit_tests/chess_compiler_tests/01_precompiled_core_function). So, I took a diff of all of the files that I had, and the version in `01_precompiled_core_function`. The difference was in `aie.mlir`. The diff is below:

```llvm
$ diff aie.mlir (old version) aie.mlir (current version from the above link)
13c13
< // RUN: aiecc.py --sysroot=%VITIS_SYSROOT% %s -I%S/../../../../runtime_lib/ %S/../../../../runtime_lib/test_library.cpp %S/test.cpp -o test.elf
---
> // RUN: aiecc.py --xbridge --sysroot=%VITIS_SYSROOT% --aie-generate-xaiev2 %s -I%S/../../../../runtime_lib/ %S/../../../../runtime_lib/test_library.cpp %S/test.cpp -o test.elf
25c25
<   func private @func(%A: memref<256xi32>, %B: memref<256xi32>) -> ()
---
>   func.func private @func(%A: memref<256xi32>, %B: memref<256xi32>) -> ()
30c30
<     call @func(%buf13_0, %buf13_1) : (memref<256xi32>, memref<256xi32>) -> ()
---
>     func.call @func(%buf13_0, %buf13_1) : (memref<256xi32>, memref<256xi32>) -> ()
```

When I use the newest version of this `aie.mlir` file, everything compiles as it should. I think this issue could be closed for now.

jgmelber, 2022-07-27:
Glad this worked!

closed 2022-07-27 by jgmelber