Xilinx/mlir-aie #151: Develop method to detect loops in routing

denolf, 2022-08-05:
When detecting routing errors, there are cases where loops might be present. Develop a method for detecting these loops along the routing path.

stephenneuendorffer, 2023-08-28:
@andrej Is your work here finished?

andrej, 2023-08-29:
If we decide to merge in #527 this issue could probably be closed. Not sure what people's thoughts on 527 are though.

AndraBisca, 2025-03-25:
Ported to [#842](https://github.com/Xilinx/mlir-aie/pull/842) which has been closed. Was this ever integrated? @stephenneuendorffer @makslevental

hunhoffe, 2026-10-26:
This is now handled: AIECreatePathFindFlows.cpp runs a hold-cycle search over the planned routes and reports the exact cycle when one blocks routing (see 78ac5cf3 "Report the shared-receiver cycle when it alone blocks routing", part of the broader router-reliability work in #3919). The two PRs referenced above (#527, #842) are superseded by this. Closing as resolved.

closed 2026-10-26 by hunhoffe