Xilinx/mlir-aie #152: Reference design `MM_2x2` requires `cardano.h`

cabreraam, 2022-08-12:
The new reference design `MM_2x2` (#147) requires [`cardano.h`](https://github.com/Xilinx/mlir-aie/blob/5f4970c6ffdeb78bda24106d728f447b2cdaeea3/test/reference_designs/MM_2x2/kernel.cc#L8). Based on some digging, it seems like `cardano` is something that doesn't exist in Vitis 2021.x versions, but they exist in Vitis 2020.x versions. (I am using Vitis 2021.2). I took a peak at `cardano.h` in Vitis 2020.2, and in the file, there is a note that says:

>  #warning "cardano.h and cardano namespace are deprecated. Use adf.h and adf namespace instead."

I do have `adf.h` in my Vitis 2021.2 install, and I tried to just replace `#include cardano.h` with `#include adf.h`. However, I get the following error:

```bash
$ xchesscc -I/tools/Xilinx/Vitis/2021.2/aietools/include -p me -P /tools/Xilinx/Vitis/2021.2/aietools/data/cervino/lib -c kernel.cc                                    
In file included from /tools/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include-lite/type_traits:10,                    
                 from /tools/Xilinx/Vitis/2021.2/aietools/include/adf/window/window.h:6,                                                                                                                            
                 from /tools/Xilinx/Vitis/2021.2/aietools/include/adf.h:8,                                                                                                                                          
                 from kernel.cc:8:                                                                                                                                                                                  
/tools/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include/type_traits:850:18: error: missing binary operator before '(' 
/tools/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include/type_traits:871:18: error: missing binary operator before '(' 
/tools/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include/type_traits:982:18: error: missing binary operator before '(' 
/tools/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include/type_traits:1456:18: error: missing binary operator before '('
/tools/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include/type_traits:1565:18: error: missing binary operator before '('
/tools/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include/type_traits:1602:18: error: missing binary operator before '('
/tools/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include/type_traits:1628:18: error: missing binary operator before '('
/tools/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include/type_traits:2382:18: error: missing binary operator before '('
/tools/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include/type_traits:2701:52: error: missing binary operator before '('
/tools/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include/type_traits:2817:18: error: missing binary operator before '('                       
/tools/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include/type_traits:2875:18: error: missing binary operator before '('                                                                                                     
/tools/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include/type_traits:2893:18: error: missing binary operator before '('                                                                                                     
/tools/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include/type_traits:2966:18: error: missing binary operator before '('                                                                                                     
/tools/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include/type_traits:3031:18: error: missing binary operator before '('                                                                                                     
/tools/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include/type_traits:3036:20: error: missing binary operator before '('                                                                                                     
/tools/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include/type_traits:3063:18: error: missing binary operator before '('
/tools/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include/type_traits:3153:18: error: missing binary operator before '('
/tools/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include/type_traits:3279:18: error: missing binary operator before '('
/tools/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include/type_traits:3315:18: error: missing binary operator before '('
/tools/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include/type_traits:3331:18: error: missing binary operator before '('                                                                                                     
/tools/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include/type_traits:3349:18: error: missing binary operator before '('                                                                                                     
Error in "/tools/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include/__nullptr", line 64, column 30: old function argument style, not supported                                                                               
... symbol being parsed: nullptr_t                                                                        
In file included from /tools/Xilinx/Vitis/2021.2/aietools/include/adf/buffer_port/buffer_port_sizes.h:13,                      
                 from /tools/Xilinx/Vitis/2021.2/aietools/include/adf/buffer_port/buffer_port.h:14,                                                                                                                 
                 from /tools/Xilinx/Vitis/2021.2/aietools/include/adf.h:10,                                                                                                                                         
                 from kernel.cc:8:                                                                                                                                                                                  
/tools/Xilinx/Vitis/2021.2/aietools/include/adf/buffer_port/buffer_port_dims.h:10:9: warning: #pragma once is obsolete                                                                                                                                        
In file included from /tools/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include-lite/__tuple:10,                                                                                                                             
                 from /tools/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include/utility:207,                                                                                                                                 
                 from /tools/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include-lite/utility:10,                                                                                                                             
                 from /tools/Xilinx/Vitis/2021.2/aietools/include/adf/buffer_port/buffer_port_dims.h:14,                       
                 from /tools/Xilinx/Vitis/2021.2/aietools/include/adf/buffer_port/buffer_port_sizes.h:13,                                                                                                           
                 from /tools/Xilinx/Vitis/2021.2/aietools/include/adf/buffer_port/buffer_port.h:14,                            
                 from /tools/Xilinx/Vitis/2021.2/aietools/include/adf.h:10,          
                 from kernel.cc:8:                                                                        
[... cut ...]
/tools/Xilinx/Vitis/2021.2/aietools/include/adf/buffer_port/buffer_port_macros.h:11:9: warning: #pragma once is obsolete                                                                                                                                      
xchesscc Failed                                                
The Program xchesscc has encountered an unexpected error exiting ... xchesscc Failed                                           
/tools/Xilinx/Vitis/2021.2/aietools/bin/xchesscc: line 58: kill: (-1656029) - No such process                                  
make: *** [Makefile:14: kernel.o] Error 255                    
```

Could someone help debug this, please?

jackl-xilinx, 2022-08-12:
Thanks Anthony for finding this issue. adf constructs like windows are not necessary for kernels using the mlir-aie framework so we'll be removing them from this kernel design. It'll be assigned to @JinmingZhuang to make the edits and push the fix.

jackl-xilinx, 2022-08-18:
Thanks for being patient. @JinmingZhuang has kindly added the fix to this so you should now be able to compile without the adf.h requirements. Give it a try and let me know if you still run into problems. We'll moving these designs to libxaiev2 drivers so board level testing will still require some updates but you should be able to compile this design with AIE compile and mlir-aie tools.

closed 2022-08-18 by jackl-xilinx