Xilinx/mlir-aie #162: Canonicalizer merges multiple DMA blocks into one

dotkrnl, 2022-09-08:
When two DMA blocks are present, the canonicalizer may merge them into one, for example:

```mlir
 ^bb5:  // 2 preds: ^bb4, ^bb5
    "AIE.useLock"(%4) {action = 0 : i32, value = 1 : i32} : (index) -> ()
    "AIE.dmaBdPacket"() {packet_id = 4 : i32, packet_type = 4 : i32} : () -> ()
    "AIE.dmaBd"(%32) {AB = 0 : i32, len = 1024 : i32, offset = 0 : i32} : (memref<1024xi32>) -> ()
    "AIE.useLock"(%4) {action = 1 : i32, value = 0 : i32} : (index) -> ()
    "AIE.useLock"(%3) {action = 0 : i32, value = 1 : i32} : (index) -> ()
    "AIE.dmaBdPacket"() {packet_id = 5 : i32, packet_type = 5 : i32} : () -> ()
    "AIE.dmaBd"(%33) {AB = 0 : i32, len = 1024 : i32, offset = 0 : i32} : (memref<1024xi32>) -> ()
    "AIE.useLock"(%3) {action = 1 : i32, value = 0 : i32} : (index) -> ()
    "cf.br"()[^bb5] : () -> ()
```

This triggers the error in #114 .  `AIETargetXAIE` seems currently depend on blocks so simply reverting #114 does not solve the issue.

fifield, 2022-09-16:
I've hit this in the past. I think the solution is to introduce a custom terminator instead of using `cf.br`

fifield, 2023-03-06:
This should be fixed by https://github.com/Xilinx/mlir-aie/pull/247

closed 2023-03-06 by fifield