Xilinx/mlir-aie #167: About vck190 boot from sd card?

HadXu, 2022-09-19:
I build aie_platform successfully. And I copy mlir-aie/platforms/vck190_bare_prod/aie_platform/sd_dir files to sd card of fat32 format. But  do not boot successfully. Do I left something or ?

[image]

jgmelber, 2022-09-19:
Did you copy the files to the SD card or flash the .img file onto the SD card? The proper procedure is to flash the .img file

HadXu, 2022-09-20:
I copy mlir-aie/platforms/vck190_bare_prod/aie_platform/sd_dir files to sd card of fat32 format, But I failed. However, I flash the sd_card.img file to sd card, and it boot successfully. Thanks. I think the full documentation should be detailed.

closed 2022-09-20 by HadXu