Xilinx/mlir-aie #179: lowing onnx mlir to mlir-aie problem?

HadXu, 2022-10-09:
mnist.mlir file in https://gist.github.com/HadXu/8987a2bca3e2a6400cc8c533b6d132c2

And I want to this file can run on vck190 with mlir-aie, but with build error
```
Found Vitis at /data/Xilinx/Vitis/2021.2
chess-clang: warning: argument unused during compilation: '-I /data/Xilinx/Vitis/2021.2/aietools/data/cervino/lib' [-Wunused-command-line-argument]
chess-clang: warning: argument unused during compilation: '-I /data/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libcxx-lite/include' [-Wunused-command-line-argument]
chess-clang: warning: argument unused during compilation: '-I /data/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include-lite' [-Wunused-command-line-argument]
chess-clang: warning: argument unused during compilation: '-I /data/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime/include' [-Wunused-command-line-argument]
chess-clang: warning: argument unused during compilation: '-include aie_core.h' [-Wunused-command-line-argument]
warning: overriding the module target triple with pdarch-unknown-unknown-elf [-Woverride-module]
1 warning generated.
Warning in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 9269, column 13: loop count of type `unsigned long long' is converted to `w32' via `int', the first stage of which is non-nil
Warning in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 19963, column 14: loop count of type `unsigned long long' is converted to `w32' via `int', the first stage of which is non-nil
Warning in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 9219, column 13: (approximate): uninitialized variable `__tmp' is used (unconditionally) by `r_update_dint_low (internal)'   (sig .67061 used by opn <90477> in arg 1)
Warning in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 9339, column 13: (approximate): uninitialized variable `__tmp' is used (unconditionally) by `r_update_dint_low (internal)'   (sig .67210 used by opn <90668> in arg 1)
Warning in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 15: (approximate): uninitialized variable `__tmp' is used (unconditionally) by `r_update_dint_low (internal)'   (sig .67255 used by opn <90721> in arg 1)
Warning in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 19913, column 14: (approximate): uninitialized variable `__tmp' is used (unconditionally) by `r_update_dint_low (internal)'   (sig .67585 used by opn <91129> in arg 1)
Warning in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 20034, column 14: (approximate): uninitialized variable `__tmp' is used (unconditionally) by `r_update_dint_low (internal)'   (sig .67734 used by opn <91320> in arg 1)
Warning in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 20082, column 14: (approximate): uninitialized variable `__tmp' is used (unconditionally) by `r_update_dint_low (internal)'   (sig .67779 used by opn <91373> in arg 1)
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 9216, column 13: undefined operation `operator void *(unsigned long long)' <50701> remaining after optimisation, used by:
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 15: (approximate): ... 2nd argument of `entry (internal)' <75486>
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 9339, column 13: undefined operation `operator void *(unsigned long long)' <50833> remaining after optimisation, used by:
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 15: (approximate): ... 2nd argument of `entry (internal)' <75562>
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 9447, column 13: ... 1st argument of `void *add(void *, amod)' <91039>
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 9447, column 13: ... 1st argument of `void *add(void *, amod)' <91080>
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 9447, column 13: ... 1st argument of `void *add(void *, amod)' <93427>
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 9387, column 13: undefined operation `operator void *(unsigned long long)' <50878> remaining after optimisation, used by:
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 15: (approximate): ... 2nd argument of `entry (internal)' <75661>
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 19952, column 5: (approximate): ... 1st argument of `void *add(void *, amod)' <92161>
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 19912, column 14: undefined operation `operator void *(unsigned long long)' <51060> remaining after optimisation, used by:
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 15: (approximate): ... 2nd argument of `entry (internal)' <75874>
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 20034, column 14: undefined operation `operator void *(unsigned long long)' <51174> remaining after optimisation, used by:
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 15: (approximate): ... 2nd argument of `entry (internal)' <76081>
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 20082, column 14: undefined operation `operator void *(unsigned long long)' <51414> remaining after optimisation, used by:
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 15: (approximate): ... 2nd argument of `entry (internal)' <76823>
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 20161, column 14: ... 1st argument of `void *add(void *, amod)' <91405>
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 20161, column 14: ... 1st argument of `void *add(void *, amod)' <91412>
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 20161, column 14: ... 1st argument of `void *add(void *, amod)' <91419>
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 20161, column 14: ... 1st argument of `void *add(void *, amod)' <91426>
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 20161, column 14: ... 1st argument of `void *add(void *, amod)' <91433>
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 20161, column 14: ... 1st argument of `void *add(void *, amod)' <91440>
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 20161, column 14: ... 1st argument of `void *add(void *, amod)' <91447>
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 20161, column 14: ... 1st argument of `void *add(void *, amod)' <91454>
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 20161, column 14: ... 1st argument of `void *add(void *, amod)' <91461>
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 20161, column 14: ... 1st argument of `void *add(void *, amod)' <91468>
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 20161, column 14: ... 1st argument of `void *add(void *, amod)' <91475>
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 20161, column 14: ... 1st argument of `void *add(void *, amod)' <91482>
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 20161, column 14: ... 1st argument of `void *add(void *, amod)' <91489>
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 20161, column 14: ... 1st argument of `void *add(void *, amod)' <91496>
Error in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 20161, column 14: ... 1st argument of `void *add(void *, amod)' <91503>
Error: 
... conversions to built-in types (even when not represented) may be due to a unique, but inappropriate, C++ matching, e.g.,
... t @ i ...becoming... t @ (T)(double)i ...
[... cut ...] =+Sinl,+Olbb=200,+Opmsa,+Olzyinl +Q3=+Sinl,+Olbb=1000,+Opmsa,+Olzyinl +Qfast=+Sinl,+Olbb=1000,+Opmsa,+Olzyinl,+Opfp +Qs=+Sinl,+Olbb=200,+Opmsa,+Olzyinl +Qz=+Sinl,+Olbb=200,+Opmsa,+Olzyinl me
xchesscc Failed No such device
The Program xchesscc has encountered an unexpected error exiting ... xchesscc Failed No such device
Error encountered while running: xchesscc_wrapper -d -f +P 4 acdc_project/core_1_3.chesslinked.ll  +l acdc_project/core_1_3.bcf -o ./core_1_3.elf
```

how to debug this problems?

stephenneuendorffer, 2022-10-11:
It looks like there's a toolflow bug and chess is trying to compile an MLIR file.  Are you running aiecc.py?  If so, what arguments are you passing?

HadXu, 2022-10-12:
Yes. I use aiecc.py, and full command is 

```bash
python /home/lay/work/vck190/mlir-aie/build/bin/aiecc.py -v \
    --sysroot=/home/lay/work/vck190/mlir-aie/platforms/vck190_bare_prod/aie_platform/sw_comp/sysroots/cortexa72-cortexa53-xilinx-linux \
    --aie-generate-xaiev2 \
    aie.mlir \
    -I/home/lay/work/vck190/mlir-aie/runtime_lib \
    /home/lay/work/vck190/mlir-aie/runtime_lib/test_library.cpp \
    main.cpp \
    -o test.elf
```

full log with -v is

```
Found Vitis at /data/Xilinx/Vitis/2021.2

compiling aie.mlir
created temporary directory acdc_project
aie-opt --lower-affine --aie-register-objectFifos --aie-objectFifo-stateful-transform --aie-lower-broadcast-packet --aie-create-packet-flows --aie-assign-buffer-addresses --convert-scf-to-cf aie.mlir -o acdc_project/input_with_addresses.mlir
aie-translate --aie-generate-corelist acdc_project/input_with_addresses.mlir
xchesscc_wrapper -c -d -f +f +P 4 /home/lay/work/vck190/mlir-aie/build/bin/aiecc/../../runtime_lib/chess_intrinsic_wrapper.cpp -o acdc_project/chess_intrinsic_wrapper.ll
sed -i s/^target.*// acdc_project/chess_intrinsic_wrapper.ll
sed -i s/noalias_sidechannel[^,]*,// acdc_project/chess_intrinsic_wrapper.ll
sed -i s/nocallback[^,]*,// acdc_project/chess_intrinsic_wrapper.ll
aie-opt --aie-localize-locks --aie-standard-lowering=tilecol=1 tilerow=3 acdc_project/input_with_addresses.mlir -o acdc_project/core_1_3.mlir
aie-opt --aie-normalize-address-spaces --canonicalize --cse --convert-vector-to-llvm --convert-memref-to-llvm --convert-func-to-llvm=use-bare-ptr-memref-call-conv --convert-cf-to-llvm --canonicalize --cse acdc_project/core_1_3.mlir -o acdc_project/core_1_3.opt.mlir
aie-translate acdc_project/input_with_addresses.mlir --aie-generate-bcf --tilecol=1 --tilerow=3 -o acdc_project/core_1_3.bcf
aie-translate acdc_project/input_with_addresses.mlir --aie-generate-ldscript --tilecol=1 --tilerow=3 -o acdc_project/core_1_3.ld.script
aie-translate --mlir-to-llvmir --opaque-pointers=0 acdc_project/core_1_3.opt.mlir -o acdc_project/core_1_3.ll
cp acdc_project/core_1_3.ll acdc_project/core_1_3.chesshack.ll
sed -i s/noundef// acdc_project/core_1_3.chesshack.ll
sed -i s/noalias_sidechannel[^,],// acdc_project/core_1_3.chesshack.ll
llvm-link acdc_project/core_1_3.chesshack.ll acdc_project/chess_intrinsic_wrapper.ll -S -o acdc_project/core_1_3.chesslinked.ll
sed -i s/noundef// acdc_project/core_1_3.chesslinked.ll
sed -i -E /define .*@/ s/%[0-9]*//g acdc_project/core_1_3.chesslinked.ll
sed -i -E s/mustprogress//g acdc_project/core_1_3.chesslinked.ll
sed -i -E s/poison/undef/g acdc_project/core_1_3.chesslinked.ll
sed -i -E s/nocallback//g acdc_project/core_1_3.chesslinked.ll
awk /_include _file/ {print($3)} acdc_project/core_1_3.bcf
xchesscc_wrapper -d -f +P 4 acdc_project/core_1_3.chesslinked.ll  +l acdc_project/core_1_3.bcf -o ./core_1_3.elf
chess-clang: warning: argument unused during compilation: '-I /data/Xilinx/Vitis/2021.2/aietools/data/cervino/lib' [-Wunused-command-line-argument]
chess-clang: warning: argument unused during compilation: '-I /data/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libcxx-lite/include' [-Wunused-command-line-argument]
chess-clang: warning: argument unused during compilation: '-I /data/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include-lite' [-Wunused-command-line-argument]
chess-clang: warning: argument unused during compilation: '-I /data/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime/include' [-Wunused-command-line-argument]
chess-clang: warning: argument unused during compilation: '-include aie_core.h' [-Wunused-command-line-argument]
warning: overriding the module target triple with pdarch-unknown-unknown-elf [-Woverride-module]
1 warning generated.
Warning in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 9269, column 13: loop count of type `unsigned long long' is converted to `w32' via `int', the first stage of which is non-nil
Warning in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 19963, column 14: loop count of type `unsigned long long' is converted to `w32' via `int', the first stage of which is non-nil
Warning in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 9219, column 13: (approximate): uninitialized variable `__tmp' is used (unconditionally) by `r_update_dint_low (internal)'   (sig .67061 used by opn <90477> in arg 1)
Warning in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 9339, column 13: (approximate): uninitialized variable `__tmp' is used (unconditionally) by `r_update_dint_low (internal)'   (sig .67210 used by opn <90668> in arg 1)
Warning in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 15: (approximate): uninitialized variable `__tmp' is used (unconditionally) by `r_update_dint_low (internal)'   (sig .67255 used by opn <90721> in arg 1)
Warning in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 19913, column 14: (approximate): uninitialized variable `__tmp' is used (unconditionally) by `r_update_dint_low (internal)'   (sig .67585 used by opn <91129> in arg 1)
Warning in "/home/lay/work/aie-mlir-tutorial/demo/acdc_project/core_1_3.opt.mlir", line 20034, column 14: (approximate): uninitialized variable `__tmp' is used (unconditionally) by `r_update_dint_low (internal)'   (sig .67734 used by opn <91320> in arg 1)
[... cut ...] +Olbb=200,+Opmsa,+Olzyinl +Q2=+Sinl,+Olbb=200,+Opmsa,+Olzyinl +Q3=+Sinl,+Olbb=1000,+Opmsa,+Olzyinl +Qfast=+Sinl,+Olbb=1000,+Opmsa,+Olzyinl,+Opfp +Qs=+Sinl,+Olbb=200,+Opmsa,+Olzyinl +Qz=+Sinl,+Olbb=200,+Opmsa,+Olzyinl me
xchesscc Failed No such device
The Program xchesscc has encountered an unexpected error exiting ... xchesscc Failed No such device
Error encountered while running: xchesscc_wrapper -d -f +P 4 acdc_project/core_1_3.chesslinked.ll  +l acdc_project/core_1_3.bcf -o ./core_1_3.elf
```

HadXu, 2022-10-13:
@stephenneuendorffer

HadXu, 2022-10-19:
@stephenneuendorffer any progress？ thanks.

stephenneuendorffer, 2022-11-03:
I had a chance to come back to this.  It looks like the input is simply too large for the single core compiler to handle. 
One fundamental problem I see is that it looks like you've got all the weights embedded in the source code, which is usually not a good idea.  I'd suggest externalizing the weights and loading them into the memory of the processor as a first step.

stephenneuendorffer, 2022-11-03:
--convert-memref-to-llvm implements a lowering for memref.alloc(alignment=16) which results in a code sequence like this:
`  %181 = llvm.ptrtoint %180 : !llvm.ptr<f32> to i64
  %182 = llvm.mlir.constant(1 : index) : i64
  %183 = llvm.sub %177, %182  : i64
  %184 = llvm.add %181, %183  : i64
  %185 = llvm.urem %184, %177  : i64
  %186 = llvm.sub %184, %185  : i64
  %187 = llvm.inttoptr %186 : i64 to !llvm.ptr<f32>`

This code sequence takes a regular pointer and converts it to an aligned pointer, but this code sequence breaks the chess compiler.  I suspect this is because there is pointer-specific handling in the frontend that eliminates int-to-pointer conversions.

One solution would be to statically allocate your buffers using AIE.buffer

HadXu, 2022-11-04:
Yes. So I change  all the weights in to hex string. The new .mlir file in [this](https://gist.github.com/HadXu/de885f77ca643fbb196a064874e000da)  , but same error. 

```
Found Vitis at /data/Xilinx/Vitis/2021.2

compiling aie-mnist-2.mlir
created temporary directory acdc_project
aie-opt --lower-affine --aie-register-objectFifos --aie-objectFifo-stateful-transform --aie-lower-broadcast-packet --aie-create-packet-flows --aie-assign-buffer-addresses -convert-scf-to-cf aie-mnist-2.mlir -o acdc_project/input_with_addresses.mlir
aie-translate --aie-generate-corelist acdc_project/input_with_addresses.mlir
xchesscc_wrapper -c -d -f +f +P 4 /home/lay/work/vck190/mlir-aie/build/bin/aiecc/../../runtime_lib/chess_intrinsic_wrapper.cpp -o acdc_project/chess_intrinsic_wrapper.ll
sed -i s/^target.*// acdc_project/chess_intrinsic_wrapper.ll
sed -i s/noalias_sidechannel[^,]*,// acdc_project/chess_intrinsic_wrapper.ll
sed -i s/nocallback[^,]*,// acdc_project/chess_intrinsic_wrapper.ll
aie-opt --aie-localize-locks --aie-standard-lowering=tilecol=1 tilerow=3 acdc_project/input_with_addresses.mlir -o acdc_project/core_1_3.mlir
aie-opt --aie-normalize-address-spaces --canonicalize --cse --convert-vector-to-llvm --convert-memref-to-llvm 
--convert-func-to-llvm=use-bare-ptr-memref-call-conv --convert-cf-to-llvm --canonicalize --cse acdc_project/core_1_3.mlir -o acdc_project/core_1_3.opt.mlir
aie-translate acdc_project/input_with_addresses.mlir --aie-generate-bcf --tilecol=1 --tilerow=3 -o acdc_project/core_1_3.bcf
aie-translate acdc_project/input_with_addresses.mlir --aie-generate-ldscript --tilecol=1 --tilerow=3 -o acdc_project/core_1_3.ld.script
aie-translate --mlir-to-llvmir --opaque-pointers=0 acdc_project/core_1_3.opt.mlir -o acdc_project/core_1_3.ll
cp acdc_project/core_1_3.ll acdc_project/core_1_3.chesshack.ll
sed -i s/noundef// acdc_project/core_1_3.chesshack.ll
sed -i s/noalias_sidechannel[^,],// acdc_project/core_1_3.chesshack.ll
llvm-link acdc_project/core_1_3.chesshack.ll acdc_project/chess_intrinsic_wrapper.ll -S -o acdc_project/core_1_3.chesslinked.ll
sed -i s/noundef// acdc_project/core_1_3.chesslinked.ll
sed -i -E /define .*@/ s/%[0-9]*//g acdc_project/core_1_3.chesslinked.ll
sed -i -E s/mustprogress//g acdc_project/core_1_3.chesslinked.ll
sed -i -E s/poison/undef/g acdc_project/core_1_3.chesslinked.ll
sed -i -E s/nocallback//g acdc_project/core_1_3.chesslinked.ll
awk /_include _file/ {print($3)} acdc_project/core_1_3.bcf
xchesscc_wrapper -d -f +P 4 acdc_project/core_1_3.chesslinked.ll  +l acdc_project/core_1_3.bcf -o ./core_1_3.elf
 AIE Compilation: ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% -:--:-- 0:00:03 0/2                                 
 Core (1, 3)      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 100% 0:00:00 0:00:00 14/14 xchesscc_wrapper -d -f +P 4 acchess-clang: warning: argument unused during compilation: '-I /data/Xilinx/Vitis/2021.2/aietools/data/cervino/lib' [-Wunused-command-line-argument]
chess-clang: warning: argument unused during compilation: '-I /data/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libcxx-lite/include' [-Wunused-command-line-argument]
chess-clang: warning: argument unused during compilation: '-I /data/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime_cxx/libs/libcxx-9.0.0/include-lite' [-Wunused-command-line-argument]
chess-clang: warning: argument unused during compilation: '-I /data/Xilinx/Vitis/2021.2/aietools/data/cervino/lib/runtime/include' [-Wunused-command-line-argument]
chess-clang: warning: argument unused during compilation: '-include aie_core.h' [-Wunused-command-line-argument]
 AIE Compilation: ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% -:--:-- 0:00:03 0/2                                 
 AIE Compilation: ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% -:--:-- 0:00:05 0/2                                 
 Core (1, 3)      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 100% 0:00:00 0:00:00 14/14 xchesscc_wrapper -d -f +P 4 acWarning in "/home/lay/work/aie-mlir-tutorial/demo/unit_tests/acdc_project/core_1_3.opt.mlir", line 125, column 11: loop count of type `unsigned long long' is converted to `w32' via `int', the first stage of which is non-nil
Warning in "/home/lay/work/aie-mlir-tutorial/demo/unit_tests/acdc_project/core_1_3.opt.mlir", line 400, column 12: loop count of type `unsigned long long' i
 AIE Compilation: ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% -:--:-- 0:00:05 0/2                                 
 Core (1, 3)      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 100% 0:00:00 0:00:00 14/14 xchesscc_wrapper -d -f +P 4 acWarning in "/home/lay/work/aie-mlir-tutorial/demo/unit_tests/acdc_project/core_1_3.opt.mlir", line 75, column 11: (approximate): uninitialized variable `__tmp' is used (unconditionally) by `r_update_dint_low (internal)'   (sig .10841 used by opn <11097> in arg 1)
Warning in "/home/lay/work/aie-mlir-tutorial/demo/unit_tests/acdc_project/core_1_3.opt.mlir", line 195, column 12: (approximate): uninitialized variable `__tmp' is used (unconditionally) by `r_update_dint_low (internal)'   (sig .10990 used by opn <11288> in arg 1)
Warning in "/home/lay/work/aie-mlir-tutorial/demo/unit_tests/acdc_project/core_1_3.opt.mlir", line 22: (approximate): uninitialized variable `__tmp' is used (unconditionally) by `r_update_dint_low (internal)'   (sig .11035 used by opn <11341> in arg 1)
[... cut ...] .h +Sinl +Olbb=200 +Opmsa +Olzyinl +wchesswork1398 chesswork1398/core_1_3.chesslinked.sfg +Q1=+Sinl,+Olbb=200,+Opmsa,+Olzyinl +Q2=+Sinl,+Olbb=200,+Opmsa,+Olzyinl +Q3=+Sinl,+Olbb=1000,+Opmsa,+Olzyinl +Qfast=+Sinl,+Olbb=1000,+Opmsa,+Olzyinl,+Opfp +Qs=+Sinl,+Olbb=200,+Opmsa,+Olzyinl +Qz=+Sinl,+Olbb=200,+Opmsa,+Olzyinl me
xchesscc Failed No such device
Error encountered while running: xchesscc_wrapper -d -f +P 4 acdc_project/core_1_3.chesslinked.ll  +l acdc_project/core_1_3.bcf -o ./core_1_3.elf
```

stephenneuendorffer, 2022-11-15:
You still need to remove the allocs, I think.  E.g.:

  %2 = memref.alloc() {alignment = 16 : i64} : memref<1x8x28x28xf32>

HadXu, 2022-11-16:
Thank you for your reply. As your advice, when I remove the allocs, how do I read weights?  I have no idel how to run this code by mlir-aie repo on vck190, Thanks.

stephenneuendorffer, 2022-11-16:
Can you use a Buffer instead?  (essentially statically allocating the memory in a particular tile, rather than relying on dynamically allocating it?)

HadXu, 2022-11-30:
Thank you for your advice. I re-code this [AIE.mlir](https://gist.github.com/HadXu/1a623b6927e87c095ece71ab5c9abf8a), but reuslt is not correct.  I think something error.

the AIE result
[image]

the x86 result

[image]

stephenneuendorffer, 2022-11-30:
Well that's some progress at least!  Can you share the full design in a way that we can more easily reproduce it?

AndraBisca, 2025-03-27:
Closed due to inactivity. Please feel free to re-open.

closed 2025-03-27 by AndraBisca