Xilinx/mlir-aie #265: Update me_basic.o to latest version so kernels can once again complete when program code is done.

jackl-xilinx, 2023-01-13:
Generated kernels don't seem to finish such that the core becomes deactivated. test/benchmarks/05_Core_Startup used to be able to measure program cycle count based on active and inactive core status. But now it doesn't seem to ever deactivate when the program is "done".

stephenneuendorffer, 2023-01-16:
The actual problem here is that we've always been subtly linking things wrong, using the chess libraries, but the include headers from LLVM.  One place where this difference shows up is in the exit() function, which is defined with a special inline version in the chess headers.

stephenneuendorffer, 2023-01-16:
I think previously, the 'done' instruction halted the core.  Now instead, the core is not inactivated directly, but event #2 is used to signal completion to the rest of the system.  Is there a way to trigger off that event instead?

jackl-xilinx, 2023-01-16:
I'm not sure which event you're referring to but I'm fine with using different even triggers to measure the boundaries of core usage to fix things like benchmark 05. Where this becomes more of an issue is where aiesimulator might expect the core to be deactivated in order to signal that the simulation is done and do host code cleanup. The latest experiments seems to highlight this behavior so I'm more concerned about this part. Do you notice that an updated me_basic.o doesn't actually cause the core to deactivate? Does the latest tool versions not deactivate after the core is complete as it did before or is this something we introduced in our linking?

jackl-xilinx, 2023-01-16:
Thanks Steve for the explanation and pointing in the right direction that the me_basic.o and "exiting" of the kernel is not what I'm looing for.

closed 2023-01-16 by jackl-xilinx