Xilinx/mlir-aie #279: Use GitHub pages for tutorial READMEs

jgmelber, 2023-01-20:
This would improve the experience and overall flow.

jackl-xilinx, 2023-03-14:
@jgmelber Do you mean moving the tutorial to github.io? I thought having them integrated in the source was a good way to work with the documentation and tutorials. What are the benefits of pages for the tutorial?

stephenneuendorffer, 2023-03-14:
I interpreted this to mean "leave the files where they are, but additionally publish them to https://xilinx.github.io/mlir-aie/ so that people can find everything in one place.  (but maybe Joe had a different idea?)

jgmelber, 2023-03-15:
@stephenneuendorffer yes that was my intent. The READMEs are fine to leave where they are but having the tutorial published to the GitHub.io pages would be a nice consolation and provide a neater interface.

jackl-xilinx, 2023-03-15:
Ah. That makes much more sense. Ok, I'll look into doing that.

AndraBisca, 2025-03-25:
I'm thinking of closing this as we have been discussing marking the mlir-tutorials as "legacy". Are you thinking of extending this to the IRON programming guide @jgmelber ?

closed 2025-03-26 by jgmelber