Xilinx/mlir-aie #290: mlir-aie does not accept non-standard libxaie locations

fifield, 2023-01-26:
* '\<sysroot\>/opt/xaienginev2` is hard-coded in several places in aircc.py

* cmake files do not allow override of location (e.g. https://github.com/Xilinx/mlir-aie/blob/main/test/CMakeLists.txt#L56) via setting `LibXAIE_DIR`. Instead we force the same defaults as `find_package` (https://github.com/Xilinx/cmakeModules/blob/master/FindLibXAIE.cmake#L36) when we should just let `find_package` do its job.

One impact of this is an inability to build mlir-air tests on x86 (with no cross compile and no sysroot) without installing something in `/opt`

AndraBisca, 2023-01-26:
I think this is also related to the fact that when I do 'ninja check-aie' none of the unit_tests work (because of the missing LibXAIE_DIR) but I can still use aiecc.py to compile them.

fifield, 2023-03-15:
Another place I bumped into this today: https://github.com/Xilinx/mlir-aie/blob/main/CMakeLists.txt#L65-L70

This code assumes that if I don't set `VitisSysroot` then I want cmake to set it for me. This is not true if I am compiling for x86 to run on x86. In this case I don't want to set `VitisSysroot` and I don't want any sysroot stuff set for me.

This is compounded by the following lines that require 1) libxaie living in a "sysroot" and 2) users can't set libxaie location:
https://github.com/Xilinx/mlir-aie/blob/main/reference_designs/CMakeLists.txt#L56
https://github.com/Xilinx/mlir-aie/blob/main/test/CMakeLists.txt#L57
https://github.com/Xilinx/mlir-aie/blob/main/tutorials/CMakeLists.txt#L57

If I comment out these lines and don't set `VitisSysroot` then cmake builds against the correct libxaie on my x86 host (which I can point to with `-DLibXAIE_DIR`)

denolf, 2023-04-04:
@fifield this should be addressed in #392

fifield, 2023-05-05:
Reopening because It seems this functionality has been removed:
```
CMake Warning:
  Manually-specified variables were not used by the project:

    LibXAIE_ROOT
```
The use case is building against libxaie that is locally installed to the system and not in Vitis or a sysroot. Example use cases are the mlir-air build for github actions workflow and PCIe platforms. https://github.com/Xilinx/mlir-air/blob/main/utils/github-clone-build-libxaie.sh#L21

denolf, 2023-05-08:
@fifield now that we support both aarch64 and x86_64, would it make sense to have 2 separate optional variables:
LibXAIE_x86_64_DIR
LibXAIE_aarch64_DIR
if one of these is defined, then we copy the include and library from the specified location to the mlir-aie install folder?

fifield, 2023-05-09:
The xaiengine target does slightly more than libxaie. It copies some additional headers which are assumed present in test_lib/CMakeLists.txt which unconditionally compiles the libraries for the AIESIM target. The AIESIM target then includes xioutils.h which cannot be present if there is no Vitis. This means the build fails when LibXAIE is present and Vitis is not even after hacking around the limitations raised in this issue.

denolf, 2023-05-22:
fixed with #496

closed 2023-05-22 by denolf