Xilinx/mlir-aie #359: Can mlir-aie generate the whole design (AIE kernels + ADF graph) in Cpp to enable the Vitis fow for the VCK5000 card?

shllvii, 2023-02-12:
[https://github.com/nqdtan/vck5000_vivado_custom_ulp_design](https://github.com/nqdtan/vck5000_vivado_custom_ulp_design) mentioned in #77 provides a solution to use the Vivado flow to target the VCK5000 card.
However, I'm more familiar with the Vitis flow, which requires the aiecompiler to provide a `libadf.a` file to v++.
So, could mlir-aie generate C++ programs for the AIE kernels and the ADF declaration? 
I've found that there are  [AIEVecToCpp](https://github.com/Xilinx/mlir-aie/tree/main/test/Targets/AIEVecToCpp) and [ADFGenerateCPPGraph](https://github.com/Xilinx/mlir-aie/tree/main/test/Targets/ADFGenerateCPPGraph) targets, but they don't aim to convert a module including essentials [Tile Components](https://github.com/Xilinx/mlir-aie/tree/main/tutorials/tutorial-1#tile-components).
Or, could mlir-aie compile a solution to a `libadf.a` file straightly?

stephenneuendorffer, 2023-02-14:
The programming model of mlir-aie is (for the most part) lower-level than what is provided by Vitis. It's more likely that targetting a C++ program with an ADF graph could be done from something like AIR.  The ADF dialect is intended to be a proof-of-concept in this direction.

Having mlir-aie generate a libadf.a directly is actually an interesting idea that we haven't really considered.  It would probably take some investigation and might have some advantages: for instance, we might need less direct involvement with the setup of the aiesimulator.  @jackl-xilinx Want to look into this?

shllvii, 2023-02-15:
Indeed, it's just the MLIR-based low-level features of mlir-aie that attract me a lot.
I acknowledge that it's inappropriate for mlir-aie to generate everything in C++. I put forward this issue since I wish to use mlir-aie to produce artifacts that could be combined with PL designs on VCK5000.  A libadf.a looks like a good choice since it may combine the existing features in mlir-aie, such as vectorization, routing, compilation, etc.
Though I don't know specific details about mlir-aie and aiecompiler, I find that mlir-aie generates LLVM files and  ELF files with the help of the Chess compiler, which is also adopted by aiecompiler. So I guess that mlir-aie probably has the capability to generate a libadf.a file.
I'm looking forward to more discussion about the libadf.a generation. Besides, I'll really appreciate it if you could provide some help about how to use mlir-aie together with the Vitis tool to target VCK5000.

jackl-xilinx, 2023-02-16:
@shllvii This is a timely topic as we're looking into adapting more of what mlir-aie can generate to interact more seamlessly with the aiesimulator and vitis analyzer. @stephenneuendorffer is right that we haven't considered creating libadf.a but generating this is in line with how we want to expand the tranformation targets for mlir-aie. This requires some more investigation, in particular, how libadf.a is used by the rest of the tools. Watch this space as we plan to look into this area in the near future.