Xilinx/mlir-aie #391: Build dependence on specific version of Vitis - not verified by CMake

SamuelBayliss, 2023-03-17:
The readme correctly specified a version of vitis that you must have to get the project to build... but if you don't read that file and try to build with another version of Vitis .. you get errors from places like runtime_lib/xaiengine/CMakeLists.txt where specific versions of the aiengine driver are referenced... 

Can we make 'FindVitis' verify that a compatible version of vitis is installed, and fail if you a) have a version b) it's the wrong one and c) you asked the runtime library to be built. We could even put in a friendly message suggesting what you might be able to build without a specific version of Vitis ( so users can get on with something while they wait for a download of a new version).

fifield, 2023-03-17:
+1
`FindVitis` is part of https://github.com/Xilinx/cmakeModules

denolf, 2023-04-29:
#415 merged in main, closing issue.

closed 2023-04-29 by denolf