Xilinx/mlir-aie #535: ObjectFifoLinkOp + One-Dimensional Memref = Segmentation Fault

andrej, 2023-07-06:
This is probably not super-high priority, since using a one-dimensional memref is not a common use case. Nonetheless, the new `objectFifo.link` operation does not play along with one-dimensional memrefs.

Here is a minimal example:
```
module @linkbug {
    AIE.device(xcve2302) {
        %tile22 = AIE.tile(2, 2)  // producer tile
        %tile12 = AIE.tile(1, 2)  // mem tile
        %tile23 = AIE.tile(2, 3)  // consumer tile
        %fifo0 = AIE.objectFifo.createObjectFifo(%tile22, {%tile12}, 4 : i32) {sym_name = "fifo0"} : !AIE.objectFifo<memref<i32>>
        %fifo1 = AIE.objectFifo.createObjectFifo(%tile12, {%tile23}, 4 : i32) {sym_name = "fifo1"} : !AIE.objectFifo<memref<i32>>
        AIE.objectFifo.link(%fifo0, {%fifo1}) : (!AIE.objectFifo<memref<i32>>, !AIE.objectFifo<memref<i32>>)
    }
}
```

When ran through `aie-opt --aie-objectFifo-stateful-transform`, this results in:

```
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Stack dump:
0.      Program arguments: aie-opt --aie-objectFifo-stateful-transform linkbug.mlir
 #0 0x00007f668679a287 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (/scratch2/androsti/mlir-aie/llvm/install/lib/libLLVM-17git.so+0x19fc287)
 #1 0x00007f66867981de llvm::sys::RunSignalHandlers() (/scratch2/androsti/mlir-aie/llvm/install/lib/libLLVM-17git.so+0x19fa1de)
 #2 0x00007f668679aa5f SignalHandler(int) Signals.cpp:0:0
 #3 0x00007f66891c8420 __restore_rt (/lib/x86_64-linux-gnu/libpthread.so.0+0x14420)
 #4 0x000056313e6717fb xilinx::AIE::ObjectFifoLinkOp::verify() (/scratch2/androsti/mlir-aie/install/bin/aie-opt+0x12357fb)
 #5 0x000056313e6a7a70 mlir::Op<xilinx::AIE::ObjectFifoLinkOp, mlir::OpTrait::ZeroRegions, mlir::OpTrait::ZeroResults, mlir::OpTrait::ZeroSuccessors, mlir::OpTrait::AtLeastNOperands<1u>::Impl, mlir::OpTrait::OpInvariants>::verifyInvariants(mlir::Operation*) (/scratch2/androsti/mlir-aie/install/bin/aie-opt+0x126ba70)
 #6 0x000056313e6a7560 mlir::RegisteredOperationName::Model<xilinx::AIE::ObjectFifoLinkOp>::verifyInvariants(mlir::Operation*) (/scratch2/androsti/mlir-aie/install/bin/aie-opt+0x126b560)
 #7 0x000056313e9ef8b3 (anonymous namespace)::OperationVerifier::verifyOperation(mlir::Operation&) Verifier.cpp:0:0
 #8 0x000056313e9efa9b (anonymous namespace)::OperationVerifier::verifyOperation(mlir::Operation&) Verifier.cpp:0:0
 #9 0x000056313e9f155a mlir::LogicalResult mlir::failableParallelForEach<mlir::Operation**, (anonymous namespace)::OperationVerifier::verifyOperation(mlir::Operation&)::$_0>(mlir::MLIRContext*, mlir::Operation**, mlir::Operation**, (anonymous namespace)::OperationVerifier::verifyOperation(mlir::Operation&)::$_0&&) Verifier.cpp:0:0
#10 0x000056313e9efed1 (anonymous namespace)::OperationVerifier::verifyOperation(mlir::Operation&) Verifier.cpp:0:0
#11 0x000056313e9f155a mlir::LogicalResult mlir::failableParallelForEach<mlir::Operation**, (anonymous namespace)::OperationVerifier::verifyOperation(mlir::Operation&)::$_0>(mlir::MLIRContext*, mlir::Operation**, mlir::Operation**, (anonymous namespace)::OperationVerifier::verifyOperation(mlir::Operation&)::$_0&&) Verifier.cpp:0:0
#12 0x000056313e9efed1 (anonymous namespace)::OperationVerifier::verifyOperation(mlir::Operation&) Verifier.cpp:0:0
#13 0x000056313e9ef77c mlir::verify(mlir::Operation*, bool) (/scratch2/androsti/mlir-aie/install/bin/aie-opt+0x15b377c)
#14 0x000056313e87ea3d mlir::parseAsmSourceFile(llvm::SourceMgr const&, mlir::Block*, mlir::ParserConfig const&, mlir::AsmParserState*, mlir::AsmParserCodeCompleteContext*) (/scratch2/androsti/mlir-aie/install/bin/aie-opt+0x1442a3d)
#15 0x000056313e866f15 mlir::parseSourceFile(std::shared_ptr<llvm::SourceMgr> const&, mlir::Block*, mlir::ParserConfig const&, mlir::LocationAttr*) (/scratch2/androsti/mlir-aie/install/bin/aie-opt+0x142af15)
#16 0x000056313e6637f9 mlir::parseSourceFileForTool(std::shared_ptr<llvm::SourceMgr> const&, mlir::ParserConfig const&, bool) (/scratch2/androsti/mlir-aie/install/bin/aie-opt+0x12277f9)
#17 0x000056313e663153 performActions(llvm::raw_ostream&, bool, bool, std::shared_ptr<llvm::SourceMgr> const&, mlir::MLIRContext*, llvm::function_ref<mlir::LogicalResult (mlir::PassManager&)>, bool, bool) MlirOptMain.cpp:0:0
#18 0x000056313e662f49 mlir::LogicalResult llvm::function_ref<mlir::LogicalResult (std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::raw_ostream&)>::callback_fn<mlir::MlirOptMain(llvm::raw_ostream&, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::function_ref<mlir::LogicalResult (mlir::PassManager&)>, mlir::DialectRegistry&, bool, bool, bool, bool, bool, bool, bool)::$_0>(long, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::raw_ostream&) MlirOptMain.cpp:0:0
#19 0x000056313e9fb348 mlir::splitAndProcessBuffer(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::function_ref<mlir::LogicalResult (std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::raw_ostream&)>, llvm::raw_ostream&, bool, bool) (/scratch2/androsti/mlir-aie/install/bin/aie-opt+0x15bf348)
#20 0x000056313e66149e mlir::MlirOptMain(llvm::raw_ostream&, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::function_ref<mlir::LogicalResult (mlir::PassManager&)>, mlir::DialectRegistry&, bool, bool, bool, bool, bool, bool, bool) (/scratch2/androsti/mlir-aie/install/bin/aie-opt+0x122549e)
#21 0x000056313e66192f mlir::MlirOptMain(int, char**, llvm::StringRef, mlir::DialectRegistry&, bool) (/scratch2/androsti/mlir-aie/install/bin/aie-opt+0x122592f)
#22 0x000056313d552298 main (/scratch2/androsti/mlir-aie/install/bin/aie-opt+0x116298)
#23 0x00007f6684882083 __libc_start_main /build/glibc-SzIz7B/glibc-2.31/csu/../csu/libc-start.c:342:3
#24 0x000056313d551f6e _start (/scratch2/androsti/mlir-aie/install/bin/aie-opt+0x115f6e)
```

jgmelber, 2023-07-06:
Did you try `1xi32` as the object type

andrej, 2023-07-06:
It does work with `1xi32` as the memref type. So it seems somewhere in the code, it is assumed that there is at least another dimension. Other uses of objectFifo also work with just `i32`, however, so it would be nice if this would too (or at least error out instead of segfault). I think this is not a high priority though!

closed 2023-09-20 by AndraBisca