Xilinx/mlir-aie #538: Cosmetic: Multiple Lock Acquires for Object Fifo

andrej, 2023-07-07:
Andra asked me to start this issue so it is documented. Please assign her to it.

Currently, in the `--aie-objectFifo-stateful-transform` pass, the following
```
%s = AIE.objectFifo.acquire<Produce>(%fifo1 : !AIE.objectFifo<memref<1xi32>>, 3) : !AIE.objectFifoSubview<memref<1xi32>>
```
lowers to something like this
```
AIE.useLock(%7, AcquireGreaterEqual, 1)
AIE.useLock(%7, AcquireGreaterEqual, 1)
AIE.useLock(%7, AcquireGreaterEqual, 1)
```

Arguably, it would be more elegant to lower this to
```
AIE.useLock(%7, AcquireGreaterEqual, 1)
```

Since the "AcquireGreaterEqual" operation also decrements the lock by one, the two are semantically equivalent as long as no one else races for the lock. In our objectFifo implementation, there is no such race, and Kristof and me convinced ourselves that the above implementation is not an issue. So this change would be purely cosmetical.

AndraBisca, 2023-07-07:
Thank you for finding this and opening the issue!

AndraBisca, 2023-07-13:
Fixed in #543

closed 2023-07-13 by AndraBisca