Xilinx/mlir-aie #544: WorkerAIEBuild license not found !!No such device

judeharis, 2023-07-16:
I have been trying to install MLIR-AIE and work through the tutorials provided: 
https://github.com/Xilinx/mlir-aie/tree/main/tutorials

After following the guide to build MLIR-AIE on Ubuntu 22.04: 
https://xilinx.github.io/mlir-aie/Building.html

When running make in tutorial-1 folder I run into the following message.
```
The Program xchesscc has encountered an unexpected error exiting ... AIEBuild license not found !!No such device
Error encountered while running: xchesscc_wrapper aie +w aie.mlir.prj/work -c -d -f +f +P 4 /mnt/Crucial/WorkspaceB/MLIR-AIE/mlir-aie/install/bin/aiecc/../../aie_runtime_lib/AIE/chess_intrinsic_wrapper.cpp -o aie.mlir.prj/chess_intrinsic_wrapper.ll
```

I have installed the AIEBuild licence (.xic file) through the License Manager.
It seems the xchesscc cannot find the license even though it shows up in license manager.
Any idea help in this matter would appreciated.

stephenneuendorffer, 2023-07-17:
Did you restart your shell?
Have you confirmed that LM_LICENSE_FILE or XILINXD_LICENSE_FILE are set appropriately within the shell?
Are you running 2022.2 or 2023.1?

judeharis, 2023-07-17:
Hi @stephenneuendorffer, yes I have tried have restarted the shell, and also used "source utils/env_setup.sh"  to setup the environment. Manually exporting LM_LICENSE_FILE and XILINXD_LICENSE_FILE to directly to Xilinx.lic file also has no effect.

I am using Xilinx 2022.2.
One thing to note I had previous had issue using the "2022.2/tps/lnx64/cmake-3.3.2" to compile MLIR-AIE using "./utils/build-mlir-aie.sh <llvm dir>/<build dir> " due to cmake version requirement of 3.21 or higher, hence I build MLIR-AIE using new version of cmake. I am not sure if this could cause of problem with unrecognised license for xchesscc.

stephenneuendorffer, 2023-07-17:
Can you try updating to 2023.1?

judeharis, 2023-07-20:
Apologies for the delayed response. Took me some time to update to 2023.1 and start with clean version of MLIR-AIE repo.  The same issues persisted but I was able to overcome them. For future reference:

- CMake which comes with Vitis is unusable for building mlir-aie but sourcing "settings64.sh" (adds CMake 3.3.2 to PATH)   is recommended before running "./utils/build-mlir-aie.sh " script.
  - Fix: I manually removed path to CMake3.3.2 from "Vitis/2023.1/settings64-Vitis.sh" 
- Problem with AIEBuild licence, was due to difference between Host-ID types supported by Linux and Windows,  I was using disk ID previously. I have not had this issue using Vivado/Vitis by itself before so i did not realise it would cause issue with xchesscc.
  - Fix: switching to Ethernet MAC address fixed this issue.

keryell, 2023-07-20:
Yes, I do not understand why some people insist on shipping super old software with Vivado and Vitis.
This is why usually I never source the configuration files as it might break any modern environment.
See the `NOTE:` https://github.com/Xilinx/mlir-aie/blob/main/docs/Building.md as a possible alternative.

Dhushyanths, 2024-02-08:
How do u configure the host ID to ethernet MAC address where the AIE license supports

judeharis, 2024-02-12:
Within the AMD's Product Licensing tool: https://www.xilinx.com/getlicense , you can a create licence with all that you need. If you have already made a licence for go to "Manage License" and "Modify License" then within the "Host ID" section select add a host. Then choose OS and set type as Ethernet MAC.