Xilinx/mlir-aie #584: Support vector.transfer_read of scalar memref

jamestcl-amd, 2023-08-17:
- For vector canonicalization, it doesn't support vector.transfer_read from scalar memref
- Sample test at https://github.com/Xilinx/mlir-aie/blob/main/test/dialect/AIEVec/precanonicalization.mlir
- Sample code at https://github.com/Xilinx/mlir-aie/blob/1cf052849d09f0583564dca1b138b165eff9a7d1/lib/Dialect/AIEVec/Transforms/VectorToVectorConversions.cpp#L128
- Workaround:
[image]

jsetoain, 2023-08-17:
@jamestcl-amd 

[PR#586](https://github.com/Xilinx/mlir-aie/pull/586) Deals with this issue. I tried to add you as a reviewer but looks like I'm not allowed. You should be able to review anyway.

jamestcl-amd, 2023-08-17:
Thanks. I will need Steve to give me permissions against this repo.

jsetoain, 2023-08-30:
Resolved by #586

closed 2023-08-30 by jsetoain