Xilinx/mlir-aie #585: Improve Vector->AIEVec lowering strategy for arith.addi/addf/muli/mulf

jamestcl-amd, 2023-08-17:
- For input example (after vector canonicalization) like below, the `arith.mulf` and `arith.addf` will not be converted to `aievec.mul_elem` and `aievec.add_elem`. This is due to the FMA/Conv check that specifically leave these operations unconverted for the aievec.conv op lowering.
```
module {
  func.func @func1(%arg0: memref<1024xbf16>, %arg1: memref<1xbf16>, %arg2: memref<1024xbf16>) {
    %cst = arith.constant dense<0.000000e+00> : vector<16xbf16>
    %cst_0 = arith.constant 0.000000e+00 : bf16
    affine.for %arg3 = 0 to 1024 step 16 {
      %0 = vector.transfer_read %arg0[%arg3], %cst_0 {in_bounds = [true]} : memref<1024xbf16>, vector<16xbf16>
      %1 = vector.transfer_read %arg1[%arg3], %cst_0 : memref<1xbf16>, vector<16xbf16>
      %2 = vector.extract %1[0] : vector<16xbf16>
      %3 = vector.broadcast %2 : bf16 to vector<16xbf16>
      %4 = arith.minf %0, %cst : vector<16xbf16>
      %5 = arith.mulf %3, %4 : vector<16xbf16>
      %6 = arith.maxf %0, %cst : vector<16xbf16>
      %7 = arith.addf %6, %5 : vector<16xbf16>
      vector.transfer_write %7, %arg2[%arg3] {in_bounds = [true]} : vector<16xbf16>, memref<1024xbf16>
    }
    return
  }
}
```
- Here are the checks that blocks the lowering of the above example:
- https://github.com/Xilinx/mlir-aie/blob/1cf052849d09f0583564dca1b138b165eff9a7d1/lib/Dialect/AIEVec/Transforms/VectorToAIEVecConversions.cpp#L838-L842
- https://github.com/Xilinx/mlir-aie/blob/c49db4fa3a82201282e8f80781bbf1f629eda81e/lib/Dialect/AIEVec/Transforms/VectorToAIEVecConversions.cpp#L526-L529
- https://github.com/Xilinx/mlir-aie/blob/c49db4fa3a82201282e8f80781bbf1f629eda81e/lib/Dialect/AIEVec/Transforms/VectorToAIEVecConversions.cpp#L1849-L1852

AndraBisca, 2025-03-25:
Have there been any updates on this issue? @jsetoain

jsetoain, 2025-03-31:
> Have there been any updates on this issue? [@jsetoain](https://github.com/jsetoain)

No, we never got back to it. The solution we were thinking about (always converting) breaks more things than it fixes, and the alternative (never converting) was waiting for better support for generic vector arithmetic operations in Peano. There's been a lot of progress in Peano since we opened this issue, so a lot of it might have been fixed on its own just because of that.