Xilinx/mlir-aie #640: Problems with opaque pointers

oowekyala, 2023-09-18:
Hi, I can't seem to run the tests in this repo. I get failures related to the use of `--opaque-pointers=0` as an argument to `llvm-link`. This happens regardless of the version of `llvm-link` that's in my PATH (I think this option has been removed with LLVM 17 so I tested it with LLVM 15 and 16).

Related to #423. Maybe you already know of a workaround?

fifield, 2023-09-19:
The workaround is to use a patched llvm or [use Vitis 2023.2](https://github.com/Xilinx/mlir-aie/pull/634) (which is not yet released). I thought I had a working llvm patch but it seems to be out of date. @makslevental do you have an updated llvm patch for this?

makslevental, 2023-09-19:
> The workaround is to use a patched llvm or [use Vitis 2023.2](https://github.com/Xilinx/mlir-aie/pull/634) (which is not yet released). I thought I had a working llvm patch but it seems to be out of date. @makslevental do you have an updated llvm patch for this?

https://github.com/makslevental/mlir-wheels/blob/main/patches/opaqueptrs.patch but I ablated it down so that one needs to explicitly pass `use-opaque-pointers=0` to the various passes (i.e., rather than it being the default).

cabreraam, 2023-09-20:
I've tried running [tutorial-1](https://github.com/Xilinx/mlir-aie/tree/main/tutorials/tutorial-1) after applying the `opaqueptrs.patch`, rebuilding LLVM, and setting `NO_USE_OPQUE_POINTERS=1`. 

After doing this, I run into the following issue:
```bash
aie-translate --opaque-pointers=0 --mlir-to-llvmir aie.mlir.prj/input_opt_with_addresses.mlir -o aie.mlir.prj/input.ll
Error encountered while running: aie-translate --opaque-pointers=0 --mlir-to-llvmir aie.mlir.prj/input_opt_with_addresses.mlir -o aie.mlir.prj/input.ll
```

When I modified `aiecc.py` to pass `use-opaque-pointers=0` instead of `opaque-pointers=0`  I run into a similar issue, i.e., an `aie-*` utility doesn't recognize the option:
```bash
Done in 0.021 sec: aie-translate --use-opaque-pointers=0 --mlir-to-llvmir aie.mlir.prj/input_opt_with_addresses.mlir -o aie.mlir.prj/input.ll
Error encountered while running: aie-translate --use-opaque-pointers=0 --mlir-to-llvmir aie.mlir.prj/input_opt_with_addresses.mlir -o aie.mlir.prj/input.ll
```

Am I doing something incorrectly?

fifield, 2023-09-20:
Yeah the patch referenced above is not sufficient. The `use-opaque-pointers` options was removed from upstream LLVM and there appears to be other changes not addressed in the patch. I do not know of a fix at this time.

makslevental, 2023-09-20:
`use-opaque-pointers` wasn't removed everywhere upstream; it remains on a pass-by-pass basis for certain of the LLVM conversion passes:

https://github.com/search?q=repo%3Allvm%2Fllvm-project+use-opaque-pointers+path%3AMLIR%2Finclude&type=code

What remains (if you're indeed using my distribution) is to set these passes to have 
`use-opaque-pointers=0` individually in `aiecc.py` (or wherever you're driving `aie-opt`). 

Note, I didn't double check whether there's a pass pipeline here (in mlir-aie) that uses some of those passes *and* relies on defaults being correct (in which case we need to accordingly adjust that pass pipeline).

I am OOO until beginning of Oct but if this is a blocker then I can investigate further.

cc @stephenneuendorffer

fifield, 2023-09-20:
`use-opaque-pointers` was removed from `mlir-translate` (i.e. `aie-translate`), which is where the error is coming from. But after fixing that I still couldn't get `aie-translate` to produce typed pointers. It's also possible I'm doing something wrong.

makslevental, 2023-09-21:
@fifield you don't need that flag if you set `use-opaque-pointers=0` on the individual passes that convert to llvm (see the [MWE here](https://github.com/Xilinx/mlir-aie/pull/590#issuecomment-1684524691)). 

I pushed up PR https://github.com/Xilinx/mlir-aie/pull/647 that was "brain compiled" since, as I do not have chess locally, I cannot test.

makslevental, 2023-09-26:
@oowekyala @cabreraam this is fixed now after [the revert](https://github.com/Xilinx/mlir-aie/pull/590). Sorry for the miscommunication/frustration (next time we bump it'll be thoroughly tested 😊).

cc @hongzhengTian

closed 2023-09-26 by makslevental