Xilinx/mlir-aie #642: Build issue

gabrielrodcanal, 2023-09-19:
Hi,
I'm having the following issue when building with the script `utils/build-mlir-aie.sh`:
```
[70/108] Performing configure step for 'xaiengine_x86_64'
loading initial cache file /home/gabriel/mlir-aie/build/runtime_lib/xaiengineTmp/x86_64/tmp/xaiengine_x86_64-cache-Release.cmake
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:18 (project):
  The CMAKE_C_COMPILER:

    clang-12

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:18 (project):
  The CMAKE_CXX_COMPILER:

    clang++-12

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
```
`clang 14` is on the `$PATH` and I have set `CC=clang` and `CXX=clang++`.

Thank you.

fifield, 2023-09-19:
I think the problem is that the CMakeLists.txt for `rutime_lib` is [setting its own toolchain](https://github.com/Xilinx/mlir-aie/blob/main/runtime_lib/CMakeLists.txt#L21) file which is conflicting with your settings. Try modifying [toolchain_x86_64.cmake](https://github.com/Xilinx/mlir-aie/blob/main/cmake/toolchainFiles/toolchain_x86_64.cmake) to use clang-14

makslevental, 2023-09-26:
@fifield Can we ditch the `toolchain_x86_64.cmake` file entirely or is there a "load bearing" reason for its existence?

cabreraam, 2023-09-26:
I wondered this myself. I found myself having to edit the version to reflect what gets built in build-llvm-local.sh and removing the version info from clang++ because for some reason, the binary for clang++-18 never got created. All of my experience was limited to the state of the repo last week though, so my opinion may already be out of date.

fifield, 2023-09-26:
> @fifield Can we ditch the `toolchain_x86_64.cmake` file entirely or is there a "load bearing" reason for its existence?

The reason for it's existence is that we often build the runtime differently than the rest of the tools (e.g. for building a cross-compiler). This is being refactored by others as I type so let's just leave it alone for now.

makslevental, 2023-09-26:
@cabreraam @gabrielrodcanal 

for the time being, you can just kill the dep on the toolchain

```diff
diff --git a/runtime_lib/CMakeLists.txt b/runtime_lib/CMakeLists.txt
index 30c54b28..092b5cc0 100644
--- a/runtime_lib/CMakeLists.txt
+++ b/runtime_lib/CMakeLists.txt
@@ -16,10 +16,10 @@ include(ExternalProject)
 project("runtime-libs")
 #add_custom_target(runtime-libs ALL)
 
-set(AIE_RUNTIME_TEST_TARGET_VAL ${AIR_RUNTIME_TEST_TARGET})
-if(NOT x86_64_TOOLCHAIN_FILE)
-  set(x86_64_TOOLCHAIN_FILE "${CMAKE_CURRENT_SOURCE_DIR}/../cmake/toolchainFiles/toolchain_x86_64.cmake")
-endif()
+#set(AIE_RUNTIME_TEST_TARGET_VAL ${AIR_RUNTIME_TEST_TARGET})
+#if(NOT x86_64_TOOLCHAIN_FILE)
+#  set(x86_64_TOOLCHAIN_FILE "${CMAKE_CURRENT_SOURCE_DIR}/../cmake/toolchainFiles/toolchain_x86_64.cmake")
+#endif()
 if(NOT aarch64_TOOLCHAIN_FILE)
   if(SysrootAarch64)
   set(aarch64_TOOLCHAIN_FILE "${CMAKE_CURRENT_SOURCE_DIR}/../cmake/toolchainFiles/toolchain_clang_crosscomp_pynq.cmake")
@@ -36,10 +36,10 @@ foreach(target ${AIE_RUNTIME_TARGETS})
     message("Testing with AIE runtime target: ${AIE_RUNTIME_TEST_TARGET_VAL}")
   endif()
 
-  if (NOT EXISTS ${${target}_TOOLCHAIN_FILE})
-    message(FATAL_ERROR "Toolchain file ${${target}_TOOLCHAIN_FILE} not found! Cannot build target ${target}.")
-  endif()
-  message("Building AIE runtime for ${target} with TOOLCHAIN=${${target}_TOOLCHAIN_FILE}")
+#  if (NOT EXISTS ${${target}_TOOLCHAIN_FILE})
+#    message(FATAL_ERROR "Toolchain file ${${target}_TOOLCHAIN_FILE} not found! Cannot build target ${target}.")
+#  endif()
+#  message("Building AIE runtime for ${target} with TOOLCHAIN=${${target}_TOOLCHAIN_FILE}")
   
   # xaiengine
   # if no LibXAIE_{target}_DIR specified, build xaiengine_${target} from Vitis
@@ -59,7 +59,7 @@ foreach(target ${AIE_RUNTIME_TARGETS})
       CMAKE_CACHE_ARGS
           -DCMAKE_MODULE_PATH:STRING=${CMAKE_MODULE_PATH}
       CMAKE_ARGS
-          -DCMAKE_TOOLCHAIN_FILE=${${target}_TOOLCHAIN_FILE}
+#          -DCMAKE_TOOLCHAIN_FILE=${${target}_TOOLCHAIN_FILE}
           -DCMAKE_BUILD_TYPE=${CMAKE_BUILD_TYPE}
           -DCMAKE_INSTALL_PREFIX=${CMAKE_INSTALL_PREFIX}
           -DAIE_RUNTIME_TARGET=${target}
@@ -91,7 +91,7 @@ foreach(target ${AIE_RUNTIME_TARGETS})
       CMAKE_CACHE_ARGS
           -DCMAKE_MODULE_PATH:STRING=${CMAKE_MODULE_PATH}
       CMAKE_ARGS
-          -DCMAKE_TOOLCHAIN_FILE=${${target}_TOOLCHAIN_FILE}
+#          -DCMAKE_TOOLCHAIN_FILE=${${target}_TOOLCHAIN_FILE}
           -DCMAKE_BUILD_TYPE=${CMAKE_BUILD_TYPE}
           -DCMAKE_INSTALL_PREFIX=${CMAKE_INSTALL_PREFIX}
           -DAIE_RUNTIME_TARGET=${target}
```

closed 2023-09-26 by makslevental