Xilinx/mlir-aie #676: License error running make tutorial-1

gabrielrodcanal, 2023-10-09:
Hi,
I tried to run make on the first tutorial in the repository and I get the following error:
`aiecc.py -j4 aie.mlir
Found Vitis at /home/gabriel/Xilinx/Vitis/2023.1
 MLIR compilation: ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% -:--:-- 0:00:01 0/1 1 WorkerAIEBuild license not found !!No such device
The Program xchesscc has encountered an unexpected error exiting ... AIEBuild license not found !!No such device
Error encountered while running: xchesscc_wrapper aie +w aie.mlir.prj/work -c -d -f +f +P 4 /home/gabriel/mlir-aie/install/aie_runtime_lib/AIE/chess_intrinsic_wrapper.cpp -o aie.mlir.prj/chess_intrinsic_wrapper.ll
 Error ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% -:--:-- 0:00:01 0/1 1 Worker
make: *** [Makefile:14: core_1_4.elf] Error 1
`

It seems xchesscc is unable to find the license? However, when I built the AIE MLIR toolchain everything went fine - after installing the AIE license, of course. Even running `aiecompiler` doesn't throw any license error.

stephenneuendorffer, 2023-10-09:
Indeed, it seems that there is a license problem.
Have you confirmed that LM_LICENSE_FILE or XILINXD_LICENSE_FILE are set appropriately within the shell?
Do you have the no charge AIEngine license in your license file?

gabrielrodcanal, 2023-10-09:
Those environment variables are not set in my shell. Where should they point to?

stephenneuendorffer, 2023-10-09:
To your license file.  It surprises me a bit, but perhaps aiecompiler doesn't itself check the license? https://www.xilinx.com/support/licensing_solution_center.html

gabrielrodcanal, 2023-10-09:
Hi, the issue was my host ID did not match the one in the license, as they explain here: https://support.xilinx.com/s/question/0D52E00006iHpnkSAC/lmlicensefile-not-set?language=en_US. I regenerated the license correcting that. Oddly enough, this does not set the environment variables you mentioned but fixes the issue.

Thanks!

closed 2023-10-09 by gabrielrodcanal

stephenneuendorffer, 2023-10-09:
Great!  I believe there is also a system global license file...