Xilinx/mlir-aie #698: 'AIE.objectFifo' op expects parent op 'AIE.device' on tutorial-4

gabrielrodcanal, 2023-10-27:
I get this error at the MLIR level when running `make` on tutorial-4. This also happens on the objectFifo version of tutorial-3.

gabrielrodcanal, 2023-10-27:
The fix is as simple as surrounding all the MLIR code with:
```
AIE.device(xcvc1902) {
// Code
}
```

All the tutorials that make use of `AIE.objectFifo` suffer from this problem.

AndraBisca, 2023-10-27:
Hey Gabriel! Thanks for pointing this out. I will prepare a patch to fix this issue in all tutorials. In the meantime, please use the fix that you proposed. Apologies for the inconvenience.

gabrielrodcanal, 2023-10-27:
> Hey Gabriel! Thanks for pointing this out. I will prepare a patch to fix this issue in all tutorials. In the meantime, please use the fix that you proposed. Apologies for the inconvenience.

I'm happy to contribute a pull request with that fix if you haven't fixed that already yourself yet :)

closed 2023-11-01 by AndraBisca