Xilinx/mlir-aie #720: Tutorial 5 fails to compile

gabrielrodcanal, 2023-11-06:
Hi there,
I've noticed that tutorial 5 doesn't compile for me. It's throwing the following errors:
[image]

I'm not sure if it has to do with my local installation or there is some problem with the lowering. Could anyone confirm if this happens to them too?

AndraBisca, 2023-11-06:
Hello Gabriel! This is not an issue that has to do with your build, rather this test uses a deprecated API for the problematic functions. You can see an example of the new API [here](https://github.com/Xilinx/mlir-aie/blob/main/test/unit_tests/aie/17_shim_dma_with_core/test.cpp#L114).

gabrielrodcanal, 2023-11-07:
Hi Andra,
I applied those changes to the `test.cpp` code and I got it to compile. However, when I run the simulation with `make -C aie.mlir.prj/sim` I get the following error:

commit 5ce027dd (2023-12-05): [tutorial] update tutorial5

Fixes #720
closed 2023-12-05 by stephenneuendorffer