Xilinx/mlir-aie #778: Support for Linux on RyzenAI desktop and laptops

chaudhariatul, 2023-11-24:
Thanks for documentation and tutorials for RyzenAI on Windows and using WSL.

Is there any documentation for Desktops or Laptops when Linux booted?

A couple documents refer to Ubuntu20.04 and Petalinux in these documents:
1/ https://github.com/Xilinx/mlir-aie/blob/main/docs/Building.md, 
2/ https://github.com/Xilinx/mlir-aie/blob/main/docs/Platform.md 
and couple more places,
which doesn't address known challenges on how to enable and verify the AI Engine when Linux booted.

jgmelber, 2023-11-24:
Linux drivers are not currently available. We will pass the request and interest on to the driver developers.

chaudhariatul, 2023-11-24:
Hope this issue on RyzenAI helps with a boost on this request:

https://github.com/amd/RyzenAI-SW/issues/2

stalkerg, 2023-11-27:
It requires a specific llvm version and can't be integrated into the system llvm (from Linux distribution). Seems like they are trying to bundle the most complex part in `Versal`, but it will not work for end-users the same as ROCm (at least for Linux desktop users)
We need the Mesa-level solution to start making Desktop applications for end-users (not for businesses OR scientists). 

Anyway, firstly, we should get drivers and docs for Ryzen AI after that, we can try to build our own stack or improve the current one. 
PS Seems like Tenstorrent has a simular arch.

chaudhariatul, 2023-11-27:
Support for linux OS where we are able to build business/research applications running on edge devices will be a big incentive to adopt RyzenAI based small form factor devices.

jgmelber, 2024-02-02:
Closed in #971

closed 2024-02-02 by jgmelber