Xilinx/mlir-aie #789: Linking Errors When Trying to Build with `build-llvm-local.sh`

cabreraam, 2023-11-27:
Hi all,

This is the first time I've tried to build since #640 

When issuing `./utils/build-llvm-local.sh`, I'm running into linking issues now; all missing symbols are in the `llvm` namespace FWIW. I'm using LLVM/Clang 14.0.0. __I've attached a CMake build log__.
[log.txt](https://github.com/Xilinx/mlir-aie/files/13481647/log.txt)

Can anyone help with this?

Thanks

cabreraam, 2023-11-28:
Apparently it's a [known issue](https://github.com/llvm/llvm-project/issues/55330) that setting `LLVM_LINK_LLVM_DYLIB` and `CMAKE_{C,CXX}_VISIBILITY_PRESET=HIDDEN` will result in the linker errors I observed. Indeed, removing the visibility preset options from `build-llvm-local.sh` fixed my linking issue.

stephenneuendorffer, 2023-11-28:
@makslevental Anything to do here?

cabreraam, 2023-11-28:
The jury's still out on whether or not this was just an isolated incident on my work machine. I'm happy to close the issue if no one else was experiencing this.

makslevental, 2023-11-28:
> @makslevental Anything to do here?

Simple - just drop `-DLLVM_LINK_LLVM_DYLIB=ON` from local build (we have no need for that).

makslevental, 2023-11-28:
Just btw - those symbol privacy flags aren't a workaround for anything - they're the [recommended way](https://github.com/llvm/llvm-project/pull/71133#discussion_r1383393924) to [build for distribution](https://github.com/llvm/torch-mlir/blob/dc9ea08db5ac295b4b3f91fc776fef6a702900b9/setup.py#L90-L92).

closed 2023-11-28 by makslevental