Xilinx/mlir-aie #799: tutorial-2a make failed

Dinngger, 2023-11-29:
ubuntu 20.04 with vitis 2023.2
tutorial-1 make ok while tutorial-2a make failed.
[image]

cabreraam, 2023-11-29:
A dirty solution is to remove the `libcdo_driver.so` dependence by removing the `-lcdo_driver` option [here](https://github.com/Xilinx/mlir-aie/blob/abbcb9a05a37e33217db8a0771594b81e3c4b95f/python/compiler/aiecc/main.py#L520C57-L520C57).

But that doesn't solve the underlying linking issue, which I am unsure of how to proceed.

After trying tutorial 2b, I observed a couple of issues:
1. When you run `make`, there is no `sim` directory created in `air.mlir.prj`, though one gets created in tutorial 2a.
2. When running `make -C {path to}/sim on the tutorial 2a `sim` directory that gets created, I observe that there is a dependence on a boost utility. Specifically:
    ```bash
     Vitis/2023.2/aietools/bin//../include/adf/wrapper/logging.h:53:10: fatal error: boost/log/trivial.hpp: No such file or directory
     #include <boost/log/trivial.hpp>
    ```
 Does the [Getting started doc](https://github.com/Xilinx/mlir-aie/blob/abbcb9a05a37e33217db8a0771594b81e3c4b95f/docs/Building.md) need to be updated to include a dependence on the Boost libraries?

makslevental, 2023-11-29:
> I've made that change, but now I'm seeing an error ld.lld: error: /auto/software/swtree/ubuntu22.04/x86_64/Xilinx/Vitis/2023.2/aietools/lib/lnx64.o/libcdo_driver.so is incompatible with elf64-littleaarch64

That sysroot is for cross-compiling to aarch64 (zynq?). Indeed looking at the CLI args in the screencap there's a lot of that littered in there. Is that what you're trying to do @Dinngger?

>  Does the [Getting started doc](https://github.com/Xilinx/mlir-aie/blob/abbcb9a05a37e33217db8a0771594b81e3c4b95f/docs/Building.md) need to be updated to include a dependence on the Boost libraries?

mlir-aie doesn't depend on boost - I made that change recently https://github.com/Xilinx/mlir-aie/pull/706. Vitis might (not in the sense that I don't know, but in the sense that it's beyond my control) but then Vitis also indeed ships with boost 1.72 somewhere in `Vitis/2023.1/tps` or `Vitis_HLS/2023.1/tps` (was just 5 minutes looking at that directory). 

In general, if you're not cross-compiling for a device or whatever, my recommendation is to mirror what the CI does on your local machine because CI does exercise the tutorials and reference designs https://github.com/Xilinx/mlir-aie/actions/runs/7038002835/workflow#L220-L223.

cabreraam, 2023-11-29:
> That sysroot is for cross-compiling to aarch64 (zynq?). Indeed looking at the CLI args in the screencap there's a lot of that littered in there. Is that what you're trying to do @Dinngger?
In the tutorials, I think the default `Makefile` rule is to cross-compile a host binary for a Versal ACAP platform (VCK190 specifically?). So, I think they're just trying to do whatever the tutorials are trying to do.
 
> mlir-aie doesn't depend on boost - I made that change recently #706. Vitis might (not in the sense that I don't know, but in the sense that it's beyond my control) but then Vitis also indeed ships [...]

I will look into this. Thanks for the pointers!

makslevental, 2023-11-29:
Boost is @ `Vitis/2023.2/aietools/tps/boost_1_72_`

cabreraam, 2023-11-29:
@makslevental Do you have any opinion on the linker error, specifically `lcdo_driver.so`?

makslevental, 2023-11-30:
Funny I just hit the boost error too:

```
******************** TEST 'AIE :: unit_tests/aie/29_aie2_nd_dma_even_odd/aie.mlir' FAILED ********************
Script:
--
: 'RUN: at line 27';   /usr/bin/python3.10 /mlir-aie/build/bin/aiecc.py --aiesim --xchesscc --xbridge  --host-target=x86_64-unknown-linux-gnu /mlir-aie/test/unit_tests/aie/29_aie2_nd_dma_even_odd/aie.mlir -I/mlir-aie/build/runtime_lib/x86_64/test_lib/include  /mlir-aie/test/unit_tests/aie/29_aie2_nd_dma_even_odd/test.cpp -o test -L/mlir-aie/build/runtime_lib/x86_64/test_lib/lib -ltest_lib
: 'RUN: at line 28';   echo ./test.elf
: 'RUN: at line 29';   aie.mlir.prj/aiesim.sh | /mlir/bin/FileCheck /mlir-aie/test/unit_tests/aie/29_aie2_nd_dma_even_odd/aie.mlir
--
Exit Code: 1

Command Output (stdout):
--
['-I/mlir-aie/build/runtime_lib/x86_64/test_lib/include', '/mlir-aie/test/unit_tests/aie/29_aie2_nd_dma_even_odd/test.cpp', '-o', 'test', '-L/mlir-aie/build/runtime_lib/x86_64/test_lib/lib', '-ltest_lib']
Error encountered while running: clang++ -O2 -fuse-ld=lld -shared -o aie.mlir.prj/sim/ps/ps.so /mlir-aie/build/aie_runtime_lib/AIE2/aiesim/genwrapper_for_ps.cpp -D__AIEARCH__=20 -I/mlir-aie/build/runtime_lib/x86_64/test_lib/include /mlir-aie/test/unit_tests/aie/29_aie2_nd_dma_even_odd/test.cpp -L/mlir-aie/build/runtime_lib/x86_64/test_lib/lib -ltest_lib -fPIC -flto -fpermissive -DAIE_OPTION_SCALAR_FLOAT_ON_VECTOR -Wno-deprecated-declarations -Wno-enum-constexpr-conversion -Wno-format-security -DSC_INCLUDE_DYNAMIC_PROCESSES -D__AIESIM__ -D__PS_INIT_AIE__ -Og -Dmain(...)=ps_main(...) -Iaie.mlir.prj -I/opt/Xilinx/Vitis/2023.2/aietools/include -I/opt/Xilinx/Vitis/2023.2/aietools/include/drivers/aiengine -I/opt/Xilinx/Vitis/2023.2/aietools/data/osci_systemc/include -I/opt/Xilinx/Vitis/2023.2/aietools/include/xtlm/include -I/opt/Xilinx/Vitis/2023.2/aietools/include/common_cpp/common_cpp_v1_0/include -I/mlir-aie/build/runtime_lib/x86_64/test_lib/include /mlir-aie/build/runtime_lib/x86_64/test_lib/lib/libmemory_allocator_sim_aie.a -L/opt/Xilinx/Vitis/2023.2/aietools/lib/lnx64.o -L/opt/Xilinx/Vitis/2023.2/aietools/data/osci_systemc/lib/lnx64 -Wl,--as-needed -lxioutils -lxaiengine -ladf_api -lsystemc -lxtlm -flto
 AIE Compilation: ━━━━━━━━━                    33% -:--:-- 0:00:05 1/3 4 Workers

--
Command Output (stderr):
--
warning: unknown warning option '-Wno-enum-constexpr-conversion' [-Wunknown-warning-option]
In file included from /mlir-aie/build/aie_runtime_lib/AIE2/aiesim/genwrapper_for_ps.cpp:1:
In file included from /opt/Xilinx/Vitis/2023.2/aietools/include/adf/wrapper/wrapper.h:5:
In file included from /opt/Xilinx/Vitis/2023.2/aietools/include/adf/wrapper/me_ip_block.h:53:
In file included from /opt/Xilinx/Vitis/2023.2/aietools/include/adf/wrapper/me_pl_channels.h:50:
/opt/Xilinx/Vitis/2023.2/aietools/include/adf/wrapper/logging.h:53:10: fatal error: 'boost/log/trivial.hpp' file not found
#include <boost/log/trivial.hpp>
         ^~~~~~~~~~~~~~~~~~~~~~~
1 warning and 1 error generated.
warning: unknown warning option '-Wno-enum-constexpr-conversion' [-Wunknown-warning-option]
1 warning generated.
```

Pretty lame :shrug:

jbelot, 2023-12-01:
Hi! 👋
I encountered exactly the same issue as @Dinngger on Ubuntu 22.04 with Vitis 2023.2.
I applied the changes of @cabreraam:
> I think this gets fixed if you edit the libstdc++ version [here](https://github.com/Xilinx/mlir-aie/blob/6f4bba37689deb3837b6d599703cdfaa02de2d36/tutorials/makefile-common#L20-L23) to reflect Vitis 2023.2.

And I do have the next issue mentioned here:
> I've made that change, but now I'm seeing an error ld.lld: error: /auto/software/swtree/ubuntu22.04/x86_64/Xilinx/Vitis/2023.2/aietools/lib/lnx64.o/libcdo_driver.so is incompatible with elf64-littleaarch64

What is different is for me is this workaround:
> A dirty solution is to remove the libcdo_driver.so dependence by removing the -lcdo_driver option [here](https://github.com/Xilinx/mlir-aie/blob/abbcb9a05a37e33217db8a0771594b81e3c4b95f/python/compiler/aiecc/main.py#L520C57-L520C57). The build completes successfully when I do that.

For me the build does not complete, and I still have the error:
`Workersld.lld: error: /MyVitisPath/2023.2/aietools/lib/lnx64.o/libcdo_driver.so is incompatible with elf64-littleaarch64`

makslevental, 2023-12-01:
@jbelot (or anyone else): can you please definitively answer for me the simple question: are you **intentionally** trying to cross-compile for aarch64/PetaLinux/zynq/whatever?

stephenneuendorffer, 2023-12-01:
The tutorial is intended to cross compile, but cross-compilation shouldn't involve libcdo_driver.so.  So there are paths crossed somewhere in aiecc.py I think.

jgmelber, 2023-12-04:
The tutorials target aiesimulator where the host is ARM, so you do need to cross compile here as far as I know. @jackl-xilinx is this correct?

jbelot, 2023-12-05:
> Hi! 👋 I encountered exactly the same issue as @Dinngger on Ubuntu 22.04 with Vitis 2023.2. I applied the changes of @cabreraam:
> 
> > I think this gets fixed if you edit the libstdc++ version [here](https://github.com/Xilinx/mlir-aie/blob/6f4bba37689deb3837b6d599703cdfaa02de2d36/tutorials/makefile-common#L20-L23) to reflect Vitis 2023.2.
> 
> And I do have the next error mentioned here:
> 
> > I've made that change, but now I'm seeing an error ld.lld: error: /auto/software/swtree/ubuntu22.04/x86_64/Xilinx/Vitis/2023.2/aietools/lib/lnx64.o/libcdo_driver.so is incompatible with elf64-littleaarch64
> 
> What is different is for me is this workaround:
> 
> > A dirty solution is to remove the libcdo_driver.so dependence by removing the -lcdo_driver option [here](https://github.com/Xilinx/mlir-aie/blob/abbcb9a05a37e33217db8a0771594b81e3c4b95f/python/compiler/aiecc/main.py#L520C57-L520C57). The build completes successfully when I do that.
> 
> For me the build does not complete, and I still have the error: `Workersld.lld: error: /MyVitisPath/2023.2/aietools/lib/lnx64.o/libcdo_driver.so is incompatible with elf64-littleaarch64`

OK my bad, even though I don't get why it worked, I had to re-build mlir-aie with: 
```
rm -rf build/ install/
./utils/build-mlir-aie.sh llvm/build
```
after doing the changes on `aiecc/main.py` file.
The build complete successfully now for me, sorry for the noise.

closed 2023-12-05 by stephenneuendorffer

makslevental, 2023-12-05:
@jbelot no worries - come back any time :)

commit 7c93880b (2023-12-05): Host code shouldn't link with the cdo_driver

This should have been removed with prior refactoring of the CDO
backend.

Fixes #799