Xilinx/mlir-aie #839: aiecc should change working directory when running chesscc

fifield, 2023-12-07:
Currently chess runs in whatever working directory aiecc was invoked from, causing it to pollute that directory. Instead aiecc should change it's working directory to the `tmpdir` before running chess, then copy any user outputs (elfs, xclbin, ...) to the appropriate place.

stephenneuendorffer, 2023-12-07:
I suspect that chess leaves it's droppings in the same directory that the outputs appear in.  So it might be better to not change the current directory, and just create temporary outputs in the tmpdir, then copy them where we want them.  the key idea is to reduce the presence of unnecessary files in the file system.

keryell, 2023-12-07:
I would suggest a unique directory under `tmpdir`  otherwise we will fall into a race-condition nightmare.