Xilinx/mlir-aie #853: Update ODS to add 'name' member to xilinx::AIE::AIEObjectFifoSubviewType and other types

frederik-h, 2023-12-11:
It seems that some ODS information needs to be update because of recent LLVM changes. 
I have tried to compile the [nod.ai shark runtime](https://github.com/nod-ai/SRT) with the [iree-amd-aie](https://github.com/nod-ai/iree-amd-aie) plugin enabled. The latter uses this project. I got the following error:

```
In file included from /home/fharwath/wd/shark/SRT/third_party/llvm-project/mlir/include/mlir/IR/Types.h:12:
/home/fharwath/wd/shark/SRT/third_party/llvm-project/mlir/include/mlir/IR/TypeSupport.h:54:28: error: no member named 'name' in 'xilinx::AIE::AIEObjectFifoSubviewType'
   54 |                         T::name);
```

This is caused by LLVM commit [3dbac2c007c1 [mlir] Expose type and attribute names in the MLIRContext and abstract type/attr classes
](https://github.com/llvm/llvm-project/commit/3dbac2c007c114a720300d2a4d79abe9ca1351e7).

stephenneuendorffer, 2023-12-11:
Are you sure your tree is up to date?  Because: [#809](https://github.com/Xilinx/mlir-aie/pull/809) should have fixed this?

frederik-h, 2023-12-12:
> Are you sure your tree is up to date? Because: [#809](https://github.com/Xilinx/mlir-aie/pull/809) should have fixed this?

Thanks for the pointer to the commit and sorry for the confusion. The iree-amd-aie project has been using an older version of this project and I did not notice that you have already solved the problem. They have made various changes to their mlir-air and mlir-aie submodule and now it includes the fix and I can build it.

closed 2023-12-12 by frederik-h