Xilinx/mlir-aie #886: aiecc unused `--no-compile-host`/`--compile-host` flags

makslevental, 2023-12-22:
aiecc has these args

https://github.com/Xilinx/mlir-aie/blob/main/python/compiler/aiecc/cl_arguments.py#L129-L142

```python
parser.add_argument(
    "--compile-host",
    dest="compile_host",
    default=not host_disable_compile,
    action="store_true",
    help="Enable compiling of the host program",
)
parser.add_argument(
    "--no-compile-host",
    dest="compile_host",
    default=host_disable_compile,
    action="store_false",
    help="Disable compiling of the host program",
)
```
but they're not used anywhere. They were lost here [fe1bc63](https://github.com/Xilinx/mlir-aie/commit/fe1bc637685785dce6f7cca7c94954d84e1f7fd1#diff-bf6a8b9f704122777f8f94813c3b4de859ca444fcceca6e11ad059a5dc175db6L151). Note sure if that was intentional or what but `--no-compile-host` can be found in various places in `reference_designs/` (so makes me wonder if it should be restored?).

closed 2025-03-25 by jgmelber

fifield, 2025-03-25:
I guess I inadvertently restored it in recent PR(s)?

jgmelber, 2025-03-25:
> I guess I inadvertently restored it in recent PR(s)?

I closed this as it seemed you restored it.