Xilinx/mlir-aie #918: LLVM/MLIR compatible version for successful compilation?

medbzkst, 2024-01-08:
I have been trying to get MLIR-AIE/AIR up and running for a thousand times. I am getting this error over the AIEVec IR build. 
`/home/bouazim/mlir-air/mlir-aie/include/aie/Dialect/AIEVec/IR/AIEVecOps.td:453:11: error: Variable not defined: 'InferTypeOpAdaptor'`. This error occurs on multiple locations in the same file (seems to be related to TableGen).

I guess this is an LLVM/MLIR compatibility issue. Can someone please point to the LLVM version that can make this version of MLIR-AIE/R installable?

Thanks!

makslevental, 2024-01-08:
The hash is here https://github.com/Xilinx/mlir-aie/blob/main/utils/clone-llvm.sh

For the record you can also download distros of LLVM that match that hash here https://github.com/Xilinx/mlir-aie/releases/tag/mlir-distro (a .whl is just a zip).

Also for the record InferTypeOpInterface is quite old so you might be having some other issue compiling.

medbzkst, 2024-01-08:
Thank you for the references. Now, a brand new error has appeared 
```CMake Error:
  Running

   '/home/user/ninja/build-cmake/ninja' '-C' '/home/user/mlir-air/mlir-aie/build' '-t' 'recompact'

  failed with:

   ninja: error: build.ninja:8556: multiple rules generate python/aie/extras/meta.py
```
I do not see how to handle these conflicts actually.

Also the InferTypeOpInterface is part of the core MLIR-AIE [code](https://github.com/Xilinx/mlir-aie/blob/cb232a43383ef3b8efd8b408545c9b74885578ad/include/aie/Dialect/AIEVec/IR/AIEVecOps.td#L453) . Does that say that the code is not compatible with the LLVM underlying commit?

Thanks!

makslevental, 2024-01-08:
> Thank you for the references. Now, a brand new error has appeared
> 
> ```cmake
>   Running
> 
>    '/home/user/ninja/build-cmake/ninja' '-C' '/home/user/mlir-air/mlir-aie/build' '-t' 'recompact'
> 
>   failed with:
> 
>    ninja: error: build.ninja:8556: multiple rules generate python/aie/extras/meta.py
> ```
> 

That means you have the right version of LLVM but the wrong version of AIE (you need to update AIE).

> Also the InferTypeOpInterface is part of the core MLIR-AIE [code](https://github.com/Xilinx/mlir-aie/blob/cb232a43383ef3b8efd8b408545c9b74885578ad/include/aie/Dialect/AIEVec/IR/AIEVecOps.td#L453) . Does that say that the code is not compatible with the LLVM underlying commit?

[InferTypeOpInterface](https://github.com/llvm/llvm-project/blob/5cbf74b012c10e9cc841a27cd5d7335e556f47dd/mlir/include/mlir/Interfaces/InferTypeOpInterface.td) is from upstream (`InferTypeOpAdaptor` is just a wrapper around the [interface](https://github.com/llvm/llvm-project/blob/5cbf74b012c10e9cc841a27cd5d7335e556f47dd/mlir/include/mlir/Interfaces/InferTypeOpInterface.td#L225)).

medbzkst, 2024-01-08:
> That means you have the right version of LLVM but the wrong version of AIE (you need to update AIE).

Thank you @makslevental . I suppose you mean MLIR-AIE by saying AIE. Probably, I am not seeing a point here.

So, the current version on the **master branch** of the MLIR-AIE repo should be the one to use, right? I am using it.

And the LLVM commit that is cloned using the cloning script (the one you referred to) is the one that is compatible with the MLIR-AIE version in use, right? Because I tried it too.

And if I want to alleviate the LLVM setup process, I should just proceed to the distro (that you referred to). Is that right? Because I tried to use it as well.

Also let me mention that I have been trying to follow both [MLIR-AIE documentation](https://xilinx.github.io/mlir-aie/Building.html) and [MLIR-AIR documentation](https://xilinx.github.io/mlir-air/building.html) and both did not lead anywhere (MLIR-AIR documentation is even not accurate but that's another story).

Any thoughts about this?

Thanks!

makslevental, 2024-01-08:
> > That means you have the right version of LLVM but the wrong version of AIE (you need to update AIE).
> 
> Thank you @makslevental . I suppose you mean MLIR-AIE by saying AIE. Probably, I am not seeing a point here.

I'm telling you that that error is because you are **not** in fact up to date on MLIR-AIE because I recognize it as a [recent transition](https://github.com/Xilinx/mlir-aie/pull/878) that we went through (going from using our own `python/aie/extras/meta.py` to upstream `python/aie/extras/meta.py`).

> So, the current version on the **master branch** of the MLIR-AIE repo should be the one to use, right? I am using it.
>
> And the LLVM commit that is cloned using the cloning script (the one you [...]

All of this is correct but I'm but I ask you to look at little more closely at your environment because we have in CI ~50 tests that exercise these paths and they are all (as of today) [in sync](https://github.com/Xilinx/mlir-aie/pull/913/checks).

> Also let me mention that I have been trying to follow both [MLIR-AIE documentation](https://xilinx.github.io/mlir-aie/Building.html) and [MLIR-AIR [...]

Yes the docs are not great that's true and it's unfortunate but this is unfortunately unavoidable for a project like this that moves basically at the speed of light. If you would like realtime debugging help feel free to ping me at the same name on the MLIR discord https://discord.gg/rhKa7TAE.

medbzkst, 2024-01-09:
Thank you, @makslevental, for the help and the reference to the right spot! It compiles now.

closed 2024-01-09 by medbzkst