Xilinx/mlir-aie #926: Fix github.io link and generation of aie dialect (https://xilinx.github.io/mlir-aie/AIEDialect.html)

jackl-xilinx, 2024-01-11:
While https://xilinx.github.io/mlir-aie/AIEDialect.html is invalid, https://xilinx.github.io/mlir-aie/AIE.html is valid. I think maybe that's because the AIEDialect.td changed to AIE.td or something like that but the places that reference it (like the [sidebar](https://github.com/Xilinx/mlir-aie/blob/e10cb3043807a9858ff75289627d1bb97f2855c0/docs/_layouts/default.html#L30) and [README](https://github.com/Xilinx/mlir-aie/blob/e10cb3043807a9858ff75289627d1bb97f2855c0/docs/README.md)) didn't change along with it. If you look at the new location, you'll see it's mostly empty and I think it has to do with the how AIE.td was changed where included *.td files were removed around [this commit](https://github.com/Xilinx/mlir-aie/commit/2270ac4fd3fbd4dd8259d6c093b32d118e3a4fa2). You can see here that the commit removed the includes for things like AIEInterfaces.td and replaced it with OpBase.td which I don't see the file for. I think the included td files add the doc definitions.

makslevental, 2024-01-23:
Docs are fixed https://xilinx.github.io/mlir-aie/AIE.html

closed 2024-01-23 by makslevental

jackl-xilinx, 2024-01-26:
Awesome! Thanks @makslevental