Xilinx/mlir-aie #942: Failure to compile due to invalid implicit type conversion in aie-rt

jsetoain, 2024-01-17:
When compiling with `AIE_ENABLE_CDO_DIRECT=ON` (default behaviour with Vitis tools configured, apparently), I see the following compilation error in the submodule `aie-rt`:

```
[38/401] Building C object runtime_lib/CMakeFiles/xaienginecdo_static.dir/xaiengine/aie-rt/driver/src/global/xaiemlgbl_reginit.c.o
FAILED: runtime_lib/CMakeFiles/xaienginecdo_static.dir/xaiengine/aie-rt/driver/src/global/xaiemlgbl_reginit.c.o
/<home>/.local/bin/clang -D_DEBUG -D_GLIBCXX_ASSERTIONS -D__AIECDO__ -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -I<project dir>/llvm/install/include -I<project dir>/aie/include -I<project dir>/build/include -I<project dir>/aie/runtime_lib/xaiengine/aie-rt/driver/src -I<project dir>/aie/runtime_lib/xaiengine/aie-rt/driver/src/common -I<project dir>/aie/runtime_lib/xaiengine/aie-rt/driver/src/core -I<project dir>/aie/runtime_lib/xaiengine/aie-rt/driver/src/device -I<project dir>/aie/runtime_lib/xaiengine/aie-rt/driver/src/dma -I<project dir>/aie/runtime_lib/xaiengine/aie-rt/driver/src/events -I<project dir>/aie/runtime_lib/xaiengine/aie-rt/driver/src/global -I<project dir>/aie/runtime_lib/xaiengine/aie-rt/driver/src/interrupt -I<project dir>/aie/runtime_lib/xaiengine/aie-rt/driver/src/io_backend -I<project dir>/aie/runtime_lib/xaiengine/aie-rt/driver/src/io_backend/ext -I<project dir>/aie/runtime_lib/xaiengine/aie-rt/driver/src/io_backend/privilege -I<project dir>/aie/runtime_lib/xaiengine/aie-rt/driver/src/lite -I<project dir>/aie/runtime_lib/xaiengine/aie-rt/driver/src/locks -I<project dir>/aie/runtime_lib/xaiengine/aie-rt/driver/src/memory -I<project dir>/aie/runtime_lib/xaiengine/aie-rt/driver/src/npi -I<project dir>/aie/runtime_lib/xaiengine/aie-rt/driver/src/perfcnt -I<project dir>/aie/runtime_lib/xaiengine/aie-rt/driver/src/pl -I<project dir>/aie/runtime_lib/xaiengine/aie-rt/driver/src/pm -I<project dir>/aie/runtime_lib/xaiengine/aie-rt/driver/src/rsc -I<project dir>/aie/runtime_lib/xaiengine/aie-rt/driver/src/stream_switch -I<project dir>/aie/runtime_lib/xaiengine/aie-rt/driver/src/timer -I<project dir>/aie/runtime_lib/xaiengine/aie-rt/driver/src/trace -I<project dir>/aie/runtime_lib/xaiengine/aie-rt/driver/src/util -fPIC -fno-semantic-interposition -Werror=date-time -Werror=unguarded-availability-new -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wmissing-field-initializers -Wimplicit-fallthrough -Wcovered-switch-default -Wstring-conversion -Wmisleading-indentation -Wctad-maybe-unsupported -fdiagnostics-color -Werror=sign-compare -Werror=return-type -Werror=implicit-function-declaration -Werror=mismatched-tags -g   -D_DEBUG -D_GLIBCXX_ASSERTIONS -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -fPIC -Wno-gnu-designator -MD -MT runtime_lib/CMakeFiles/xaienginecdo_static.dir/xaiengine/aie-rt/driver/src/global/xaiemlgbl_reginit.c.o -MF runtime_lib/CMakeFiles/xaienginecdo_static.dir/xaiengine/aie-rt/driver/src/global/xaiemlgbl_reginit.c.o.d -o runtime_lib/CMakeFiles/xaienginecdo_static.dir/xaiengine/aie-rt/driver/src/global/xaiemlgbl_reginit.c.o -c <project dir>/aie/runtime_lib/xaiengine/aie-rt/driver/src/global/xaiemlgbl_reginit.c
<project dir>/aie/runtime_lib/xaiengine/aie-rt/driver/src/global/xaiemlgbl_reginit.c:2290:14: error: incompatible function pointer types initializing 'AieRC (*)(XAie_DevInst *, u8, u8)' (aka 'AieRC (*)(XAie_DevInst *, unsigned char, unsigned char)') with an expression of type 'AieRC (XAie_DevInst *, u32, u32)' (aka 'AieRC (XAie_DevInst *, unsigned int, unsigned int)') [-Wincompatible-function-pointer-types]
 2290 |         .RstShims = _XAieMl_RstShims,
      |                     ^~~~~~~~~~~~~~~~
1 error generated.
```

makslevental, 2024-01-17:
Yea so this is legit; the purported assignment is

```cpp
static const XAie_ShimRstMod AieMlShimTileRst =
{
	.RegOff = 0,
	.RstCntr = {0},
	.RstShims = _XAieMl_RstShims,
};
```
which tries to assign `RstShims` which requires

```cpp
typedef struct {
	u32 RegOff;
	XAie_RegFldAttr RstCntr;
	AieRC (*RstShims)(XAie_DevInst *DevInst, u8 StartCol, u8 NumCols);
} XAie_ShimRstMod;
```

in [aie-rt/driver/src/global/xaiegbl_regdef.h#L562](https://github.com/stephenneuendorffer/aie-rt/blob/phoenix_v2023.2/driver/src/global/xaiegbl_regdef.h#L562). Compared with 

```cpp
AieRC _XAieMl_RstShims(XAie_DevInst *DevInst, u32 StartCol, u32 NumCols)
{
```

in [aie-rt/driver/src/pm/xaie_reset_aieml.c#L48](https://github.com/stephenneuendorffer/aie-rt/blob/phoenix_v2023.2/driver/src/pm/xaie_reset_aieml.c#L48).

I'll send a fix to @stephenneuendorffer soon but what I'm actually curious about is how/why you're failing to compile when no one else is. Do you have `-Werror=incompatible-function-pointer-types` or something to that effect?

jsetoain, 2024-01-17:
> Yea so this is legit; the purported assignment is
> 
> ```c++
> static const XAie_ShimRstMod AieMlShimTileRst =
> {
> 	.RegOff = 0,
> 	.RstCntr = {0},
> 	.RstShims = _XAieMl_RstShims,
> };
> ```
> 
> which tries to assign `RstShims` which requires
> 
> ```c++
> typedef struct {
> 	u32 RegOff;
> 	XAie_RegFldAttr RstCntr;
> 	AieRC (*RstShims)(XAie_DevInst *DevInst, u8 StartCol, u8 NumCols);
> } XAie_ShimRstMod;
> ```
> 
> in [aie-rt/driver/src/global/xaiegbl_regdef.h#L562](https://github.com/stephenneuendorffer/aie-rt/blob/phoenix_v2023.2/driver/src/global/xaiegbl_regdef.h#L562). Compared with
> 
> ```c++
> AieRC _XAieMl_RstShims(XAie_DevInst *DevInst, u32 StartCol, u32 NumCols)
> {
> ```
> 
> in [aie-rt/driver/src/pm/xaie_reset_aieml.c#L48](https://github.com/stephenneuendorffer/aie-rt/blob/phoenix_v2023.2/driver/src/pm/xaie_reset_aieml.c#L48).
> 
> I'll send a fix to @stephenneuendorffer soon but what I'm actually curious about is how/why you're failing to compile when no one else is. Do you have `-Werror=incompatible-function-pointer-types` or something to that effect?

We've clarified this in private, but for documentation purposes, this mismatch is a warning in gcc 10 and clang 12, but it's an error in clang 17, which is the one I'm using.

makslevental, 2024-03-05:
Fixed in https://github.com/Xilinx/mlir-aie/pull/1068.

closed 2024-03-05 by makslevental