Xilinx/mlir-aie #950: Ping-pong with neighbouring tiles lowers to using DMA when broadcasting

medbzkst, 2024-01-23:
I was running a simple example where I implemented a ping-pong buffer that directly broadcasts to the neighbouring tiles that the underlying tile connects to.

```
module @multi_ping_pong {
  aie.device(xcvc1902) {
    %tile_0_2 = aie.tile(0, 2)
    %tile_1_1 = aie.tile(1, 1)
    %tile_1_2 = aie.tile(1, 2)
    %tile_1_3 = aie.tile(1, 3)
    aie.objectfifo @objfifo12(%tile_1_2, {%tile_0_2, %tile_1_1, %tile_1_3}, 2 : i32) : !aie.objectfifo<memref<256xi32>>
    func.func private @doSomething(memref<256xi32>)
    %core_1_2 = aie.core(%tile_1_2) {
      %0 = aie.objectfifo.acquire @objfifo12(Produce, 1) : !aie.objectfifosubview<memref<256xi32>>
      %1 = aie.objectfifo.subview.access %0[0] : !aie.objectfifosubview<memref<256xi32>> -> memref<256xi32>
      func.call @doSomething(%1) : (memref<256xi32>) -> ()
      aie.objectfifo.release @objfifo12(Produce, 1)
      aie.end
    }
  }
}
```

As given in the code, the tile (1,2) has an object FIFO that implements a ping-pong buffer with the tiles that it directly connects to, i.e. tile (0,2), (1,1), (1,3). However, when I lower this code using
`aie-opt --aie-canonicalize-device aie.mlir | aie-opt --aie-objectFifo-stateful-transform`
I see that there are tile DMA instantiations 
```
module @multi_ping_pong {
  aie.device(xcvc1902) {
    memref.global "public" @objfifo12_0_cons : memref<256xi32>
    memref.global "public" @objfifo12_1_cons : memref<256xi32>
    memref.global "public" @objfifo12_2_cons : memref<256xi32>
    memref.global "public" @objfifo12 : memref<256xi32>
    %tile_0_2 = aie.tile(0, 2)
    %tile_1_1 = aie.tile(1, 1)
    %tile_1_2 = aie.tile(1, 2)
    %tile_1_3 = aie.tile(1, 3)
    %objfifo12_0_cons_buff_0 = aie.buffer(%tile_0_2) {sym_name = "objfifo12_0_cons_buff_0"} : memref<256xi32>
    %objfifo12_0_cons_buff_1 = aie.buffer(%tile_0_2) {sym_name = "objfifo12_0_cons_buff_1"} : memref<256xi32>
    %objfifo12_0_cons_lock_0 = aie.lock(%tile_0_2, 0) {init = 0 : i32, sym_name = "objfifo12_0_cons_lock_0"}
    %objfifo12_0_cons_lock_1 = aie.lock(%tile_0_2, 1) {init = 0 : i32, sym_name = "objfifo12_0_cons_lock_1"}
    %objfifo12_1_cons_buff_0 = aie.buffer(%tile_1_1) {sym_name = "objfifo12_1_cons_buff_0"} : memref<256xi32>
    %objfifo12_1_cons_buff_1 = aie.buffer(%tile_1_1) {sym_name = "objfifo12_1_cons_buff_1"} : memref<256xi32>
    %objfifo12_1_cons_lock_0 = aie.lock(%tile_1_1, 0) {init = 0 : i32, sym_name = "objfifo12_1_cons_lock_0"}
    %objfifo12_1_cons_lock_1 = aie.lock(%tile_1_1, 1) {init = 0 : i32, sym_name = "objfifo12_1_cons_lock_1"}
    %objfifo12_2_cons_buff_0 = aie.buffer(%tile_1_3) {sym_name = "objfifo12_2_cons_buff_0"} : memref<256xi32>
    %objfifo12_2_cons_buff_1 = aie.buffer(%tile_1_3) {sym_name = "objfifo12_2_cons_buff_1"} : memref<256xi32>
    %objfifo12_2_cons_lock_0 = aie.lock(%tile_1_3, 0) {init = 0 : i32, sym_name = "objfifo12_2_cons_lock_0"}
    %objfifo12_2_cons_lock_1 = aie.lock(%tile_1_3, 1) {init = 0 : i32, sym_name = "objfifo12_2_cons_lock_1"}
    %objfifo12_buff_0 = aie.buffer(%tile_1_2) {sym_name = "objfifo12_buff_0"} : memref<256xi32>
    %objfifo12_buff_1 = aie.buffer(%tile_1_2) {sym_name = "objfifo12_buff_1"} : memref<256xi32>
    %objfifo12_lock_0 = aie.lock(%tile_1_2, 0) {init = 0 : i32, sym_name = "objfifo12_lock_0"}
    %objfifo12_lock_1 = aie.lock(%tile_1_2, 1) {init = 0 : i32, sym_name = "objfifo12_lock_1"}
    aie.flow(%tile_1_2, DMA : 0, %tile_1_3, DMA : 0)
    aie.flow(%tile_1_2, DMA : 0, %tile_1_1, DMA : 0)
    aie.flow(%tile_1_2, DMA : 0, %tile_0_2, DMA : 0)
    func.func private @doSomething(memref<256xi32>)
    %core_1_2 = aie.core(%tile_1_2) {
      aie.use_lock(%objfifo12_lock_0, Acquire, 0)
      func.call @doSomething(%objfifo12_buff_0) : (memref<256xi32>) -> ()
      aie.use_lock(%objfifo12_lock_0, Release, 1)
      aie.end
    }
    %mem_1_2 = aie.mem(%tile_1_2) {
      %0 = aie.dma_start(MM2S, 0, ^bb1, ^bb3)
    ^bb1:  // 2 preds: ^bb0, ^bb2
      aie.use_lock(%objfifo12_lock_0, Acquire, 1)
      aie.dma_bd(%objfifo12_buff_0 : memref<256xi32>, 0, 256)
      aie.use_lock(%objfifo12_lock_0, Release, 0)
      aie.next_bd ^bb2
    ^bb2:  // pred: ^bb1
      aie.use_lock(%objfifo12_lock_1, Acquire, 1)
      aie.dma_bd(%objfifo12_buff_1 : memref<256xi32>, 0, 256)
      aie.use_lock(%objfifo12_lock_1, Release, 0)
      aie.next_bd ^bb1
    ^bb3:  // pred: ^bb0
      aie.end
    }
    %mem_0_2 = aie.mem(%tile_0_2) {
      %0 = aie.dma_start(S2MM, 0, ^bb1, ^bb3)
    ^bb1:  // 2 preds: ^bb0, ^bb2
      aie.use_lock(%objfifo12_0_cons_lock_0, Acquire, 0)
      aie.dma_bd(%objfifo12_0_cons_buff_0 : memref<256xi32>, 0, 256)
      aie.use_lock(%objfifo12_0_cons_lock_0, Release, 1)
      aie.next_bd ^bb2
    ^bb2:  // pred: ^bb1
      aie.use_lock(%objfifo12_0_cons_lock_1, Acquire, 0)
      aie.dma_bd(%objfifo12_0_cons_buff_1 : memref<256xi32>, 0, 256)
      aie.use_lock(%objfifo12_0_cons_lock_1, Release, 1)
      aie.next_bd ^bb1
    ^bb3:  // pred: ^bb0
      aie.end
    }
    %mem_1_1 = aie.mem(%tile_1_1) {
      %0 = aie.dma_start(S2MM, 0, ^bb1, ^bb3)
    ^bb1:  // 2 preds: ^bb0, ^bb2
      aie.use_lock(%objfifo12_1_cons_lock_0, Acquire, 0)
      aie.dma_bd(%objfifo12_1_cons_buff_0 : memref<256xi32>, 0, 256)
      aie.use_lock(%objfifo12_1_cons_lock_0, Release, 1)
      aie.next_bd ^bb2
    ^bb2:  // pred: ^bb1
      aie.use_lock(%objfifo12_1_cons_lock_1, Acquire, 0)
      aie.dma_bd(%objfifo12_1_cons_buff_1 : memref<256xi32>, 0, 256)
      aie.use_lock(%objfifo12_1_cons_lock_1, Release, 1)
      aie.next_bd ^bb1
    ^bb3:  // pred: ^bb0
      aie.end
    }
    %mem_1_3 = aie.mem(%tile_1_3) {
      %0 = aie.dma_start(S2MM, 0, ^bb1, ^bb3)
    ^bb1:  // 2 preds: ^bb0, ^bb2
      aie.use_lock(%objfifo12_2_cons_lock_0, Acquire, 0)
      aie.dma_bd(%objfifo12_2_cons_buff_0 : memref<256xi32>, 0, 256)
      aie.use_lock(%objfifo12_2_cons_lock_0, Release, 1)
      aie.next_bd ^bb2
    ^bb2:  // pred: ^bb1
      aie.use_lock(%objfifo12_2_cons_lock_1, Acquire, 0)
      aie.dma_bd(%objfifo12_2_cons_buff_1 : memref<256xi32>, 0, 256)
      aie.use_lock(%objfifo12_2_cons_lock_1, Release, 1)
      aie.next_bd ^bb1
    ^bb3:  // pred: ^bb0
      aie.end
    }
  }
}
```
And this always holds as long as the object FIFO broadcasts to more than one tile. So even when I keep only the horizontal ones, i.e. (1,1) and (1,3), it is still the same behaviour.

_**HOWEVER!!, if I keep one and only one of those tiles, e.g. (1,2) <-> (1,1) (or any other combination) as in the code below**_ 
```
module @multi_ping_pong {
  aie.device(xcvc1902) {
    %tile_1_1 = aie.tile(1, 1)
    %tile_1_2 = aie.tile(1, 2)
    aie.objectfifo @objfifo12(%tile_1_2, {%tile_1_1}, 2 : i32) : !aie.objectfifo<memref<256xi32>>
    func.func private @doSomething(memref<256xi32>)
    %core_1_2 = aie.core(%tile_1_2) {
      %0 = aie.objectfifo.acquire @objfifo12(Produce, 1) : !aie.objectfifosubview<memref<256xi32>>
      %1 = aie.objectfifo.subview.access %0[0] : !aie.objectfifosubview<memref<256xi32>> -> memref<256xi32>
      func.call @doSomething(%1) : (memref<256xi32>) -> ()
      aie.objectfifo.release @objfifo12(Produce, 1)
      aie.end
    }
    func.func private @doOtherThing(memref<256xi32>)
    %core_1_1 = aie.core(%tile_1_1) {
      %0 = aie.objectfifo.acquire @objfifo12(Consume, 1) : !aie.objectfifosubview<memref<256xi32>>
      %1 = aie.objectfifo.subview.access %0[0] : !aie.objectfifosubview<memref<256xi32>> -> memref<256xi32>
      func.call @doOtherThing(%1) : (memref<256xi32>) -> ()
      aie.objectfifo.release @objfifo12(Consume, 1)
      aie.end
    }
  }
}
```

It lowers and as expected to something without DMAs as in here

```
module @multi_ping_pong {
  aie.device(xcvc1902) {
    memref.global "public" @objfifo12 : memref<256xi32>
    %tile_1_1 = aie.tile(1, 1)
    %tile_1_2 = aie.tile(1, 2)
    %objfifo12_buff_0 = aie.buffer(%tile_1_2) {sym_name = "objfifo12_buff_0"} : memref<256xi32>
    %objfifo12_buff_1 = aie.buffer(%tile_1_2) {sym_name = "objfifo12_buff_1"} : memref<256xi32>
    %objfifo12_lock_0 = aie.lock(%tile_1_2, 0) {init = 0 : i32, sym_name = "objfifo12_lock_0"}
    %objfifo12_lock_1 = aie.lock(%tile_1_2, 1) {init = 0 : i32, sym_name = "objfifo12_lock_1"}
    func.func private @doSomething(memref<256xi32>)
    %core_1_2 = aie.core(%tile_1_2) {
      aie.use_lock(%objfifo12_lock_0, Acquire, 0)
      func.call @doSomething(%objfifo12_buff_0) : (memref<256xi32>) -> ()
      aie.use_lock(%objfifo12_lock_0, Release, 1)
      aie.end
    }
    func.func private @doOtherThing(memref<256xi32>)
    %core_1_1 = aie.core(%tile_1_1) {
      aie.use_lock(%objfifo12_lock_0, Acquire, 1)
      func.call @doOtherThing(%objfifo12_buff_0) : (memref<256xi32>) -> ()
      aie.use_lock(%objfifo12_lock_0, Release, 0)
      aie.end
    }
  }
}
```

So, what is the problem with double buffering with multiple tiles and why the use DMAs if they are neighbours? and why that DMA if it is more than one and only one connection?

AndraBisca, 2024-01-23:
Hello @medbzkst and thank you for your question!

As you have pointed out, the lowering of the object FIFO is different depending on whether there is broadcast or not. In 1-to-1 connections, it will either lower to using the shared memory module, or to using DMAs if there is none (note that two tiles can be neighbors without having access to a shared memory module between them, this can happen depending on where the tiles are placed relative to odd and even rows).

Then, regarding the broadcast, you are right that the lowering will always use DMAs no matter where the consumer tiles are placed relative to the producer tile. Historically, this decision was made because, in the AIE1 architecture, the locks did not support more than two actors at a time. So in the case of a broadcast where some of the consumers had access to a shared memory module, while others did not, it was difficult to synchronize the locks of the objects between the producer, the neighboring consumers, and the DMAs that would send data to non-neighboring consumers. We have yet to look into whether the locks in the AIE2 architecture could support a hybrid lowering (using both shared and non-shared memory), but it is quite likely.

Another reason for the decision to always go through DMAs in a broadcast was to leverage the AXI-stream interconnect architecture and buffering to synchronize the data being sent from a single producer to multiple consumers, instead of handling it explicitly using locks within the core code.

I have been working on putting together a more complete, single documentation resource for the object FIFO primitive with all of the different scenarios and lowerings it supports, but it is not ready yet. If you would like to directly add onto existing documentation, please feel free to make a PR! 

I hope this helps. Please let us know if anything is still unclear!

medbzkst, 2024-01-24:
@AndraBisca, I can't thank you enough for this explanation. Cristal clear!

closed 2024-01-24 by medbzkst