Xilinx/mlir-aie #962: Simulation fails for Tutorials

tanvirarafin, 2024-01-29:
The simulation seems broken for the tutorials.

`aiecc.py` does not generate the `sim` directory inside `aie.mlir.prj/` by default with the `make` command. `make sim` command fails in the AIE Compilation step with `fatal error: 'boost/log/trivial.hpp' file not found` error. 

Output of `make sim` on tutorial 2b
```
make sim
aiecc.py -j4 --xchesscc --xbridge --aiesim aie.mlir -I/home/marafin/mlir-aie_2/install/runtime_lib/x86_64/test_lib/include -L/home/marafin/mlir-aie_2/install/runtime_lib/x86_64/test_lib/lib -ltest_lib ./test.cpp
Found Vitis at /tools/Xilinx/Vitis/2023.2
 MLIR compilation: ━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% -:--:-- 0:00:03 0/1 1 Workerwarning: overriding the module target triple with pdarch-unknown-unknown-elf [-Woverride-module]
1 warning generated.
 MLIR compilation: ━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% -:--:-- 0:00:03 0/1 1 WorkerWarning in "../../../../../../tools/Xilinx/Vitis/2023.2/aietools/data/versal_prod/lib/me_chess.h", line 647, column 131: ignoring attribute [[deprecated]] on class declaration
 MLIR compilation: ━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% -:--:-- 0:00:05 0/1 1 WorkerWarning in "../../../../../../tools/Xilinx/Vitis/2023.2/aietools/data/versal_prod/lib/me_common.h", line 63, column 132: ignoring attribute [[deprecated]] on class declaration
Warning in "../../../../../../tools/Xilinx/Vitis/2023.2/aietools/data/versal_prod/lib/me_common.h", line 96, column 131: ignoring attribute [[deprecated]] on class declaration
 AIE Compilation: ━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% -:--:-- 0:00:00 0/2 4 Workers['-I/home/marafin/mlir-aie_2/install/runtime_lib/x86_64/test_lib/include', '-L/home/marafin/mlir-aie_2/install/runtime_lib/x86_64/test_lib/lib', '-ltest_lib', './test.cpp']
warning: unknown warning option '-Wno-enum-constexpr-conversion' [-Wunknown-warning-option]
 AIE Compilation: ━━━━━━━━━━━━━╸━━━━━━━━━━━━━  50% -:--:-- 0:00:01 1/2 4 WorkersIn file included from /home/marafin/mlir-aie_2/install/aie_runtime_lib/AIE/aiesim/genwrapper_for_ps.cpp:1:
In file included from /tools/Xilinx/Vitis/2023.2/aietools/include/adf/wrapper/wrapper.h:5:
In file included from /tools/Xilinx/Vitis/2023.2/aietools/include/adf/wrapper/me_ip_block.h:53:
In file included from /tools/Xilinx/Vitis/2023.2/aietools/include/adf/wrapper/me_pl_channels.h:50:
/tools/Xilinx/Vitis/2023.2/aietools/include/adf/wrapper/logging.h:53:10: fatal error: 'boost/log/trivial.hpp' file not found
#include <boost/log/trivial.hpp>
         ^~~~~~~~~~~~~~~~~~~~~~~
 AIE Compilation: ━━━━━━━━━━━━━╸━━━━━━━━━━━━━  50% -:--:-- 0:00:01 1/2 4 Workers1 warning and 1 error generated.
warning: unknown warning option '-Wno-enum-constexpr-conversion' [-Wunknown-warning-option]
 AIE Compilation: ━━━━━━━━━━━━━╸━━━━━━━━━━━━━  50% -:--:-- 0:00:01 1/2 4 Workers1 warning generated.
Error encountered while running: clang++ -O2 -fuse-ld=lld -shared -o /home/marafin/mlir-aie/tutorials/tutorial-2/tutorial-2a/aie.mlir.prj/sim/ps/ps.so /home/marafin/mlir-aie_2/install/aie_runtime_lib/AIE/aiesim/genwrapper_for_ps.cpp -D__AIEARCH__=10 -I/home/marafin/mlir-aie_2/install/runtime_lib/x86_64/test_lib/include -L/home/marafin/mlir-aie_2/install/runtime_lib/x86_64/test_lib/lib -ltest_lib ./test.cpp -fPIC -flto -fpermissive -DAIE_OPTION_SCALAR_FLOAT_ON_VECTOR -Wno-deprecated-declarations -Wno-enum-constexpr-conversion -Wno-format-security -DSC_INCLUDE_DYNAMIC_PROCESSES -D__AIESIM__ -D__PS_INIT_AIE__ -Og -Dmain(...)=ps_main(...) -I/home/marafin/mlir-aie/tutorials/tutorial-2/tutorial-2a/aie.mlir.prj -I/tools/Xilinx/Vitis/2023.2/aietools/include -I/tools/Xilinx/Vitis/2023.2/aietools/include/drivers/aiengine -I/tools/Xilinx/Vitis/2023.2/aietools/data/osci_systemc/include -I/tools/Xilinx/Vitis/2023.2/aietools/include/xtlm/include -I/tools/Xilinx/Vitis/2023.2/aietools/include/common_cpp/common_cpp_v1_0/include -I/home/marafin/mlir-aie_2/install/runtime_lib/x86_64/test_lib/include /home/marafin/mlir-aie_2/install/runtime_lib/x86_64/test_lib/lib/libmemory_allocator_sim_aie.a -L/tools/Xilinx/Vitis/2023.2/aietools/lib/lnx64.o -L/tools/Xilinx/Vitis/2023.2/aietools/data/osci_systemc/lib/lnx64 -Wl,--as-needed -lxioutils -lxaiengine -ladf_api -lsystemc -lxtlm -flto
 AIE Compilation: ━━━━━━━━━━━━━╸━━━━━━━━━━━━━  50% -:--:-- 0:00:01 1/2 4 Workers

```
`

medbzkst, 2024-01-30:
Did you try installing the [Boost](https://www.boost.org/) library?

tanvirarafin, 2024-01-30:
Now, `make sim` fails with a linker error as below. However, a `sim` directory is created inside `aie.mlir.prj`. 

```
(sandbox) marafin@ubuntu:~/mlir-aie_2/tutorials/tutorial-2/tutorial-2b$ make sim
aiecc.py -j4 --xchesscc --xbridge --aiesim aie.mlir -I/home/marafin/mlir-aie_2/install/runtime_lib/x86_64/test_lib/include -L/home/marafin/mlir-aie_2/install/runtime_lib/x86_64/test_lib/lib -ltest_lib ./test.cpp
Found Vitis at /tools/Xilinx/Vitis/2023.2
 MLIR compilation: ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% -:--:-- 0:00:03 0/1 1 Workerwarning: overriding the module target triple with pdarch-unknown-unknown-elf [-Woverride-module]
1 warning generated.
 MLIR compilation: ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% -:--:-- 0:00:04 0/1 1 WorkerWarning in "../../../../../../tools/Xilinx/Vitis/2023.2/aietools/data/versal_prod/lib/me_chess.h", line 647, column 131: ignoring attribute [[deprecated]] on class declaration
 MLIR compilation: ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% -:--:-- 0:00:05 0/1 1 WorkerWarning in "../../../../../../tools/Xilinx/Vitis/2023.2/aietools/data/versal_prod/lib/me_common.h", line 63, column 132: ignoring attribute [[deprecated]] on class declaration
Warning in "../../../../../../tools/Xilinx/Vitis/2023.2/aietools/data/versal_prod/lib/me_common.h", line 96, column 131: ignoring attribute [[deprecated]] on class declaration
 AIE Compilation: ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% -:--:-- 0:00:00 0/2 4 Workers['-I/home/marafin/mlir-aie_2/install/runtime_lib/x86_64/test_lib/include', '-L/home/marafin/mlir-aie_2/install/runtime_lib/x86_64/test_lib/lib', '-ltest_lib', './test.cpp']
 AIE Compilation: ━━━━━━━━━━━━━━━━━━━━╺━━━━━━━━━━━━━━━━━━━  50% -:--:-- 0:00:00 1/2 4 Workerswarning: unknown warning option '-Wno-enum-constexpr-conversion' [-Wunknown-warning-option]
 AIE Compilation: ━━━━━━━━━━━━━━━━━━━━╺━━━━━━━━━━━━━━━━━━━  50% -:--:-- 0:00:01 1/2 4 WorkersIn file included from /home/marafin/mlir-aie_2/install/aie_runtime_lib/AIE/aiesim/genwrapper_for_ps.cpp:1:
In file included from /tools/Xilinx/Vitis/2023.2/aietools/include/adf/wrapper/wrapper.h:5:
/tools/Xilinx/Vitis/2023.2/aietools/include/adf/wrapper/me_ip_block.h:69:5: warning: inline variables are a C++17 extension [-Wc++17-extensions]
    inline static bool DepMsgPrinted = false;
    ^
 AIE Compilation: ━━━━━━━━━━━━━━━━━━━━╺━━━━━━━━━━━━━━━━━━━  50% -:--:-- 0:00:02 1/2 4 Workers2 warnings generated.
warning: unknown warning option '-Wno-enum-constexpr-conversion' [-Wunknown-warning-option]
 AIE Compilation: ━━━━━━━━━━━━━━━━━━━━╺━━━━━━━━━━━━━━━━━━━  50% -:--:-- 0:00:02 1/2 4 Workers1 warning generated.
ld.lld: error: -plugin-opt=Og: number expected, but got 'g'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
Error encountered while running: clang++ -O2 -fuse-ld=lld -shared -o /home/marafin/mlir-aie_2/tutorials/tutorial-2/tutorial-2b/aie.mlir.prj/sim/ps/ps.so /home/marafin/mlir-aie_2/install/aie_runtime_lib/AIE/aiesim/genwrapper_for_ps.cpp -D__AIEARCH__=10 -I/home/marafin/mlir-aie_2/install/runtime_lib/x86_64/test_lib/include -L/home/marafin/mlir-aie_2/install/runtime_lib/x86_64/test_lib/lib -ltest_lib ./test.cpp -fPIC -flto -fpermissive -DAIE_OPTION_SCALAR_FLOAT_ON_VECTOR -Wno-deprecated-declarations -Wno-enum-constexpr-conversion -Wno-format-security -DSC_INCLUDE_DYNAMIC_PROCESSES -D__AIESIM__ -D__PS_INIT_AIE__ -Og -Dmain(...)=ps_main(...) -I/home/marafin/mlir-aie_2/tutorials/tutorial-2/tutorial-2b/aie.mlir.prj -I/tools/Xilinx/Vitis/2023.2/aietools/include -I/tools/Xilinx/Vitis/2023.2/aietools/include/drivers/aiengine -I/tools/Xilinx/Vitis/2023.2/aietools/data/osci_systemc/include -I/tools/Xilinx/Vitis/2023.2/aietools/include/xtlm/include -I/tools/Xilinx/Vitis/2023.2/aietools/include/common_cpp/common_cpp_v1_0/include -I/home/marafin/mlir-aie_2/install/runtime_lib/x86_64/test_lib/include /home/marafin/mlir-aie_2/install/runtime_lib/x86_64/test_lib/lib/libmemory_allocator_sim_aie.a -L/tools/Xilinx/Vitis/2023.2/aietools/lib/lnx64.o -L/tools/Xilinx/Vitis/2023.2/aietools/data/osci_systemc/lib/lnx64 -Wl,--as-needed -lxioutils -lxaiengine -ladf_api -lsystemc -lxtlm -flto
 AIE Compilation: ━━━━━━━━━━━━━━━━━━━━╺━━━━━━━━━━━━━━━━━━━  50% -:--:-- 0:00:02 1/2 4 Workers
make: *** [Makefile:17: sim] Error 1
```

AndraBisca, 2024-02-14:
@tanvirarafin thank you for bringing this issue to our attention. We are currently looking into it. You said that after running `make sim`, a `sim` folder is generated. Is the folder empty or does it have other subfolders? Could you try adding an empty `.target` file inside of your `sim` folder and then running `make -C aie.mlir.prj/sim`?

tanvirarafin, 2024-02-14:
@AndraBisca `sim` folder has some contents. For example, for tutorial 2b, here are the contents:
```marafin@ubuntu:~/mlir-aie_2/tutorials/tutorial-2/tutorial-2b/aie.mlir.prj/sim$ ls -al
total 32
drwxrwxr-x 6 marafin marafin 4096 Feb 14 07:40 .
drwxrwxr-x 4 marafin marafin 4096 Feb 14 07:40 ..
drwxrwxr-x 2 marafin marafin 4096 Feb 14 07:40 arch
drwxrwxr-x 2 marafin marafin 4096 Feb 14 07:40 config
-rw-rw-r-- 1 marafin marafin  677 Feb 14 07:40 flows_physical.mlir
-rw-r--r-- 1 marafin marafin 2643 Feb 14 07:40 Makefile
drwxrwxr-x 2 marafin marafin 4096 Feb 14 07:40 ps
drwxrwxr-x 2 marafin marafin 4096 Feb 14 07:40 reports
```
After creating a `.target` file inside the `sim` directory and then running `make -C aie.mlir.prj/sim` works 👍. Here is the terminal output for that command:

```
marafin@ubuntu:~/mlir-aie_2/tutorials/tutorial-2/tutorial-2b$ make -C aie.mlir.prj/sim
make: Entering directory '/home/marafin/mlir-aie_2/tutorials/tutorial-2/tutorial-2b/aie.mlir.prj/sim'
cd ../..; aiesimulator --pkg-dir=./aie.mlir.prj/sim --dump-vcd foo
/tools/Xilinx/Vitis/2023.2/aietools/bin/aiesimulator: line 77: [: !=: unary operator expected
WARNING: XILINX_VITIS environment variable unset, python unavailable
AIEsim feature license is found.
INFO: Executing config: ./aie.mlir.prj/sim/config/scsim_config.json
[INFO]: If you are running trace or profile, Xilinx recommends to use --xlopt=0. This will give a greater visibility into the design. One can verify --xlopt level in the aiecompiler command line.
set use_simple_noc to: 1
Loading device config from: /tools/Xilinx/Vitis/2023.2/aietools/data/devices/VC1902.json
IP-INFO: [ps_ps_main] IP loaded.

Info: DEVICE FILE: /tools/Xilinx/Vitis/2023.2/aietools/data/devices/VC1902.json

Info: AIE SOLUTION FILE: ./aie.mlir.prj/sim/arch/aieshim_solution.aiesol
[AIESIM_OPTIONS]: aiesim_options file path aiesimulator_output/aiesim_options.txt
[INFO]: Disable Unused Tiles
[INFO]: Xpe File: /home/marafin/mlir-aie_2/tutorials/tutorial-2/tutorial-2b/./aie.mlir.prj/sim/reports/graph.xpe
INFO  : INFO: Running AIE MTMODEL Simulation with 1 threads
ISS disables unused tiles
Optimized ISS Wrapper (R+W), r2p24
INFO  : ROWS: 8 COLS: 50
Running Dispatch Server on port: 35653
Summary File: Warning - Can't find aie compiler summary file!
Current Directory=/home/marafin/mlir-aie_2/tutorials/tutorial-2/tutorial-2b
Enabled fast PM writes.
Enabled fast DM writes.
Info: (I702) default timescale unit used for tracing: 1 ps (foo.vcd)
[INFO]: Enabled Stream Switch Port Latency 
IP-INFO: [ps_ps_main] IP started.
Tutorial-2b test start.
Turning ecc off
Start cores
Acquired lock14_0 (1) in tile (1,4). Done.
Checking buf[3] = 14.
PASS!
Tutorial-2b test done.
Exiting!
Cores are done executing but the simulation will run for some more cycles to allow PLIO to be flushed
Stopping Simulator.

Info: /OSCI/SystemC: Simulation stopped by user.
IP-INFO: deleting ip PSIP_ps_i6 
[INFO] : Simulation Finished, Sim result: 0 Total Simulation time 5008800 ps
AIEsim feature license is released.
make: Leaving directory '/home/marafin/mlir-aie_2/tutorials/tutorial-2/tutorial-2b/aie.mlir.prj/sim'
```

AndraBisca, 2024-02-14:
Thank you @tanvirarafin for testing this temporary fix! While this isn't an ideal flow, you can at least run your designs using the simulator, even if you do encounter errors along the way. I have been able to reproduce this behaviour as well. We'll keep looking into this and leave the issue open.

AndraBisca, 2025-03-25:
Duplicate issue with [#976](https://github.com/Xilinx/mlir-aie/issues/976)

closed 2025-03-25 by AndraBisca