Xilinx/mlir-aie #1094: `isShimNOCTile` is wrong for 5x4 designs.

makslevental, 2024-03-05:
Physically I believe the (0,0) tile is the Shim PL Tile; if you set the partition start column to the 0th column, then `aie-rt` identifies that tile as the [shim PL tile](https://github.com/stephenneuendorffer/aie-rt/blob/4ce26c8cd2a635ec27055f4ddd689f954ed30660/driver/src/device/xaie_device_aieml.c#L59-L58). This is an issue for 5x4 designs because our [isShimNOCTile](https://github.com/Xilinx/mlir-aie/blob/a63eb5dac38c89257c94174d0cb214332083d597/include/aie/Dialect/AIE/IR/AIETargetModel.h#L468) does not take into account "virtualization" i.e. partition start column. Not sure what the fix is without extending the Target model to capture partition start.

jgmelber, 2024-03-06:
Yes, currently we consider the IPU target a virtualized 4x4 device to limit the complexity. The device model would need to be extended, perhaps with a special case for 5 column designs if we decide that this is something we want to support.

AndraBisca, 2025-03-25:
@jgmelber is this still an issue?

AndraBisca, 2025-05-28:
Closed due to inactivity. Please feel free to re-open.