Xilinx/mlir-aie #1105: Interfacing with the PL

medbzkst, 2024-03-07:
Following the tutorials, it is only clear how to interface with the DDR directly through the ShimDMA in [tutorial 5](https://github.com/Xilinx/mlir-aie/tree/main/tutorials/tutorial-5), i.e. from L1 directly to L3. What about interfacing with the PL using the PLIOs (to enable L2 or do some calculations while streaming)? [Tutorial 5](https://github.com/Xilinx/mlir-aie/tree/main/tutorials/tutorial-5) only states that

> Note that shim_dma are defined for the shim tiles (row 0). Also note that not every column in row 0 is shim_dma capable. The list of capable tiles in the S70 device is (2,3,6,7,10,11,18,19,26,27,34,35,42,43,46,47)

So is it exactly the same way with the 39 PL Interface tiles? If yes, what are the shim tiles that have the PL Interface tiles? And how to precise the data width carried by a PLIO as we do with the ADF API?

medbzkst, 2024-03-07:
By looking up the [AIETargetModel](https://github.com/Xilinx/mlir-aie/blob/5164b34812eedb6995b6bd95853a5573cd3c5abe/include/aie/Dialect/AIE/IR/AIETargetModel.h#L338) it seems to be that there is what is called ShimPL tiles that are mutually exclusive to ShimDMA tiles in the shim row.

However, having [16 ShimDMA tiles](https://github.com/Xilinx/mlir-aie/blob/5164b34812eedb6995b6bd95853a5573cd3c5abe/include/aie/Dialect/AIE/IR/AIETargetModel.h#L325) means that we are only left with 34 ShimPL tiles, whereas in [AM009](https://docs.xilinx.com/r/en-US/am009-versal-ai-engine/AI-Engine-to-Programmable-Logic-Interface#:~:text=In%20the%20VC1902%20device%2C%20there%20are%2050%20columns%20of%20AI%20Engine%20tiles%20and%20AI%20Engine%20array%20interface%20tiles%2C%20however%2C%20only%2039%20array%20interface%20tiles), it states that there 39 PL interface tiles. This is also visible on Vitis analyzer for a large enough design by experience.

How is that possible?

stephenneuendorffer, 2024-03-11:
See: https://support.xilinx.com/s/question/0D54U00007nlZntSAE/how-to-know-39-plio-locations?language=en_US

It's possible to query vivado for the exact information.  To my knowledge this is the only place the valid sites are documented.

```
link_design -part [get_parts xcvc1902-vsvd1760-2MP-i-L]
foreach tile [get_tiles -of [get_sites AIE_PL_*]] {puts $tile} // returns 39 tiles
foreach tile [get_tiles -of [get_sites AIE_PL_*]] {puts $tile} // returns 16 tiles.
```

In particular, note that at the very ends of the device, the NOC exists and can be used, even though there is not PL below the AIE tile

medbzkst, 2024-03-12:
Thanks @stephenneuendorffer! That was the most accurate way to figure it out. (I believe there is a typo in the last line of your commands where it should have been AIE_NOC instead of AIE_PL, right?)

Looking at the results 

```
# foreach tile [get_tiles -of [get_sites AIE_PL_*]] {puts $tile}
AIE_INTF_B1_CORE_X6Y0
AIE_INTF_B2_CORE_X7Y0
AIE_INTF_B3_CORE_X8Y0
AIE_INTF_B0_CORE_X9Y0
AIE_INTF_B1_CORE_X10Y0
AIE_INTF_B2_CORE_X11Y0
AIE_INTF_B3_CORE_X12Y0
AIE_INTF_B0_CORE_X13Y0
AIE_INTF_B1_CORE_X14Y0
AIE_INTF_B2_CORE_X15Y0
AIE_INTF_B3_CORE_X16Y0
AIE_INTF_B0_CORE_X17Y0
AIE_INTF_B1_CORE_X18Y0
AIE_INTF_B2_CORE_X19Y0
AIE_INTF_B3_CORE_X20Y0
AIE_INTF_B0_CORE_X21Y0
AIE_INTF_B1_CORE_X22Y0
AIE_INTF_B2_CORE_X23Y0
AIE_INTF_B3_CORE_X24Y0
AIE_INTF_B0_CORE_X25Y0
AIE_INTF_B1_CORE_X26Y0
AIE_INTF_B2_CORE_X27Y0
AIE_INTF_B3_CORE_X28Y0
AIE_INTF_B0_CORE_X29Y0
AIE_INTF_B1_CORE_X30Y0
AIE_INTF_B2_CORE_X31Y0
AIE_INTF_B3_CORE_X32Y0
AIE_INTF_B0_CORE_X33Y0
AIE_INTF_B1_CORE_X34Y0
AIE_INTF_B2_CORE_X35Y0
AIE_INTF_B3_CORE_X36Y0
AIE_INTF_B0_CORE_X37Y0
AIE_INTF_B1_CORE_X38Y0
AIE_INTF_B2_CORE_X39Y0
AIE_INTF_B3_CORE_X40Y0
AIE_INTF_B0_CORE_X41Y0
AIE_INTF_B1_CORE_X42Y0
AIE_INTF_B2_CORE_X43Y0
AIE_INTF_B3_CORE_X44Y0
# foreach tile [get_tiles -of [get_sites AIE_NOC_*]] {puts $tile}
AIE_INTF_B1_CORE_X2Y0
AIE_INTF_B2_CORE_X3Y0
AIE_INTF_B1_CORE_X6Y0
AIE_INTF_B2_CORE_X7Y0
AIE_INTF_B1_CORE_X10Y0
AIE_INTF_B2_CORE_X11Y0
AIE_INTF_B1_CORE_X18Y0
AIE_INTF_B2_CORE_X19Y0
AIE_INTF_B1_CORE_X26Y0
AIE_INTF_B2_CORE_X27Y0
AIE_INTF_B1_CORE_X34Y0
AIE_INTF_B2_CORE_X35Y0
AIE_INTF_B1_CORE_X42Y0
AIE_INTF_B2_CORE_X43Y0
AIE_INTF_B1_CORE_X46Y0
AIE_INTF_B2_CORE_X47Y0
```

As I have known before, on the AIE1 xcvc1902 device, we have 39 adjacent PL interface tiles centred in the shim row from 6 to 44, whereas the NOC/DMA tiles are the same as mentioned in [tutorial 5](https://github.com/Xilinx/mlir-aie/tree/main/tutorials/tutorial-5). That brings a point: ShimPL and ShimDMA tiles are **NOT** placed in a mutually exclusive manner as given in the [AIETargetModel](https://github.com/Xilinx/mlir-aie/blob/5164b34812eedb6995b6bd95853a5573cd3c5abe/include/aie/Dialect/AIE/IR/AIETargetModel.h#L323) of the xcvc1902. I believe that should be fixed. Otherwise, MLIR-AIE would not work properly when the PL is in the game, right?

stephenneuendorffer, 2024-03-12:
> That brings a point: ShimPL and ShimDMA tiles are **NOT** placed in a mutually exclusive manner as given in the [...]

Yes the model is incorrect here...  Care to make a patch?   Note that in shim tiles that include both a PL interface and a NOC interface, they are somewhat mutually exclusive since they share the stream interconnect.  Both cannot be used with full bandwidth. Would love to see people using the PL interface more!

jgmelber, 2024-10-22:
Maybe this is helpful #1623 and [passthrough_dmas_plio](https://github.com/Xilinx/mlir-aie/tree/main/programming_examples/basic/passthrough_dmas_plio)

closed 2025-03-25 by jgmelber