Xilinx/mlir-aie #1130: AIE_ENABLE_XRT_PYTHON_BINDINGS=ON is incompatible with AIE_ENABLE_PYTHON_PASSES=ON

keryell, 2024-03-15:
For example configuring with:
```bash
cmake \
  -GNinja \
  -DAIE_COMPILER=NONE \
  -DAIE_ENABLE_PYTHON_PASSES=ON \
  -DAIE_ENABLE_BINDINGS_PYTHON=ON \
  -DAIE_ENABLE_XRT_PYTHON_BINDINGS=ON \
  -DAIE_INCLUDE_INTEGRATION_TESTS=ON \
  -DAIE_LINKER=NONE \
  -DAIE_RUNTIME_TARGETS="x86_64;aarch64" \
  -DAIE_RUNTIME_TEST_TARGET=aarch64 \
  -DCMAKE_BUILD_TYPE=Debug \
  -DCMAKE_EXPORT_COMPILE_COMMANDS=ON \
  -DCMAKE_INSTALL_PREFIX=$MLIR_AIE_HOME/../install-mlir-aie \
  -DCMAKE_MODULE_PATH=$MLIR_AIE_HOME/cmake/modulesXilinx \
  -DHOST_COMPILER=NONE \
  -DLLVM_DIR=$LLVM_DIR/../install-llvm/lib/cmake/llvm \
  -DLLVM_ENABLE_ASSERTIONS=ON \
  -DLLVM_EXTERNAL_LIT=$LLVM_DIR/build/bin/llvm-lit \
  -DLLVM_USE_LINKER=lld \
  -DMLIR_DIR=$LLVM_DIR/../install-llvm/lib/cmake/mlir \
  -DXRT_BIN_DIR=$XILINX_XRT/bin \
  -DXRT_INCLUDE_DIR=$XILINX_XRT/include \
  -DXRT_LIB_DIR=$XILINX_XRT/lib \
  -B $MLIR_AIE_HOME/build -S $MLIR_AIE_HOME
```
leads to:
```
Testing with AIE runtime target: x86_64
Building AIE runtime for x86_64 with TOOLCHAIN=/home/rkeryell/Xilinx/Projects/AIE/ACDC/mlir-aie/runtime_lib/../cmake/toolchainFiles/toolchain_x86_64.cmake
LibXAIE_x86_64_DIR not specified: Building xaiengine for x86_64 from Vitis at /opt/xilinx/Vitis/2023.2.
-- setting up test_lib
Building AIE runtime for aarch64 with TOOLCHAIN=/home/rkeryell/Xilinx/Projects/AIE/ACDC/mlir-aie/runtime_lib/../cmake/modulesXilinx/toolchain_vitis_crosscomp_arm.cmake
LibXAIE_aarch64_DIR not specified: Building xaiengine for aarch64 from Vitis at /opt/xilinx/Vitis/2023.2.
-- setting up test_lib
xaienginecdo_static xaiengine/aie-rt/driver/src /home/rkeryell/Xilinx/Projects/AIE/ACDC/mlir-aie/build/include
CMake Error at python/CMakeLists.txt:374 (add_custom_command):
  No TARGET 'AIEPythonModules.extension._xrt.dso' has been created in this
  directory.
```

makslevental, 2024-03-15:
Dunno what's up but you shouldn't have `AIE_ENABLE_PYTHON_PASSES=ON` (unless you're really actually curious about the "black magic" as you called it...). Is that on somewhere?

keryell, 2024-03-15:
I do not know why I had this. Probably some old compilation recipe?

makslevental, 2024-03-15:
:shrug:

closed 2024-03-15 by makslevental

keryell, 2024-03-20:
But how to validate for example `test/python/python_passes.py`?

makslevental, 2024-03-20:
Again: are you really sure you want to? Right now there's only one such pass ([this one](https://github.com/Xilinx/mlir-aie/pull/761)) and those are thoroughly validated because well `random_seed` is fixed and so the solve is deterministic :shrug:. I had plans for much more here but priorities got in the way so there's literally nothing else to verify.

Maybe I should flip this question-answer session: what are you trying to accomplish here with that ("python passes") infra?

keryell, 2024-03-20:
I am just looking at all the tests to have a maximum of them running on my laptop. So I am curious about all the unsupported tests too, including this one.

makslevental, 2024-03-20:
> I am just looking at all the tests to have a maximum of them running on my laptop. So I am curious about all the unsupported tests too, including this one.

Yea so without a doubt, if you're trying to avoid long-ish runtimes for the testsuite, probably best to avoid the tests that include an ILP solve :smile:.

keryell, 2024-03-20:
> Yea so without a doubt, if you're trying to avoid long-ish runtimes for the testsuite, probably best to avoid the tests that include an ILP solve 😄.

Ouch! The name `test/python/python_passes.py` sounded so innocuous to me . :-)