Xilinx/mlir-aie #1200: use after free in `AIEObjectFifoRegisterProcess`

makslevental, 2024-04-09:
https://github.com/Xilinx/mlir-aie/blob/abbcb9a05a37e33217db8a0771594b81e3c4b95f/lib/Dialect/AIE/Transforms/AIEObjectFifoRegisterProcess.cpp#L147-L149

the above is a use after free that has gone undetected until now for whatever reason (undefined behavior?). Compiling against LLVM-17 caught it [here](https://github.com/Xilinx/mlir-aie/actions/runs/8619873012/job/23625619735#step:8:993) (note that gcc does not care/complain). I'm going to fix this one but this pattern actually appears in many many places in the code base.

AndraBisca, 2024-04-09:
Thank you Maks. I apologize, I haven't maintained this code in quite a while and it's not used aside from these tests (I actually wanted to move it to the AIEX dialect but never got to it). Did you say you fixed it already? Otherwise, I can take a look at it tomorrow, let me know!

makslevental, 2024-04-09:
> Did you say you fixed it already? Otherwise, I can take a look at it tomorrow, let me know!

It's fixed in https://github.com/Xilinx/mlir-aie/pull/1206 (I believe :crossed_fingers:) but it would be good to do an audit and track down wherelse this kind of `getOps` type thing is done. Of course that's not like a command to you, but more of a mental note to myself to remember to do it.

makslevental, 2024-04-09:
But also if this is unused we should remove it - the best PR is one that deletes code :smile:

AndraBisca, 2024-04-09:
Thank you Maks! I'll make the same kind of note to myself and pay attention if I catch it while working on the repo. I also agree with potentially deleting that code. I'll give it a harder think after the next few weeks and either delete it or move it to my fork.

closed 2024-04-10 by makslevental