Xilinx/mlir-aie #1314: Update required for Linux Ryzen AI setup instructions

Yu-Zhewen, 2024-04-19:
https://github.com/Xilinx/mlir-aie/blob/main/docs/buildHostLin.md

1. In the BIOS settings, besides disable **Secure Boot**, also check if **Advanced -> CPU Configuration->IPU** is already enabled. This is a known issue as https://github.com/amd/xdna-driver/issues/14. On a minisforum PC, the IPU is actually disabled out of the box.

2. The current  XDNA Driver installation specifies a checkpoint that stops working.  This is due to the broken gitlab links to the firmware  files https://github.com/amd/xdna-driver/blob/317e0c67747cbf88e5b5a3a81ba4bdf7bf5b3fc3/tools/info.json.

jgmelber, 2024-04-19:
This is partially addressed in #1311.

We are working on verifying the latest xdna-driver with our repository: #1211

AndraBisca, 2024-07-10:
Good to close @jgmelber ?

closed 2024-07-12 by Yu-Zhewen

fxmarty-amd, 2024-09-13:
Is disabling secureboot still a requirement to use mlir-aie with a specific driver version in https://github.com/Xilinx/mlir-aie/blob/main/docs/buildHostWin.md?

I wonder if my issue https://github.com/Xilinx/mlir-aie/issues/1757 could be linked, as I have secureboot enabled.