Xilinx/mlir-aie #1448: upstream emitc dialect change breaks aievec to cpp lowering

fifield, 2024-05-02:
bumping llvm results in these failures:
```
  AIE :: unit_tests/aievec_tests/bf16_exp_lut/bf16_exp_lut.mlir
  AIE :: unit_tests/aievec_tests/bf16_inv_lut/bf16_inv_lut.mlir
  AIE :: unit_tests/aievec_tests/bf16_rsqrt_v16/bf16_rsqrt.mlir
  AIE :: unit_tests/aievec_tests/bf16_rsqrt_v32/bf16_rsqrt.mlir
  AIE :: unit_tests/aievec_tests/bf16_softmax/bf16_softmax.mlir
  AIE :: unit_tests/aievec_tests/bf16_softmax_2/bf16_softmax.mlir
  AIE :: unit_tests/aievec_tests/bf16_sqrt_v16/bf16_sqrt.mlir
  AIE :: unit_tests/aievec_tests/bf16_sqrt_v32/bf16_sqrt.mlir
  AIE :: unit_tests/aievec_tests/i16_abs_v32_2/i16_abs.mlir
  AIE :: unit_tests/aievec_tests/i32_abs_v16_2/i32_abs.mlir
  AIE :: unit_tests/aievec_tests/i8_abs_v64_2/i8_abs.mlir
  AIE_TUTORIALS :: ml/softmax/run_makefile.lit
```

see discussion in https://github.com/llvm/llvm-project/pull/88391

jsetoain, 2024-05-02:
> bumping llvm results in these failures:
> 
> ```
>   AIE :: unit_tests/aievec_tests/bf16_exp_lut/bf16_exp_lut.mlir
>   AIE :: unit_tests/aievec_tests/bf16_inv_lut/bf16_inv_lut.mlir
>   AIE :: unit_tests/aievec_tests/bf16_rsqrt_v16/bf16_rsqrt.mlir
>   AIE :: unit_tests/aievec_tests/bf16_rsqrt_v32/bf16_rsqrt.mlir
>   AIE :: unit_tests/aievec_tests/bf16_softmax/bf16_softmax.mlir
>   AIE :: unit_tests/aievec_tests/bf16_softmax_2/bf16_softmax.mlir
>   AIE :: unit_tests/aievec_tests/bf16_sqrt_v16/bf16_sqrt.mlir
>   AIE :: unit_tests/aievec_tests/bf16_sqrt_v32/bf16_sqrt.mlir
>   AIE :: unit_tests/aievec_tests/i16_abs_v32_2/i16_abs.mlir
>   AIE :: unit_tests/aievec_tests/i32_abs_v16_2/i32_abs.mlir
>   AIE :: unit_tests/aievec_tests/i8_abs_v64_2/i8_abs.mlir
>   AIE_TUTORIALS :: ml/softmax/run_makefile.lit
> ```
> 
> see discussion in [llvm/llvm-project#88391](https://github.com/llvm/llvm-project/pull/88391)

I'm on it, thanks!

jsetoain, 2024-05-03:
@fifield #1453 addresses this issue. There are a lot of failures I suspect are related to the CI configuration, I can probably use your help and/or Maks' for that, I'm not sure what is it that I'm missing. The tests themselves pass in my local machine.

jsetoain, 2024-05-03:
> @fifield #1453 addresses this issue. There are a lot of failures I suspect are related to the CI configuration, I can probably use your help and/or Maks' for that, I'm not sure what is it that I'm missing. The tests themselves pass in my local machine.

I've been looking around, and I believe there are a bunch of wheel packages that need to be bumped before CI tests with the new version will pass. If I go back to the previous version, tests pass both upstream and downstream, it should be ready to bump (and I got rid of quite a lot of warnings in the process).

closed 2024-05-03 by fifield